esphome/issues #1585: Tuya MCU - Light Integration Not Scaling Correctly or Enforcing Min_Value.

grrraham, 2020-10-28:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io image installed on a Raspberry Pi 4 (everything updated as of 29/10/2020).
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP: ESP8266 in Moes Dimmable Wall Switches, set as: 
platform: ESP8266
board: esp01_1m

**ESPHome version (latest production, beta, dev branch)**

ESPHome (dev) 

**Affected component:**

Tuya MCU
Tuya Dimming

**Description of problem:**
So I have Moes Dimmable (wall) light switches.
They have large rectangular buttons with touch sensitive dimmable strips up the middle. 
The older style with the rounded corners on the buttons work fine with Tuya Dimmer integration and  0-255 (default) scaling.
The newer style with the more square corners on the button, requires scaling, and HomeAssistant / the button has issues.
After a power cycle, the dimmer wall switch will work perfectly at the device.
If you play with it from home assistant and set the brightness to minimum, ESPHome will let the Brightness go below the
minimum setpoint (150) and this breaks the switch such that the local brightness control will no longer work.

Can the Tuya Light (dimmer) integration be updated so that the min_value and max_value get scaled between 0-100% in Home Assistant?
Can the Tuya Light (dimmer) integration be updated so that the min_value and max_value are 'respected' and it is not possible to send a value lower than the min_value?

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: dimmer_lounge
  platform: ESP8266
  board: esp01_1m

wifi:
  networks:
  - ssid: !secret wifi_SSD1
    password: !secret wifi_password
  # Optional manual IP
  manual_ip:
    static_ip: 192.168.0.18  
    gateway: 192.168.0.1
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Lounge Dimmer Hotspot"
    password: !secret wifi_password2

# Make sure logging is not using the serial port
logger:
  baud_rate: 0

# Enable Home Assistant API
api:

# Make sure you can upload new firmware OTA
ota:

time:
  - platform: homeassistant
    id: ha_time
    
uart:
  rx_pin: GPIO3
  tx_pin: GPIO1
  baud_rate: 9600

# Register the Tuya MCU connection
tuya:

light:
  - platform: "tuya"
    name: "Lounge"
    id: dimmableoutput
    dimmer_datapoint: 3
    switch_datapoint: 1
    max_value: 1000
    min_value: 150 
#The issue seems to be with the min_value.   When the value is equal to 150, Home Assistant is still showing 15% instead of 0%.  
# If I move the slider to 0% in home assistant, a number lower than 150 is sent to the ESP which then upsets (breaks) the 
# microcontroller on the other end of the serial comms.

# This sensor was added to try to work out what was going on.
# This sensor seems to be feed back from the microcontroller to say what  the output (and local slider display) are set to.
sensor:  
  - platform: "tuya"
    name: "lounge data2"
    id: dimmablewallswitch
    sensor_datapoint: 2

```

**Logs (if applicable):**

```
[00:34:03][C][logger:185]: Logger:
[00:34:03][C][logger:186]:   Level: DEBUG
[00:34:03][C][logger:187]:   Log Baud Rate: 0
[00:34:03][C][logger:188]:   Hardware UART: UART0
[00:34:03][C][light:178]: Light 'Lounge'
[00:34:03][C][light:180]:   Default Transition Length: 0.0s
[00:34:03][C][light:181]:   Gamma Correct: 1.00
[00:34:03][C][tuya.light:027]: Tuya Dimmer:
[00:34:03][C][tuya.light:029]:    Dimmer has datapoint ID 3
[00:34:03][C][tuya.light:031]:    Switch has datapoint ID 1
[00:34:03][C][tuya.sensor:028]: Tuya Sensor 'lounge data2'
[00:34:03][C][tuya.sensor:028]:   Unit of Measurement: ''
[00:34:03][C][tuya.sensor:028]:   Accuracy Decimals: 0
[00:34:03][C][tuya.sensor:029]:   Sensor has datapoint ID 2
[00:34:03][C][homeassistant.time:010]: Home Assistant Time:
[00:34:03][C][homeassistant.time:011]:   Timezone: 'AEST-10'
[00:34:03][C][ota:029]: Over-The-Air Updates:
[00:34:03][C][ota:030]:   Address: 192.168.2.18:8266
[00:34:03][C][api:095]: API Server:
[00:34:03][C][api:096]:   Address: 192.168.2.18:6053
[00:34:03][C][tuya:023]: Tuya:
[00:34:03][C][tuya:034]:   Datapoint 2: int value (value: 1000)
[00:34:03][C][tuya:032]:   Datapoint 1: switch (value: ON)
[00:34:03][C][tuya:034]:   Datapoint 3: int value (value: 1000)
[00:34:03][C][tuya:036]:   Datapoint 4: enum (value: 0)
[00:34:03][C][tuya:046]:   Product: '{"p":"r3fg1ytyd4sd67ga","v":"4.1.3","m":0}'
[00:34:09][D][light:265]: 'Lounge' Setting:
[00:34:09][D][light:274]:   State: OFF
[00:34:09][D][sensor:092]: 'lounge data2': Sending state 1000.00000  with 0 decimals of accuracy
[00:34:09][D][tuya.sensor:016]: MCU reported sensor is: 1000
[00:34:09][D][light:265]: 'Lounge' Setting:
[00:34:10][D][light:265]: 'Lounge' Setting:
[00:34:11][D][light:274]:   State: ON
[00:34:11][D][sensor:092]: 'lounge data2': Sending state 1000.00000  with 0 decimals of accuracy
[00:34:11][D][tuya.sensor:016]: MCU reported sensor is: 1000
[00:34:11][D][light:265]: 'Lounge' Setting:
[00:34:13][D][light:265]: 'Lounge' Setting:
[00:34:13][D][light:278]:   Brightness: 84%
[00:34:14][D][sensor:092]: 'lounge data2': Sending state 843.00000  with 0 decimals of accuracy
[00:34:14][D][tuya.sensor:016]: MCU reported sensor is: 843
[00:34:14][D][light:265]: 'Lounge' Setting:
[00:34:14][D][light:278]:   Brightness: 84%
[00:34:15][D][light:265]: 'Lounge' Setting:
[00:34:15][D][light:278]:   Brightness: 45%
[00:34:15][D][sensor:092]: 'lounge data2': Sending state 450.00000  with 0 decimals of accuracy
[00:34:15][D][tuya.sensor:016]: MCU reported sensor is: 450
[00:34:15][D][light:265]: 'Lounge' Setting:
[00:34:15][D][light:278]:   Brightness: 45%
[00:34:16][D][light:265]: 'Lounge' Setting:
[00:34:16][D][light:278]:   Brightness: 24%
[00:34:17][D][sensor:092]: 'lounge data2': Sending state 235.00000  with 0 decimals of accuracy
[00:34:17][D][tuya.sensor:016]: MCU reported sensor is: 235
[00:34:17][D][light:265]: 'Lounge' Setting:
[00:34:17][D][light:278]:   Brightness: 24%
[00:34:18][D][light:265]: 'Lounge' Setting:
[00:34:18][D][light:278]:   Brightness: 17%
[00:34:18][D][sensor:092]: 'lounge data2': Sending state 172.00000  with 0 decimals of accuracy
[00:34:18][D][tuya.sensor:016]: MCU reported sensor is: 172
[00:34:18][D][light:265]: 'Lounge' Setting:
[00:34:18][D][light:278]:   Brightness: 17%
[00:34:20][D][light:265]: 'Lounge' Setting:
[00:34:20][D][light:278]:   Brightness: 0%
[00:34:20][D][sensor:092]: 'lounge data2': Sending state 150.00000  with 0 decimals of accuracy
[00:34:20][D][tuya.sensor:016]: MCU reported sensor is: 150
[00:34:20][D][light:265]: 'Lounge' Setting:
[00:34:20][D][light:278]:   Brightness: 15%

```

**Additional information and things you've tried:**

So in the above log I change the output in home assistant from 100% down to 0%.  You can see that 0% = 0 and not 150.
You can also see that when the input was set to 0%, the wall switch responded with 150 and changed the input back to 150.
In the config I have 150 as the minimum input.

cprussin, 2020-10-29:
I was just about to open a similar ticket, but I'm guessing this issue is caused by esphome not synchronizing the minimum brightness setting with the tuya mcu. For instance, in Tasmota, you must manually do something like `Backlog TuyaSend2 3,10; DimmerRange 10,1000` to set both the MCU dimmer range and the Tasmota dimmer range to match each other. This is a total hunch, but my symptoms across devices some of which previously had Tasmota and some that did not leads me to believe that esphome is only setting the dimming range in esphome itself and not the MCU.

cprussin, 2020-10-29:
Also and loosely related but I don't see any way to change the dimming mode with esphome. In Tasmota that too would be done with manual `TuyaSend` commands. So it seems that controlling the MCU settings in general is a missing feature in esphome.

cprussin, 2020-11-01:
@grrraham I think I may have fixed your issue in https://github.com/esphome/esphome/pull/1347.  By any chance could you build that branch and run it on your device and see if that does indeed fix your issue?

grrraham, 2020-11-01:
Hi Connor, I would love too give it a whirl, I just need a little more instruction on how to actually do that. I just have ESP Home (dev) installed via a RPi4.

closed 2020-11-05 by jesserockz

grrraham, 2020-11-17:
I may be too late to the party…  However I updated the  Hassio ESPhome Dev install and it accepted the new commands.
The dimming wall switches are working at 10% minimum dimming (they don’t work at 5%) and the wife approval factor has not dropped. (which is a good thing).

Thankyou Connor for your efforts!

Closed #1585<https://github.com/esphome/issues/issues/1585> via esphome/esphome#1347<https://github.com/esphome/esphome/pull/1347>.