esphome/issues #1587: APDS-9960 proximity breaks with gestures

dedepene, 2020-10-29:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Supervised HA on Ubuntu, always updated to the latest
**ESP (ESP32/ESP8266, Board/Sonoff):**

NodeMCU v2 w/ ESP8266
**ESPHome version (latest production, beta, dev branch)**

ESPHome version 1.15.3 compiled on Oct 28 2020, 19:38:24
**Affected component:**

https://esphome.io/components/sensor/apds9960.html

**Description of problem:**
Proximity alone works fine - it shows some percentage (0-100), greater as you get nearer the sensor (100% reading). The proximity reading breaks as soon as gestures are introduced by defining a binary sensor (LEFT & RIGHT in my setup). Left and right gestures seem to work fine but proximity is detected up to 16% - anything closer to the sensor doesn't change the 16% that also stops to update (as it should every 1s in my setup).

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: capacitive
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: "ssid"
  password: "pass"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Capacitive Fallback Hotspot"
    password: "pass"

captive_portal:

# Enable logging
logger:
  level: verbose

# Enable Home Assistant API
api:

ota:

i2c:
  sda: D2
  scl: D1
  scan: True

apds9960:
  address: 0x39
  update_interval: 1s
  
sensor:
  - platform: apds9960
    type: PROXIMITY
    name: "APDS9960 Proximity Channel"
    
binary_sensor:
  - platform: apds9960
    direction: LEFT
    name: "APDS960 Left Movement"
    
  - platform: apds9960
    direction: RIGHT
    name: "APDS960 Right Movement"
```

**Logs (if applicable):**

[... 144 lines cut ...]
[11:38:51][D][sensor:092]: 'APDS9960 Proximity Channel': Sending state 3.52941 % with 1 decimals of accuracy
[11:38:52][D][apds9960:206]: Got proximity=4.7%
[11:38:52][V][sensor:013]: 'APDS9960 Proximity Channel': Received new state 4.705883
[11:38:52][D][sensor:092]: 'APDS9960 Proximity Channel': Sending state 4.70588 % with 1 decimals of accuracy
[11:38:53][D][apds9960:206]: Got proximity=10.2%
[11:38:53][V][sensor:013]: 'APDS9960 Proximity Channel': Received new state 10.196078
[11:38:53][D][sensor:092]: 'APDS9960 Proximity Channel': Sending state 10.19608 % with 1 decimals of accuracy
[11:38:54][D][apds9960:206]: Got proximity=16.1%
[11:38:54][V][sensor:013]: 'APDS9960 Proximity Channel': Received new state 16.078432
[11:38:54][D][sensor:092]: 'APDS9960 Proximity Channel': Sending state 16.07843 % with 1 decimals of accuracy
[11:39:00][D][apds9960:206]: Got proximity=0.0%
[11:39:00][V][sensor:013]: 'APDS9960 Proximity Channel': Received new state 0.000000
[11:39:00][D][sensor:092]: 'APDS9960 Proximity Channel': Sending state 0.00000 % with 1 decimals of accuracy
[11:39:01][D][apds9960:206]: Got proximity=0.0%
[11:39:01][V][sensor:013]: 'APDS9960 Proximity Channel': Received new state 0.000000
[11:39:01][D][sensor:092]: 'APDS9960 Proximity Channel': Sending state 0.00000 % with 1 decimals of accuracy
[11:39:02][D][apds9960:206]: Got proximity=0.0%
[11:39:02][V][sensor:013]: 'APDS9960 Proximity Channel': Received new state 0.000000
[11:39:02][D][sensor:092]: 'APDS9960 Proximity Channel': Sending state 0.00000 % with 1 decimals of accuracy
[11:39:03][D][apds9960:206]: Got proximity=0.0%
[11:39:03][V][sensor:013]: 'APDS9960 Proximity Channel': Received new state 0.000000
[11:39:03][D][sensor:092]: 'APDS9960 Proximity Channel': Sending state 0.00000 % with 1 decimals of accuracy
```

**Additional information and things you've tried:**

nixx3d, 2020-11-16:
I can confirm this issues with the RobotDyn version of the APDS.
When using gestures the proximity / Ambient (CLEAR) & RGB do not work.

dedepene, 2020-12-14:
Hello? Is this even being recognized as an issue at all?

dedepene, 2021-04-15:
This is still an active issue, rendering the sensor useless.

dedepene, 2021-04-30:
Has this been looked into?

lordrak007, 2021-08-16:
Hello, i have face same problem. I am using proximity and when enable binary sensor for movement, apds stuck completely. So proximity stop working and gestures does not work too.
In verbose mode i have got this error: [09:43:24][V][apds9960:222]: FIFO buffer has filled to capacity!