esphome/issues #1588: WiFi is broken on ESP8266 when using a display modul

netpok, 2020-10-29:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 nodemcu clone

**ESPHome version (latest production, beta, dev branch)**
1.15.3

**Affected component:**

[GPIO LCD](https://esphome.io/components/display/lcd_display.html#gpio) and probably all displays

**Description of problem:**
I tried to use my old 16x2 gpio lcd so I set up config A from below and got the attached error:

The problem is present whether the display is connected or not.

**Problem-relevant YAML-configuration entries:**
Base config:
```yaml
esphome:
  name: scaledisplay
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
# debug:

# # Enable logging
logger:
#   baud_rate: 0

ota:

web_server:
  port: 80

time:
  - platform: sntp

mqtt:
  broker: !secret mqtt_broker
  username: !secret mqtt_username
  password:  !secret mqtt_password

# <INSERT CONFIG HERE>
```

A) GPIO LCD
```yaml
display:
  - platform: lcd_gpio
    dimensions: 16x2
    data_pins:
      - D5
      - D6
      - D7
      - D8
    enable_pin: D3
    rs_pin: D4
    lambda: |-
      it.print("Hello World!");
```

B) I2C LCD
```yaml
i2c:
  sda: D0
  scl: D1

display:
  - platform: lcd_pcf8574
    dimensions: 18x4
    address: 0x3F
    lambda: |-
      it.print("Hello World!");
```

C) I2C SSD1306
```yaml
i2c:
  sda: D0
  scl: D1

font:
  - file: "a.ttf"
    id: afont
    size: 20

display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x64"
    reset_pin: D0
    address: 0x3C
    lambda: |-
      it.print(0, 0, id(afont), "Hello World!");
```

D) Simple I2C
```yaml
# Example configuration entry
i2c:
  sda: D0
  scl: D1
```

**Logs (if applicable):**

```
[I][logger:166]: Log initialized
[21:35:07][C][ota:366]: There have been 2 suspected unsuccessful boot attempts.
[21:35:07][I][app:029]: Running through setup()...
[21:35:07][C][lcd_gpio:010]: Setting up GPIO LCD Display...
[21:35:07][C][wifi:033]: Setting up WiFi...
[21:35:07][D][wifi:324]: Starting scan...
[21:35:13][D][wifi:339]: Found networks:
[21:35:13][I][wifi:385]: - '<REMOVED>' (<REMOVED>) ▂▄▆█
[21:35:13][D][wifi:386]:     Channel: 1
[21:35:13][D][wifi:387]:     RSSI: -75 dB
[21:35:13][D][wifi:389]: - <REMOVED: a list of like 15 other networks>
[21:35:14][I][wifi:194]: WiFi Connecting to <REMOVED>...
[21:35:24][I][wifi:457]: WiFi Connected!
[21:35:24][C][wifi:303]:   SSID: <REMOVED>
[21:35:24][C][wifi:304]:   IP Address: 192.168.114.190
[21:35:24][C][wifi:306]:   BSSID: <REMOVED>
[21:35:24][C][wifi:307]:   Hostname: 'scaledisplay'
[21:35:24][C][wifi:311]:   Signal strength: -88 dB ▂▄▆█
[21:35:24][C][wifi:315]:   Channel: 1
[21:35:24][C][wifi:316]:   Subnet: 255.255.255.0
[21:35:24][C][wifi:317]:   Gateway: 192.168.114.1
[21:35:24][C][wifi:318]:   DNS1: 192.168.114.1
[21:35:24][C][wifi:319]:   DNS2: (IP unset)
[21:35:24][C][web_server:068]: Setting up web server...
[21:35:24][C][ota:029]: Over-The-Air Updates:
[21:35:24][C][ota:030]:   Address: scaledisplay.local:8266
[21:35:24][W][ota:036]: Last Boot was an unhandled reset, will proceed to safe mode in 8 restarts
[21:35:24][C][mqtt:025]: Setting up MQTT...
[21:35:24][D][mqtt:103]: Resolving MQTT broker IP address...
[21:35:30][W][mqtt:127]: Couldn't resolve IP address for 'mosquitto.lan'!
[21:35:30][D][mqtt:103]: Resolving MQTT broker IP address...
[21:35:32][W][wifi_esp8266:395]: Event: Disconnected ssid='<REMOVED>' bssid=<REMOVED> reason='Beacon Timeout'
[21:35:33][W][wifi:100]: WiFi Connection lost... Reconnecting...
[21:35:33][W][wifi:522]: Restarting WiFi adapter...
[21:35:33][I][wifi:194]: WiFi Connecting to '<REMOVED>'...
[21:35:33][W][wifi_esp8266:191]: wifi_apply_hostname_(scaledisplay): lwIP error -16 on interface st (index 0)
[21:35:37][W][mqtt:127]: Couldn't resolve IP address for 'mosquitto.lan'!
[21:35:37][D][mqtt:103]: Resolving MQTT broker IP address...
[21:35:39][W][wifi_esp8266:392]: Event: Disconnected ssid=<REMOVED> reason='Probe Request Unsuccessful'
[21:35:39][W][wifi:485]: Error while connecting to network.
[21:35:39][W][wifi:522]: Restarting WiFi adapter...
[21:35:44][D][wifi:324]: Starting scan...
[21:35:44][W][mqtt:127]: Couldn't resolve IP address for 'mosquitto.lan'!
[21:35:44][D][mqtt:103]: Resolving MQTT broker IP address...
[21:35:50][D][wifi:339]: Found networks:
[21:35:50][D][wifi:389]: - <REMOVED: one or mostly zero networks not including mine>
[21:35:50][W][wifi:394]: No matching network found!
```

**Additional information and things you've tried:**
After playing around with the config I noticed the problem is only present if the display is set up. On further testing the problem is still present with config B and C (but I have none of those displays) and not if only the i2c is configured (config D).

I tried the same code with an ESP32 and it had no connection issues (but as it turned out my display is 5v only).

Since the i2c displays were missing I could imagin some problems but none should result in the connection loss and as far as I know the GPIO lcd is write only.

***Extra information:***
My access point is Unifi AP-AC-PRO which had some problems with earlier arduino cores, but since the second scan does not pick up the other networks either I don't think it's the issue here.

deftdawg, 2021-01-11:
Maybe try setting  
```
wifi: 
 power_save_mode: none
```
If you haven't already

moe01324, 2021-02-03:
Experiencing the same problem. Also with "none" power_save_mode.
My display config:
```
font:
  - file: 'arial.ttf'
    id: font3
    size: 15  

spi:
  clk_pin: D5
  mosi_pin: D7

display:
  - platform: ssd1351_spi
    model: "SSD1351 128x128"
    reset_pin: D1
    cs_pin: D8
    dc_pin: D2
    lambda: |-
      it.print(0, 0, id(font3), "Hello World!");
```

Error log:

> [22:31:26][I][logger:166]: Log initialized
> [22:31:26][C][ota:366]: There have been 1 suspected unsuccessful boot attempts.
> [22:31:26][I][app:029]: Running through setup()...
> [22:31:26][C][spi:022]: Setting up SPI bus...
> [22:31:26][D][binary_sensor:034]: 'button_kitchen': Sending initial state OFF
> [22:31:26][C][rotary_encoder:127]: Setting up Rotary Encoder 'knob_buro'...
> [22:31:26][C][ssd1351_spi:011]: Setting up SPI SSD1351...
> [22:31:28][C][wifi:033]: Setting up WiFi...
> [22:31:28][W][wifi_esp8266:067]: Setting WiFi mode failed!
> [22:31:28][W][wifi_esp8266:067]: Setting WiFi mode failed!
> [22:31:28][D][wifi:324]: Starting scan...
> [22:31:28][W][wifi_esp8266:067]: Setting WiFi mode failed!

Casino, 2021-02-07:
Same issue here with esp8266! The SPI display works when wifi is disabled. Otherwise an exception 29 occurs.

yaml:
`spi:
  clk_pin: D5
  mosi_pin: D7

font:
  - file: 'slkscr.ttf'
    id: font1
    size: 8

display:
  - platform: ssd1351_spi
    model: "SSD1351 128x128"
    reset_pin: D1
    cs_pin: D6
    dc_pin: D2
    lambda: |-
      it.print(0, 0, id(font1), "Hello World!");`

log:

> [22:04:57]Exception (28):
←[33mWARNING Exception type: unknown←[0m
[22:04:57]epc1=0x4021b761 epc2=0x00000000 epc3=0x00000000 excvaddr=0x0000000b depc=0x00000000
←[32mINFO Need to fetch platformio IDE-data, please stand by←[0m
←[32mINFO Running:  platformio run -d esp8266_hydro_sensors -t idedata←[0m
[22:05:00]
[22:05:00]>>>stack>>>
←[33mWARNING Found stack trace! Trying to decode it←[0m
[22:05:00]
[22:05:00]ctx: sys
[22:05:00]sp: 3fffeb70 end: 3fffffb0 offset: 0190
[22:05:00]3fffed00:  3fffb868 40215eb9 00000020 40100f96
[22:05:00]3fffed10:  0000003a 00000006 00000000 4021bb5d
[22:05:00]3fffed20:  3fffed70 3fffed58 00000018 4021c634
[22:05:00]3fffed30:  3fffb868 00000000 3fffb834 40215ec2
[22:05:00]3fffed40:  3fffbe00 000d000f 80ffb834 72677000
[22:05:00]3fffed50:  00656461 80ffb868 65636300 3f007470
[22:05:00]3fffed60:  80000090 3fffbe00 000d000f 8021b947
[22:05:00]3fffed70:  3fffbd44 000f000f 0000746e 00000008
[22:05:00]3fffed80:  80000560 00000035 3fffb868 00000036
[22:05:00]3fffed90:  3fffbb87 00000034 3fffb834 40217619
[22:05:00]3fffeda0:  00000005 00000000 00000020 00000036
[22:05:00]3fffedb0:  3fffbb87 00000059 3fffb868 4021bd48
[22:05:00]3fffedc0:  3fffbb87 40104d93 3ffee890 00000036
[22:05:00]3fffedd0:  3fffbb87 00000059 3fffb834 40217830
[22:05:00]3fffede0:  00000000 00000000 4bc6a7f0 00000000
[22:05:00]3fffedf0:  3ffeb700 00000000 401007ec 0000dc22
[22:05:00]3fffee00:  3fffb868 00000000 40102d1a 00000100
[22:05:00]3fffee10:  00fe880b 7fffffff 00002200 3fffb974
[22:05:00]3fffee20:  00000000 3fffee70 3fffb754 40217889
[22:05:00]3fffee30:  00000000 3fffee70 3fffb754 4020f36a
[22:05:00]3fffee40:  3fffb354 2c9f0300 4000050c 3fffc278
[22:05:00]3fffee50:  40100000 3fffc200 00000022 3fffb354
[22:05:00]3fffee60:  3fffb974 00000000 3fffb754 4020f516
[22:05:00]3fffee70:  3fffbfec 3fff15fc 00000020 3fff0e18
[22:05:00]3fffee80:  3fff0f64 3fff1004 3fffb974 3fff0e18
[22:05:00]3fffee90:  3fffb354 3fff0dfc 3fff0dfd 4022c03c
[22:05:00]3fffeea0:  00000000 3fffdc80 3fff114c 0000025c
[22:05:00]3fffeeb0:  3fff0f64 10b2a8c0 00000020 40100f96
[22:05:00]3fffeec0:  00000000 000015ed 000015ed 3fff0f64
[22:05:00]3fffeed0:  3fffb992 3fff1004 3fff0f64 4023038d
[22:05:00]3fffeee0:  3fffb974 00000014 00000000 40100f5f
[22:05:00]3fffeef0:  4024206c 3fffbfd4 3fff156c 3fff1154
[22:05:00]3fffef00:  00000008 3fff0f64 3fffb974 4022941d
[22:05:00]3fffef10:  3fffdc80 3fff156c 3fffbfd4 40229243
[22:05:00]3fffef20:  4024c522 3fff156c 3fffbfd4 4024c533
[22:05:00]3fffef30:  3fffb984 3fffb974 00000000 3ffe8520
[... 242 lines cut ...]
[22:05:01]3ffffe60:  00000000 00000000 4bc6a7f0 00000000
[22:05:01]3ffffe70:  0000dc0b 3fffb23c 401007ec 0000dc1b
[22:05:01]3ffffe80:  00000000 3fff1fe4 000000ff 402061ad
[22:05:01]3ffffe90:  00fe803a 4bc6a7f0 ae978d4f 00000000
[22:05:01]3ffffea0:  00000000 00000000 4bc6a7f0 00000003
[22:05:01]3ffffeb0:  00000001 3fff08bc 3fff16c0 401004fb
[22:05:01]3ffffec0:  00000000 4bc6a7f0 0c083126 00000000
[22:05:01]3ffffed0:  00000000 00000000 4bc6a7f0 00000000
[22:05:01]3ffffee0:  00000000 4bc6a7f0 401007ec 0000dc1c
[22:05:01]3ffffef0:  401054f9 00f799a1 0872b020 00000000
[22:05:01]3fffff00:  401055a6 3fff0138 00f799a1 00000000
[22:05:01]3fffff10:  4010577d 00f799a1 3fff0b3c 00000000
[22:05:01]3fffff20:  3fff0160 3fff0b3c 00000010 3fff0b3c
[22:05:01]3fffff30:  0000dc1c 3fff1e94 4021c9a4 3fffefa0
[22:05:01]3fffff40:  3fff0b3c 0000dc1c 3fff1e94 4021cffe
[22:05:01]3fffff50:  3fffb248 00000010 3fff07a0 4020a2b8
[22:05:01]3fffff60:  3fffb201 000000a9 feefef00 feefeffe
[22:05:01]3fffff70:  3fffb248 3fff1c34 000000d8 3fff2734
[22:05:01]3fffff80:  00000000 00000000 00000001 3fff0adc
[22:05:01]3fffff90:  3fffdad0 00000000 3fff0a9c 4020d704
[22:05:01]3fffffa0:  3fffdad0 00000000 3fff0a9c 4021cac8
[22:05:01]<<<stack<<<
[22:05:01]
[22:05:01]last failed alloc call: 40215EB9(24)
←[33mWARNING Memory allocation of 24 bytes failed at 40215EB9←[0m

moe01324, 2021-06-16:
any chance that somebody checks this?
can provide additional logs if needed...

DeanCording, 2022-01-08:
I'm experiencing a similar issue with LCD_GPIO

```
substitutions:
  device_name: lcd-display
  board: esp12e
esphome:
  name: lcd-display
  build_path: ./build/lcd-display
  platformio_options: {}
  includes: []
  libraries: []
  name_add_mac_suffix: false
esp8266:
  board: esp12e
  framework:
    version: 2.7.4
    source: ~3.20704.0
    platform_version: platformio/espressif8266 @ 2.6.3
  restore_from_flash: false
  board_flash_mode: dout
wifi:
  ap:
    ssid: lcd-display
    ap_timeout: 1min
  domain: .local
  reboot_timeout: 15min
  power_save_mode: NONE
  fast_connect: false
  output_power: 20.0
  networks:
  - ssid: !secret 'wifi_ssid'
    password: !secret 'wifi_password'
    priority: 0.0
  use_address: lcd-display.local
captive_portal: {}
ota:
  safe_mode: true
  port: 8266
  reboot_timeout: 5min
  num_attempts: 10
time:
- platform: sntp
  servers:
  - ntp.cording.id.au
  id: sntp_time
  timezone: AEST-10
  update_interval: 15min
light:
- platform: status_led
  id: led
  pin:
    number: 2
    inverted: true
    mode:
      output: true
      analog: false
      input: false
      open_drain: false
      pullup: false
      pulldown: false
  disabled_by_default: false
  restore_mode: RESTORE_DEFAULT_OFF
  name: led
  internal: true
text_sensor:
- platform: version
  name: ESPHome Version
  disabled_by_default: false
  icon: mdi:new-box
[... 153 lines cut ...]
[13:00:44]3ffffed0:  3fff0e74 3fff0ce4 00000020 4021f8b8  
WARNING Decoded 0x4021f8b8: AsyncServer::setNoDelay(bool)
[13:00:44]3ffffee0:  3ffef198 3fff0e74 00000000 4020e9c5  
WARNING Decoded 0x4020e9c5: setup
[13:00:44]3ffffef0:  3fff0798 00000000 00000000 3ffef670  
[13:00:44]3fffff00:  00000000 00000000 00000000 00000000  
[13:00:44]3fffff10:  00000000 00000000 00000000 00000000  
[13:00:44]3fffff20:  3fff0728 00000000 00000000 3fff0740  
[13:00:44]3fffff30:  00000000 00000000 00000000 00000000  
[13:00:44]3fffff40:  00000000 00000000 00000000 00000000  
[13:00:44]3fffff50:  3fff0ce4 00000000 00000000 3fff0c14  
[13:00:44]3fffff60:  3fff0b54 3fff0afc 3fff0a0c 3fff08c4  
[13:00:44]3fffff70:  3fff0db0 feefeffe feefeffe feefeffe  
[13:00:44]3fffff80:  3ffef194 00000010 3ffef18c 3ffef190  
[13:00:44]3fffff90:  feefeffe feefeffe feefeffe 3ffef474  
[13:00:44]3fffffa0:  3fffdad0 00000000 3ffef434 4021c0b8  
WARNING Decoded 0x4021c0b8: loop_wrapper() at core_esp8266_main.cpp
[13:00:44]3fffffb0:  feefeffe feefeffe 3ffe8540 401006bd  
WARNING Decoded 0x401006bd: cont_wrapper
[13:00:44]<<<stack<<<
[13:00:44]
[13:00:44]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[13:00:44]

```

DeanCording, 2022-01-08:
VERY_VERBOSE log:

```
[16:48:32]
[16:48:32]SDK:2.2.2-dev(38a443e)/Core:2.7.3-3-g2843a5ac=20703003/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-30-g92add50/BearSSL:5c771be
[16:48:32][I][logger:214]: Log initialized
[16:48:32][I][app:029]: Running through setup()...
[16:48:32][V][app:030]: Sorting components by setup priority...
[16:48:32][VV][scheduler:057]: set_interval(name='', interval=60000, offset=29983)
[16:48:32][C][lcd_gpio:010]: Setting up GPIO LCD Display...
[16:48:32][VV][scheduler:057]: set_interval(name='update', interval=1000, offset=259)
[16:48:32][C][wifi:037]: Setting up WiFi...
[16:48:32][V][wifi_esp8266:060]: Disabling STA.
[16:48:32]del if0
[16:48:32]usl
[16:48:32]mode : null
[16:48:32][V][wifi_esp8266:058]: Enabling STA.
[16:48:32]mod[V][wifi_esp8266:535]: Event: Changed Mode old=STA new=OFF
[16:48:32]wifi evt: 8
[16:48:32]e : sta(ec:fa:bc:0b:f7:8d)
[16:48:32]add if0
[16:48:32][V][wifi_esp8266:535]: Event: Changed Mode old=OFF new=STA
[16:48:32]wifi evt: 8
[16:48:32]sleep disable
[16:48:32][D][wifi:370]: Starting scan...
[16:48:32][VV][scheduler:185]: Running interval '' with interval=60000 last_execution=4294877415 (now=165)
[16:48:32][VV][scheduler:185]: Running interval 'update' with interval=1000 last_execution=4294966167 (now=165)
[16:48:32]
[16:48:32]User exception (panic/abort/assert)
[16:48:32]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[16:48:32]
[16:48:32]Abort called
[16:48:32]
[16:48:32]>>>stack>>>
WARNING Found stack trace! Trying to decode it
[16:48:32]
[16:48:32]ctx: cont
[16:48:32]sp: 3ffffdc0 end: 3fffffc0 offset: 0000
[16:48:32]3ffffdc0:  00000007 00000000 3ffef80c 40202124  
WARNING Decoded 0x40202124: esphome::logger::Logger::log_vprintf_(int, char const*, int, __FlashStringHelper const*, __va_list_tag)
[16:48:32]3ffffdd0:  000000fe 00000000 00000000 00000000  
[16:48:32]3ffffde0:  00000000 00000000 00000000 00000002  
[16:48:32]3ffffdf0:  00000020 0000003c 3ffee70c 3fff002c  
[16:48:32]3ffffe00:  3fff007c 000000a5 3ffefe14 4020f21e  
WARNING Decoded 0x4020f21e: raise_exception at core_esp8266_postmortem.cpp
[16:48:32]3ffffe10:  00000000 00000000 4bc6a7f0 4020f230  
WARNING Decoded 0x4020f230: __assert_func
[16:48:32]3ffffe20:  3ffffe50 3ffffe40 401007e6 4021969c  
WARNING Decoded 0x401007e6: millis
WARNING Decoded 0x4021969c: std::__detail::_List_node_base::_M_hook(std::__detail::_List_node_base*) at /workdir/repo/gcc/libstdc++-v3/src/c++98/list.cc:131
[16:48:32]3ffffe30:  00000000 3ffffe40 00000010 402019c0  
WARNING Decoded 0x402019c0: esphome::lcd_gpio::GPIOLCDDisplay::call_writer()
[16:48:32]3ffffe40:  3fff007c 3ffefe14 3ffffe80 402019a1  
WARNING Decoded 0x402019a1: esphome::lcd_base::LCDDisplay::update()
[16:48:32]3ffffe50:  3ffe88d4 3ffefff8 3ffee70c 40210ab4  
WARNING Decoded 0x40210ab4: std::_Function_handler<void (), esphome::PollingComponent::call_setup()::{lambda()#1}>::_M_invoke(std::_Any_data const&) at component.cpp
[16:48:32]3ffffe60:  3fff007c 000000a5 3ffee70c 401003f1  
WARNING Decoded 0x401003f1: esphome::Scheduler::call()
[16:48:32]3ffffe70:  000003e8 fffffb97 000000a5 3ffeff0c  
[16:48:32]3ffffe80:  000000b7 3ffefe14 3ffee70c 00000000  
[16:48:32]3ffffe90:  3fff0084 3fff007c 00000001 fffea0e7  
[16:48:32]3ffffea0:  3fff00f4 00000004 3ffee70c 00000100  
[16:48:32]3ffffeb0:  3ffefbec 00000003 3ffee70c 4020719a  
WARNING Decoded 0x4020719a: esphome::Application::setup()
[16:48:32]3ffffec0:  3fff00f4 00000004 3ffefedc 402064af  
WARNING Decoded 0x402064af: std::vector<esphome::Component*, std::allocator<esphome::Component*> >::push_back(esphome::Component* const&)
[16:48:32]3ffffed0:  00000100 3fff00f4 3ffeff1c 4020651c  
WARNING Decoded 0x4020651c: esphome::Application::register_component_(esphome::Component*)
[16:48:32]3ffffee0:  3ffefedc 3ffee70c 3ffefbec 3ffee794  
[16:48:32]3ffffef0:  3ffefedc 3ffee70c 00000000 40208304  
WARNING Decoded 0x40208304: setup
[16:48:32]3fffff00:  3ffefd98 00000000 00000000 3ffeebc4  
[16:48:32]3fffff10:  00000000 00000000 00000000 00000000  
[16:48:32]3fffff20:  00000000 00000000 00000000 00000000  
[16:48:32]3fffff30:  3ffefd08 00000000 00000000 3ffefd30  
[16:48:32]3fffff40:  00000000 00000000 00000000 00000000  
[16:48:32]3fffff50:  00000000 00000000 00000000 00000000  
[16:48:32]3fffff60:  3ffefd98 feefeffe feefeffe feefeffe  
[16:48:32]3fffff70:  3ffee788 3ffee78c 3ffee790 00000010  
[16:48:32]3fffff80:  00000020 feefeffe feefeffe feefeffe  
[16:48:32]3fffff90:  feefeffe feefeffe feefeffe 3ffeea00  
[16:48:32]3fffffa0:  3fffdad0 00000000 3ffee9c0 4020edd0  
WARNING Decoded 0x4020edd0: loop_wrapper() at core_esp8266_main.cpp
[16:48:32]3fffffb0:  feefeffe feefeffe 3ffe84f0 4010065d  
WARNING Decoded 0x4010065d: cont_wrapper
[16:48:32]<<<stack<<<
[16:48:32]
[16:48:32]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[16:48:32]
```

I've tried many different versions of the Arduino_ESP and Expressif libraries with no success.

grob6000, 2023-05-10:
Not to revive a dead issue, but I had a similar issue with the I2C SH1306. Changing i2c **frequency** to ``800kHz`` fixed the issue.

```
i2c:
  sda: 4
  scl: 5
  frequency: 800kHz
```

It appears the display update routine blocks the wifi, and this results in disconnection / dropout.

To address other displays, it might be worth considering ways to increase the speed of the update routine.

Others have had success increasing **update_frequency** to ``2s`` or more.