esphome/issues #1590: Mcp23017 (i2c) stops working after a few minutes  with esp32 devkitv1

Primo83, 2020-10-30:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP32 devkitv1

**ESPHome version (latest production, beta, dev branch)**
ESPHome version 1.15.2

**Affected component:**
I2C Bus (MPC23017)
https://esphome.io/components/i2c.html#i2c
https://esphome.io/components/mcp230xx.html?highlight=mcp23017#mcp23017-label

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
i2c:
  sda: 21
  scl: 22
  scan: True
  
mcp23017:
  - id: 'mcp23017_hub'
    address: 0x20

switch:
  - platform: gpio
    name: "MCP23017 Pin #7"
    pin:
      mcp23017: mcp23017_hub
      # Use pin number 0
      number: 7
      mode: OUTPUT

```

**Logs (if applicable):**

first e few minutes log looks like: 
```
[15:08:23][C][wifi:307]:   Hostname: 'esp32_test'
[15:08:23][C][wifi:311]:   Signal strength: -73 dB ▂▄▆█
[15:08:23][V][wifi:313]:   Priority: 0.0
[15:08:23][C][wifi:315]:   Channel: 6
[15:08:23][C][wifi:316]:   Subnet: 255.255.255.0
[15:08:23][C][wifi:317]:   Gateway: 192.168.0.1
[15:08:23][C][wifi:318]:   DNS1: 192.168.0.1
[15:08:23][C][wifi:319]:   DNS2: 0.0.0.0
[15:08:23][C][i2c:028]: I2C Bus:
[15:08:23][C][i2c:029]:   SDA Pin: GPIO21
[15:08:23][C][i2c:030]:   SCL Pin: GPIO22
[15:08:23][C][i2c:031]:   Frequency: 50000 Hz
[15:08:23][I][i2c:033]: Scanning i2c bus for active devices...
[15:08:23][I][i2c:040]: Found i2c device at address 0x20
[15:08:23][C][switch.gpio:042]: GPIO Switch 'MCP23017 Pin #7'
[15:08:23][C][switch.gpio:043]:   Pin: GPIO7 (Mode: OUTPUT)
[15:08:23][C][switch.gpio:059]:   Restore Mode: Restore (Defaults to OFF)
[15:08:23][C][logger:185]: Logger:
[15:08:23][C][logger:186]:   Level: VERY_VERBOSE
[15:08:23][C][logger:187]:   Log Baud Rate: 115200
[15:08:23][C][logger:188]:   Hardware UART: UART0
[15:08:23][C][captive_portal:169]: Captive Portal:
[15:08:23][C][ota:029]: Over-The-Air Updates:
[15:08:23][C][ota:030]:   Address: esp32_test.local:3232
[15:08:23][W][ota:036]: Last Boot was an unhandled reset, will proceed to safe mode in 8 restarts
[15:08:23][C][api:095]: API Server:
[15:08:23][C][api:096]:   Address: esp32_test.local:6053
[15:08:28][VV][api.service:220]: on_ping_request: PingRequest {}
[15:08:28][VV][api.service:032]: send_ping_response: PingResponse {}
[15:08:33][VV][api.service:220]: on_ping_request: PingRequest {}
[15:08:33][VV][api.service:032]: send_ping_response: PingResponse {}
[15:08:35][VV][api.service:220]: on_ping_request: PingRequest {}
[15:08:35][VV][api.service:032]: send_ping_response: PingResponse {}
[15:08:38][VV][api.service:290]: on_switch_command_request: SwitchCommandRequest {
  key: 1326097480
  state: YES
}
[15:08:38][D][switch:021]: 'MCP23017 Pin #7' Turning ON.
[15:08:38][VV][i2c:056]: Beginning Transmission to 0x20:
[15:08:38][VV][i2c:093]:     Writing 0b00010100 (0x14)
[15:08:38][VV][i2c:093]:     Writing 0b10000000 (0x80)
[15:08:38][VV][i2c:061]:     Transmission ended. Status code: 0x00
[15:08:38][VV][preferences:051]: SAVE 1: 0=0x00000001 1=0x00111894 (Type=1326097480, CRC=0x00111894)
[15:08:38][D][switch:045]: 'MCP23017 Pin #7': Sending state ON
[15:08:38][VV][api.service:117]: send_switch_state_response: SwitchStateResponse {
  key: 1326097480
  state: YES
}
[15:08:38][VV][api.service:220]: on_ping_request: PingRequest {}
[15:08:38][VV][api.service:032]: send_ping_response: PingResponse {}
[15:08:41][VV][api.service:290]: on_switch_command_request: SwitchCommandRequest {
  key: 1326097480
  state: NO
}
[15:08:41][D][switch:025]: 'MCP23017 Pin #7' Turning OFF.
[15:08:41][VV][i2c:056]: Beginning Transmission to 0x20:
[15:08:41][VV][i2c:093]:     Writing 0b00010100 (0x14)
[15:08:41][VV][i2c:093]:     Writing 0b00000000 (0x00)
[15:08:41][VV][i2c:061]:     Transmission ended. Status code: 0x00
[15:08:41][VV][preferences:051]: SAVE 1: 0=0x00000000 1=0x4F0AA448 (Type=1326097480, CRC=0x4F0AA448)
[15:08:41][D][switch:045]: 'MCP23017 Pin #7': Sending state OFF
[15:08:41][VV][api.service:117]: send_switch_state_response: SwitchStateResponse {
  key: 1326097480
  state: NO
}
``
after a few minutes log looks like: 
```
[14:58:04][V][wifi:313]:   Priority: 0.0
[14:58:04][C][wifi:315]:   Channel: 6
[14:58:04][C][wifi:316]:   Subnet: 255.255.255.0
[14:58:04][C][wifi:317]:   Gateway: 192.168.0.1
[14:58:04][C][wifi:318]:   DNS1: 192.168.0.1
[14:58:04][C][wifi:319]:   DNS2: 0.0.0.0
[14:58:04][C][i2c:028]: I2C Bus:
[14:58:04][C][i2c:029]:   SDA Pin: GPIO21
[14:58:04][C][i2c:030]:   SCL Pin: GPIO22
[14:58:04][C][i2c:031]:   Frequency: 50000 Hz
[14:58:04][I][i2c:033]: Scanning i2c bus for active devices...
[14:58:05][I][i2c:049]: Found no i2c devices!
[14:58:05][C][switch.gpio:042]: GPIO Switch 'MCP23017 Pin #7'
[14:58:05][C][switch.gpio:043]:   Pin: GPIO7 (Mode: OUTPUT)
[14:58:05][C][switch.gpio:059]:   Restore Mode: Restore (Defaults to OFF)
[14:58:05][C][logger:185]: Logger:
[14:58:05][C][logger:186]:   Level: VERY_VERBOSE
[14:58:05][C][logger:187]:   Log Baud Rate: 115200
[14:58:05][C][logger:188]:   Hardware UART: UART0
[14:58:05][C][captive_portal:169]: Captive Portal:
[14:58:05][C][ota:029]: Over-The-Air Updates:
[14:58:05][C][ota:030]:   Address: esp32_test.local:3232
[14:58:05][C][api:095]: API Server:
[14:58:05][C][api:096]:   Address: esp32_test.local:6053

[15:04:23][D][switch:025]: 'MCP23017 Pin #7' Turning OFF.
[15:04:23][VV][i2c:056]: Beginning Transmission to 0x20:
[15:04:23][VV][i2c:093]:     Writing 0b00010100 (0x14)
[15:04:23][VV][i2c:093]:     Writing 0b00000000 (0x00)
[15:04:23][VV][i2c:061]:     Transmission ended. Status code: 0x02
[15:04:23][W][i2c:070]: Received NACK on transmit of address 0x20
[15:04:23][VV][preferences:051]: SAVE 1: 0=0x00000000 1=0x4F0AA448 (Type=1326097480, CRC=0x4F0AA448)
[15:04:23][D][switch:045]: 'MCP23017 Pin #7': Sending state OFF
[15:04:23][VV][api.service:117]: send_switch_state_response: SwitchStateResponse {
  key: 1326097480
  state: NO
}
[15:04:23][VV][api.service:220]: on_ping_request: PingRequest {}
[15:04:23][VV][api.service:032]: send_ping_response: PingResponse {}
[15:04:28][VV][api.service:290]: on_switch_command_request: SwitchCommandRequest {
  key: 1326097480
  state: YES
}
[15:04:28][D][switch:021]: 'MCP23017 Pin #7' Turning ON.
[15:04:28][VV][i2c:056]: Beginning Transmission to 0x20:
[15:04:28][VV][i2c:093]:     Writing 0b00010100 (0x14)
[15:04:28][VV][i2c:093]:     Writing 0b10000000 (0x80)
[15:04:28][VV][i2c:061]:     Transmission ended. Status code: 0x02
[15:04:28][W][i2c:070]: Received NACK on transmit of address 0x20
[15:04:28][VV][preferences:051]: SAVE 1: 0=0x00000001 1=0x00111894 (Type=1326097480, CRC=0x00111894)
[15:04:28][D][switch:045]: 'MCP23017 Pin #7': Sending state ON
[15:04:28][VV][api.service:117]: send_switch_state_response: SwitchStateResponse {
  key: 1326097480
  state: YES
}
```

**Additional information and things you've tried:**
I connected everything in different ways and the result is always the same. The system runs for about 3-10 minutes and suddenly stops. I have no idea what I could do more. Does anyone know the solution? 

<!-- LEAVE THIS LINE AS-IS AND DON'T DELETE IT, OTHERWISE THE ISSUE WILL BE CLOSED AUTOMATICALLY. -->

kakaki, 2020-10-31:
Have same problem (nack error) after updating to 15.2, was working fine on 2 devices esp8285 m3

randybb, 2020-10-31:
Interesting. I am having similar issue with my heating. I had some power issues and during investigation I completely rebuilded power delivery to remote relay boards and replaced PCF8574 with MCP23017. Since then there are no power issues anymore, only MCP23017 stops responding after around 12 hours. Restart doesn't help and to fix that I need to disconnect it from power and connect it again. I thought there must be something in my setup, because I use MCP23017 in my DEV environment without any issues.

kakaki, 2020-10-31:
A have updated today to 1.15.3 and it’s working stable for over 5 hours.

randybb, 2020-11-01:
I am on DEV branch, so I pulled the last docker image yesterday and flashed it with a fresh binary. After one day it is still working, lets wait a couple more days to confirm functionality.

andriej, 2021-01-21:
I've switched my light controls from Arduino to ESP32 with 4 MCP23017 modules (for input and output) and since 2-3 weeks I've had 2 times that I needed power-reboot whole esp32 (reset button didn't help).

1) ESPHome seemed to 'invert' GPIO states, so those which were low became high, but were still working 
2) INPUT pins stopped reporting their state, output worked seamlessly on same chip

Do these issues connect with this thread anyhow?

closed 2021-06-05 by stale[bot]