esphome/issues #1591: E-ink Waveshare 2.13 not working

melwinek, 2020-10-31:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

Wemos D1 mini

**ESPHome version (latest production, beta, dev branch)**

Current stable

**Affected component:**

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: salonpanel
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: "lipowo"
  password: "xxxx"
  power_save_mode: none
  ap:
    ssid: "oh_salonpanel"
    password: "xxxx"

captive_portal:

logger:

api:
  password: "xxxx"

ota:
  password: "xxxx"

web_server:
  port: 80
  auth:
    username: admin
    password: "xxxx" 

font:
  - file: "arial.ttf"
    id: my_font
    size: 20

spi:
  clk_pin: D5
  mosi_pin: D7

display:
  - platform: waveshare_epaper
    cs_pin: D8
    dc_pin: D3
    busy_pin: D2
    reset_pin: D4
    update_interval: 30s 
    model: 2.13in
    full_update_every: 30
    lambda: |-
      it.print(0, 10, id(my_font), "Hello World!");

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
INFO Reading configuration /config/esphome/salonpanel.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/salonpanel
Processing salonpanel (board: d1_mini; framework: arduino; platform: espressif8266@2.6.2)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.20704.0 (2.7.4) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20800.0 (2.8.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.3
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.7
|   |-- <ESPAsyncTCP-esphome> 1.2.3
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|   |-- <ArduinoJson-esphomelib> 5.13.3
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <DNSServer> 1.1.1
|   |-- <ESP8266WiFi> 1.0
|-- <SPI> 1.0
Compiling /data/salonpanel/.pioenvs/salonpanel/src/main.cpp.o
Linking /data/salonpanel/.pioenvs/salonpanel/firmware.elf
Building /data/salonpanel/.pioenvs/salonpanel/firmware.bin
Retrieving maximum program size /data/salonpanel/.pioenvs/salonpanel/firmware.elf
Checking size /data/salonpanel/.pioenvs/salonpanel/firmware.elf
RAM:   [=====     ]  45.8% (used 37516 bytes from 81920 bytes)
Flash: [====      ]  39.6% (used 413484 bytes from 1044464 bytes)
Creating BIN file "/data/salonpanel/.pioenvs/salonpanel/firmware.bin" using "/root/.platformio/packages/framework-arduinoespressif8266/bootloaders/eboot/eboot.elf" and "/data/salonpanel/.pioenvs/salonpanel/firmware.elf"
========================= [SUCCESS] Took 24.39 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of salonpanel.local
INFO  -> 192.168.3.243
INFO Uploading /data/salonpanel/.pioenvs/salonpanel/firmware.bin (417632 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from salonpanel.local using esphome API
INFO Connecting to salonpanel.local:6053 (192.168.3.243)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to 192.168.3.243: [Errno 111] Connection refused). Re-Trying in 1 seconds
INFO Connecting to salonpanel.local:6053 (192.168.3.243)
INFO Successfully connected to salonpanel.local
[16:42:36][I][app:105]: ESPHome version 1.15.3 compiled on Oct 31 2020, 16:41:41
[16:42:36][C][wifi:443]: WiFi:
[16:42:36][C][wifi:303]:   SSID: [redacted]
[16:42:36][C][wifi:304]:   IP Address: 192.168.3.243
[16:42:36][C][wifi:306]:   BSSID: [redacted]
[16:42:36][C][wifi:307]:   Hostname: 'salonpanel'
[16:42:36][C][wifi:311]:   Signal strength: -85 dB ▂▄▆█
[16:42:36][C][wifi:315]:   Channel: 1
[16:42:36][C][wifi:316]:   Subnet: 255.255.255.0
[16:42:36][C][wifi:317]:   Gateway: 192.168.3.49
[16:42:36][C][wifi:318]:   DNS1: 8.8.8.8
[16:42:36][C][wifi:319]:   DNS2: (IP unset)
[16:42:37][E][waveshare_epaper:107]: Timeout while displaying image!
[16:42:37][C][spi:081]: SPI bus:
[16:42:37][C][spi:082]:   CLK Pin: GPIO14 (Mode: OUTPUT)
[16:42:37][C][spi:084]:   MOSI Pin: GPIO13 (Mode: OUTPUT)
[16:42:37][C][spi:085]:   Using HW SPI: NO
[16:42:38][E][waveshare_epaper:107]: Timeout while displaying image!
[16:42:38][C][logger:185]: Logger:
[16:42:38][C][logger:186]:   Level: DEBUG
[16:42:38][C][logger:187]:   Log Baud Rate: 115200
[16:42:38][C][logger:188]:   Hardware UART: UART0
[16:42:39][E][waveshare_epaper:107]: Timeout while displaying image!
[16:42:39][C][waveshare_epaper:183]: Waveshare E-Paper
[16:42:39][C][waveshare_epaper:183]:   Rotations: 0 °
[16:42:39][C][waveshare_epaper:183]:   Dimensions: 128px x 250px
[16:42:39][C][waveshare_epaper:189]:   Model: 2.13in
[16:42:39][C][waveshare_epaper:201]:   Full Update Every: 30
[16:42:39][C][waveshare_epaper:202]:   Reset Pin: GPIO2 (Mode: OUTPUT)
[16:42:39][C][waveshare_epaper:203]:   DC Pin: GPIO0 (Mode: OUTPUT)
[16:42:39][C][waveshare_epaper:204]:   Busy Pin: GPIO4 (Mode: INPUT)
[16:42:39][C][waveshare_epaper:205]:   Update Interval: 1.0s
[16:42:40][E][waveshare_epaper:107]: Timeout while displaying image!
[16:42:41][E][waveshare_epaper:107]: Timeout while displaying image!
[16:42:42][C][captive_portal:169]: Captive Portal:
[16:42:43][E][waveshare_epaper:107]: Timeout while displaying image!
[16:42:43][C][web_server:131]: Web Server:
[16:42:43][C][web_server:132]:   Address: salonpanel.local:80
[16:42:43][C][web_server:134]:   Basic authentication enabled
[16:42:44][E][waveshare_epaper:107]: Timeout while displaying image!
[16:42:44][C][ota:029]: Over-The-Air Updates:
[16:42:44][C][ota:030]:   Address: salonpanel.local:8266
[16:42:44][C][ota:032]:   Using Password.
[16:42:45][E][waveshare_epaper:107]: Timeout while displaying image!
[16:42:45][C][api:095]: API Server:
[16:42:45][C][api:096]:   Address: salonpanel.local:6053
[16:42:46][E][waveshare_epaper:107]: Timeout while displaying image!
[16:42:47][E][waveshare_epaper:107]: Timeout while displaying image!
[16:42:48][E][waveshare_epaper:107]: Timeout while displaying image!
[16:42:49][E][waveshare_epaper:107]: Timeout while displaying image!
```

**Additional information and things you've tried:**

chris-nite, 2020-11-17:
Hello @melwinek , did you already try one of those two alternative display-models in your config?
- 2.13in-ttgo
- 2.13in-ttgo-b73
It might be the case, that you got in posession of one of the newer panels from waveshare, which needs a different display-model-configuration (display --> model: <insert one of the above here>).
Good luck and please let us know if it worked! :)

brewston, 2021-03-06:
@melwinek did you ever get this working ? I have a 2.13 display (SKU 12672) connected via a e-Paper ESP8266 Driver Board (I bought them together and there's a ribbon cable between the board and the display. I have similar yaml to yourself and also see 'Timeout while displaying image' in the logs

yaml looks like : 

```
spi:
  clk_pin: D5
  mosi_pin: D7

font:

  - file: 'materialdesignicons-webfont.ttf'
    id: icon_font_30
    size: 30

display:
  - platform: waveshare_epaper
    cs_pin: D8    # GPIO 15
    dc_pin: D2    # GPIO 4
    busy_pin: D0  # GPIO 16
    reset_pin: D1 # GPIO 5
    model: 2.13in
    full_update_every: 6000
    update_interval: 30s

    lambda: |-
       it.print(0, 0, id(icon_font_30), "Hello World!");
```
The manual for the board is here :  https://www.waveshare.com/w/upload/a/a7/E-Paper_ESP8266_Driver_Board_EN.pdf
I needed it to deduce the correct pin assignments. It also has a paper type switch but I've tried in both A & B and still get the same error :(

closed 2021-07-18 by stale[bot]

swifty99, 2021-08-19:
Hi, 
on ESPHome 2021.8.0 with the same config on a ESP32 I have the same issue. The text gets displayed, however updated every second (the display kind of blinks) with error messages at random times.
Has anyone solved this?

swifty