esphome/issues #1592: ESP8266 output change causes PCF8574 output changes

evolucja, 2020-10-31:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**
Wemos D1 mini + PCF8574

**ESPHome version (latest production, beta, dev branch)**
1.13.6

**Affected component:**
https://esphome.io/components/pcf8574.html

**Description of problem:**
Today I connected PCF8574 to have additional output GPIO.  At first I thought that everything is working fine but it started to behave strangly:

I have..
livingroom_lamp_1 = ON
livingroom_lamp_2 = ON

then I change livingroom_shutter_down to ON for 3 seconds and then to OFF and in this moment
livingroom_lamp_1, livingroom_lamp_2 go OFF but only in reality (buttons in HA remain ON)

then I click livingroom_lamp_1 to OFF in HA and livingroom_lamp_2 goes ON in reality

then I click livingroom_lamp_1 to ON in HA and livingroom_lamp_1 goes ON in reality

then I click livingroom_shutter_up to ON and livingroom_lamp_1, livingroom_lamp_2 go OFF

Basically changing ESP8266 output causes PCF8574 to change its state but just in reality.

**Problem-relevant YAML-configuration entries:**
```yaml

esphome:
  name: espek_03
  platform: ESP8266
  board: d1_mini_lite

wifi:
  ssid: "xxx"
  password: "xxx"
  manual_ip:
    static_ip: 10.0.0.23
    gateway: 10.0.0.1
    subnet: 255.255.255.0

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

switch:
  - platform: gpio
    pin: 
      #number: A6 #D1
      #inverted: yes
      
      pcf8574: pcf8574_hub
      number: 6
      mode: OUTPUT
      inverted: true
      
      
    name: livingroom_lamp_1
  - platform: gpio
    pin:  
      #number: A7 #D2
[... 198 lines cut ...]
[18:21:10][C][gpio.binary_sensor:016]:   Pin: GPIO2 (Mode: INPUT_PULLUP)
[18:21:10][C][esp8266_pwm:022]: ESP8266 PWM:
[18:21:10][C][esp8266_pwm:023]:   Pin: GPIO3 (Mode: OUTPUT)
[18:21:10][C][esp8266_pwm:024]:   Frequency: 1000.0 Hz
[18:21:10][C][logger:137]: Logger:
[18:21:10][C][logger:138]:   Level: DEBUG
[18:21:10][C][logger:139]:   Log Baud Rate: 115200
[18:21:10][C][logger:140]:   Hardware UART: UART0
[18:21:10][C][light:174]: Light 'hall_led'
[18:21:10][C][light:176]:   Default Transition Length: 1.0s
[18:21:10][C][light:177]:   Gamma Correct: 2.80
[18:21:10][C][restart:021]: Restart Switch 'espek_03 restart'
[18:21:10][C][restart:021]:   Icon: 'mdi:restart'
[18:21:10][C][time_based.cover:012]: Time Based Cover 'livingroom_shutter'
[18:21:10][C][time_based.cover:012]:   Assumed State: YES
[18:21:10][C][time_based.cover:013]:   Open Duration: 18.3s
[18:21:10][C][time_based.cover:014]:   Close Duration: 18.3s
[18:21:10][C][ota:029]: Over-The-Air Updates:
[18:21:10][C][ota:030]:   Address: 10.0.0.23:8266
[18:21:10][C][api:103]: API Server:
[18:21:10][C][api:104]:   Address: 10.0.0.23:6053

```

**Additional information and things you've tried:**

evolucja, 2021-03-13:
Does anyone have any idea? It's really strange to me and I don't know what can be wrong..

closed 2021-07-25 by stale[bot]

sweetpants, 2021-08-04:
I have a similar thing going on using relay boards (optically isolated). p0 = on (low), when i switch p1 = on, no problem, but when I switch p1 = off (high), sometimes p0 goes off too, sometimes other ports also. It looks like the PCF8574 is latching the wrong value for some reason. I suspect something external as the output from ESPHOME looks ok.