esphome/issues #1593: IR Transmitter Pioneer Update

GhettoDubs, 2020-11-01:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HassOS 4.15
Supervisor 2020.10.0
Pi 4 2gb

**ESP (ESP32/ESP8266, Board/Sonoff):**

Mirabella genio flashed to esphome
https://esphome-configs.io/devices/mirabella-genio-smart-universal-ir-controller

**ESPHome version (latest production, beta, dev branch)**

ESPhome version 1.15.3 

**Affected component:**

IR transmitter

**Description of problem:**

I'm trying to use IR transmitter to turn on Pioneer receiver model VSX 523 but no luck.  Transmitter works for other components.

I've decoded the signal from the original remote and get outputs as per logs below.  I also have IR blaster on my phone and using Anymote I can turn on the receiver and get exactly the same output.  I have another receiver on ESPhome and when I send this code I get same code back.

I've tried various combinations with repeat but I don't think that's the problem.  I also tried to send raw but did not work.  I did try to adjust the spacing as per some of the discussion in pull request below.

Strangely if i keep basking the send button in HASS it sometimes works.

I believe it may be related to the timings in the code.  The current build does not match up to latest version of IRremoteESP8266 and pull request was raised for similar problem.
https://github.com/arendst/Tasmota/issues/6100
https://github.com/crankyoldgit/IRremoteESP8266/pull/830

**Problem-relevant YAML-configuration entries:**
```yaml
remote_transmitter:
  pin: GPIO14
  carrier_duty_percent: 33%
  
remote_receiver:
  pin:
    number: GPIO5
    inverted: True
  dump: all

switch:
  - platform: restart
    name: "ESPHome_ir1 restart"
  - platform: template
    name: "IR Receiver power nec"
    turn_on_action:
      - remote_transmitter.transmit_nec:
          address: 0xA55A
          command: 0x38C7
          repeat:
            times: 4
            wait_time: 100ms
  - platform: template
    name: "IR Receiver power pioneer"
    turn_on_action:
      - remote_transmitter.transmit_pioneer:
          rc_code_1: 0xA51C
          repeat:
            times: 4
            wait_time: 100ms
  - platform: template
    name: "IR Receiver power raw"
    turn_on_action:
      - remote_transmitter.transmit_raw:
          code: [8450, -4225, 500, -1500, 522, -555, 519, -1599, 525, -549, 522, -551, 522, -1600, 521, -552, 521, -1601, 522, -551, 522, -1600, 521, -552, 521, -1601, 522, -1599, 522, -552, 521, -1600, 522, -552, 521, -553, 520, -553, 521, -1600, 522, -1600, 521, -1601, 521, -552, 521, -553, 516, -557, 521, -1600, 522, -1600, 521, -553, 520, -553, 521, -553, 497, -1624, 521, -1601, 520, -1601, 498, -25000,8437, -4269, 521, -1601, 521, -556, 517, -1601, 514, -560, 497, -576, 521, -1601, 519, -554, 520, -1602, 521, -552, 497, -1625, 520, -553, 520, -1602, 521, -1600, 520, -554, 497, -1624, 498, -576, 497, -576, 497, -577, 497, -1624, 521, -1601, 497, -1624, 498, -576, 497, -577, 496, -577, 497, -1624, 498, -1624, 511, -563, 497, -576, 497, -577, 496, -1625, 499, -1623, 497, -1624, 497]
          carrier_frequency: 40kHz
```

**Logs (if applicable):**

Receiver output when using original remote
```
[16:37:05][D][remote.jvc:048]: Received JVC: data=0xA55A
[16:37:05][D][remote.lg:053]: Received LG: data=0xA55A38C7, nbits=32
[16:37:05][D][remote.nec:068]: Received NEC: address=0xA55A, command=0x38C7
[16:37:05][D][remote.pioneer:144]: Received Pioneer: rc_code_X=0xA51C
[16:37:05][D][remote.jvc:048]: Received JVC: data=0xA55A
[16:37:05][D][remote.lg:053]: Received LG: data=0xA55A38C7, nbits=32
[16:37:05][D][remote.nec:068]: Received NEC: address=0xA55A, command=0x38C7
[16:37:05][D][remote.pioneer:144]: Received Pioneer: rc_code_X=0xA51C
[16:37:05][D][remote.jvc:048]: Received JVC: data=0xA55A
[16:37:05][D][remote.lg:053]: Received LG: data=0xA55A38C7, nbits=32
[16:37:05][D][remote.nec:068]: Received NEC: address=0xA55A, command=0x38C7
[16:37:05][D][remote.pioneer:144]: Received Pioneer: rc_code_X=0xA51C
[16:37:05][D][remote.jvc:048]: Received JVC: data=0xA55A
[16:37:05][D][remote.lg:053]: Received LG: data=0xA55A38C7, nbits=32
[16:37:05][D][remote.nec:068]: Received NEC: address=0xA55A, command=0x38C7
[16:37:05][D][remote.pioneer:144]: Received Pioneer: rc_code_X=0xA51C
```

Receiver output raw when using original remote
```
[16:39:15][D][remote.raw:028]: Received Raw: 8463, -4245, 522, -1600, 522, -552, 521, -1600, 522, -552, 521, -552, 522, -1600, 521, -552, 522, -1600, 522, -552, 521, -1600, 522, -552, 521, -1600, 522, -1600, 521, -552, 521, -1601, 521, -552, 522, -552, 521, -552, 521, -1601, 521, 
[16:39:15][D][remote.raw:041]:   -1600, 522, -1600, 521, -553, 520, -553, 521, -553, 520, -1601, 520, -1602, 514, -559, 521, -553, 497, -576, 520, -1602, 497, -1624, 520, -1602, 497
[16:39:15][D][remote.raw:028]: Received Raw: 8437, -4270, 521, -1600, 521, -553, 520, -1602, 520, -553, 519, -555, 518, -1603, 497, -577, 497, -1624, 497, -577, 497, -1624, 520, -554, 497, -1624, 498, -1624, 497, -577, 496, -1625, 497, -577, 496, -577, 497, -577, 496, -1625, 497, 
[16:39:15][D][remote.raw:041]:   -1625, 497, -1624, 497, -577, 497, -577, 496, -577, 496, -1626, 496, -1625, 497, -577, 496, -577, 496, -578, 496, -1625, 497, -1625, 496, -1625, 497
[16:39:16][D][remote.raw:028]: Received Raw: 8437, -4270, 497, -1625, 497, -577, 496, -1625, 497, -577, 497, -579, 494, -1625, 518, -555, 497, -1625, 519, -554, 500, -1622, 497, -577, 496, -1625, 497, -1625, 497, -576, 497, -1625, 520, -554, 496, -577, 496, -577, 497, -1625, 497, 
[16:39:16][D][remote.raw:041]:   -1624, 520, -1602, 497, -577, 496, -577, 497, -577, 496, -1625, 519, -1603, 497, -576, 497, -577, 496, -577, 497, -1625, 496, -1625, 497, -1625, 497
[16:39:16][D][remote.raw:028]: Received Raw: 8436, -4270, 521, -1601, 520, -554, 496, -1625, 497, -576, 497, -577, 497, -1624, 520, -554, 520, -1601, 521, -553, 497, -1624, 521, -553, 497, -1624, 498, -1624, 521, -553, 520, -1601, 520, -554, 497, -576, 497, -577, 497, -1624, 498, 
[16:39:16][D][remote.raw:041]:   -1624, 497, -1625, 520, -553, 497, -577, 497, -576, 497, -1625, 518, -1603, 497, -577, 497, -576, 497, -577, 496, -1625, 497, -1625, 497, -1625, 497
```

Signal received by separate receiver when I try to send
```
[16:46:41][D][remote.lg:053]: Received LG: data=0xA55A38C7, nbits=32
[16:46:41][D][remote.nec:068]: Received NEC: address=0xA55A, command=0x38C7
[16:46:41][D][remote.pioneer:144]: Received Pioneer: rc_code_X=0xA51C
[16:46:41][D][remote.lg:053]: Received LG: data=0xA55A38C7, nbits=32
[16:46:41][D][remote.nec:068]: Received NEC: address=0xA55A, command=0x38C7
[16:46:41][D][remote.pioneer:144]: Received Pioneer: rc_code_X=0xA51C
[16:46:41][D][remote.lg:053]: Received LG: data=0xA55A38C7, nbits=32
[16:46:41][D][remote.nec:068]: Received NEC: address=0xA55A, command=0x38C7
[16:46:41][D][remote.pioneer:144]: Received Pioneer: rc_code_X=0xA51C
[16:46:41][D][remote.lg:053]: Received LG: data=0xA55A38C7, nbits=32
[16:46:41][D][remote.nec:068]: Received NEC: address=0xA55A, command=0x38C7
[16:46:41][D][remote.pioneer:144]: Received Pioneer: rc_code_X=0xA51C
```

**Additional information and things you've tried:**

lawrence-jeff, 2020-12-05:
I am working on a similar effort with VSX-454 (Circa 1994). I am working on getting it working via the system control bus input but I successfully started with using the IR transmitter off an ESP8266. My codes are the same as yours A55A38C7 when using NEC and A51C when using Pioneer
I do have 50% duty cycle though - maybe this will help you they both work fine. I have the system control port working as well with a raw signal and a small circuit. I can share that when complete if you are interested

remote_transmitter:
  pin: D6
  carrier_duty_percent: 50%

switch:
  - platform: template
    name: "Pioneer Power"
    turn_on_action:
      remote_transmitter.transmit_nec:
        address: 0xA55A
        command: 0x38C7

  - platform: template
    name: "Pioneer Power PIonCode"
    turn_on_action:
      remote_transmitter.transmit_pioneer:
        rc_code_1: 0xA51C

DTTerastar, 2021-02-23:
> I am working on a similar effort with VSX-454 (Circa 1994). I am working on getting it working via the system control bus input but I successfully started with using the IR transmitter off an [...]

I'm interested in your circuit using the system control port.  I'm trying to do the same.

lawrence-jeff, 2021-02-23:
I used the circuit here https://sudonull.com/post/110286-RaspberryPi-Pioneer-System-Remote with a random transistor I had on hand. For an ESP I used the 5 volt USB power on my ESP.. so the transistor both inverted the output (which you need) and allowed the 3.3 GPIO to generate the 5v signal

On my receiver it didn't work with the ground attached to the shield of the plug, but i had to connect my circuit ground to the receiver chassis (which I saw reference to on other blogs so must be normal for some devices). Then it worked fine. I have a pretty comprehensive ESPhome config if you want it.

DTTerastar, 2021-02-23:
I do!

closed 2021-07-07 by stale[bot]