esphome/issues #1595: Climate or transmitter broken in 1.15.3

ppd13, 2020-11-04:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassio in Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**
D1 mini Esp8266

**ESPHome version (latest production, beta, dev branch)**
ESPHome version v1.15.3

**Affected component:**
climate or transmitter

**Description of problem:**
I was running 1.15.2 and was working fine. Since i upgraded to 1.15.3 cant control aircon anymore. When revert back to 1.15.2 it works again. Not sure if climate or transmitter module is broken in this version

**Problem-relevant YAML-configuration entries:**
```yaml
remote_transmitter:
  pin: D1
  carrier_duty_percent: 50%

climate:
  - platform: daikin
    name: "gbr aircon"
    supports_heat: False

```

**Logs (if applicable):**

```
[08:19:55][D][climate:262]: 'mbr aircon' - Sending state:
[08:19:55][D][climate:265]:   Mode: OFF
[08:19:55][D][climate:270]:   Fan Mode: AUTO
[08:19:55][D][climate:273]:   Swing Mode: OFF
[08:19:55][D][climate:282]:   Target Temperature: 27.00°C

```

**Additional information and things you've tried:**
version changes

Mummer, 2020-11-09:
I think that's broken by this commit: 3c440b5
https://github.com/sashao/esphome/commit/3c440b5af3bfc699c8326fe73ba5b7233ca4fd37

Number like 16383 is not a random number.
delayMicroseconds() maximum number is coincidentally 16383 and that's why someone divide number in uint32_t by 16383, to prevent disaster. Previous approach was not good either but better than current one. 
It should look like this 
` 
delay(usec / 1000UL); `
`delayMicroseconds(usec % 16383UL);
`

glmnet, 2020-11-10:
If this commit broke the daikin then daikin is buggy in a way that sends a value which would fool the issue that was fixed by this commit. 

Otherwise it's not clear to me. 

Can you elaborate?

glmnet, 2020-11-10:
[this ](https://github.com/esphome/esphome/blob/eb5c4b7c4f9ad66c2f8612024865109e4dda4777/esphome/components/daikin/daikin.h#L38) here is a delay for 32300 us:

```
// new algorithm
    delay(usec / 1000UL); // delays 32 ms
    delayMicroseconds(usec % 1000UL); // delays 300 microseconds
// total delay 32300us as expected
```

```
// old algorithm
   delay(usec / 16383); // division returns 1.97 converted to integer, delays 1 ms
   delayMicroseconds(usec % 16383UL); // delays 15917 microseconds
// total delay 16 ms and 917 microsecons
```

Does this make sense?
If this is the case, then many other stuff might be broken by this commit

Mummer, 2020-11-10:
You are totally right and I'm an idiot. Sorry :-(

glmnet, 2020-11-10:
That commit was included in 1.15.3, you can try 1.15.2, a better idea will be to try modifying the 32300 value there by 16900

I'm adding @buxtronix and @puuu as they are the main contributors of this feature for better testing and a proper fix.

puuu, 2020-11-12:
The discussed patch for the esp8266 platform (esphome/esphome#1322) looks correct. I would also like to remind that this is only relevant for times longer than 16383 µs.

With regard to the Daikin implementation, the message space is typically 32300 µs. See also other implementations:
- [DaikinHeatpumpARC417IR.h 30000 µs](https://github.com/ToniA/arduino-heatpumpir/blob/master/DaikinHeatpumpARC417IR.h#L16)
- [DaikinHeatpumpARC480A14IR.h 32300 µs](https://github.com/ToniA/arduino-heatpumpir/blob/master/DaikinHeatpumpARC480A14IR.h#L16)
- [DaikinHeatpumpIR.h 32300 µs](https://github.com/ToniA/arduino-heatpumpir/blob/master/DaikinHeatpumpIR.h#L16)

For the remote control I analysed, the message space was between 25100 µs and 34700 µs.

Personally, I use the Daikin transmitter with an ESP32 with 3 different Daikin models. The ESP32 platform in esphome uses the RMT implementation, which is not affected by this patch. I did not saw any issue, even with 1.15.3.

The patch likely brings the timing results of the ESP8266 platform closer to the ESP32 platform. It would be nice if someone could confirm this with measurements.

Assuming the patch is correct, then @ppd13 and @Mummer Daikin devices don't like this message space time and we need to find a better one.

We need the help of @ppd13 and @Mummer:
- Can you check the raw code of your daikin remote?
- Please try with different values of [`DAIKIN_MESSAGE_SPACE`](https://github.com/esphome/esphome/blob/dev/esphome/components/daikin/daikin.h#L38)? 16900 will bring you back to the old behaviour, but is probably too short. What value will cause this issue?

ofalvai, 2020-11-15:
I'm also struggling with my ESP8266 and Daikin climate after upgrading ESPHome to 1.15.3. After finding this thread I tried adjusting `DAIKIN_MESSAGE_SPACE`, and here are my findings:
- The original `32300` value doesn't work, as expected
- Using `16900` to emulate the old behavior doesn't work for me
- Tried increasing this value, but none of the following values works (for me at least): `18900, 25100, 28100, 30000`
- Tried `15000` and it works! I still don't understand why, but this my experience with my setup. This might be useful to others.

hectorgimenez, 2021-01-06:
> I'm also struggling with my ESP8266 and Daikin climate after upgrading ESPHome to 1.15.3. After finding this thread I tried adjusting `DAIKIN_MESSAGE_SPACE`, and here are my findings:
>
> * The [...]

Same here, when upgrading to ESPHome 1.15.3 the Daikin IR stopped working for me, with `15000` value on `DAIKIN_MESSAGE_SPACE ` as @ofalvai suggested works perfectly, but no idea why. But not sure if changing this to 15000 could affect ESP32 boards then? Not even sure WHY `15000` (despite it works)

hcoohb, 2021-01-08:
Hi guys,
I had the same problem in my custom component since 1.15.3.
I did a bit of debug with 2 esp8266 (1 receiver, 1 transmitter) and I believe I have found the issue:
 The `delay()` function has never actually worked in this case!
- When on 1.15.2, because of the wrong code, the `delayMicroseconds(usec % 16383UL)` would still create a delay of up to 16383us. In the case of the daikin it would create 15.917ms space which was enough for the AC to recognise that as a message break
- With the new corrected code, the delay generated now is only up to 1ms, therefore breaking stuff. @hectorgimenez that is why 15000 works for you right now (only delayMicroseconds() called = create 15000us delay) whereas 16900 only generates an actual delay of 16900 % 1000 = 900us

I also confirmed this using this code:
```cpp
if (usec <= 16383UL) {
    delayMicroseconds(usec);
  } else {
    uint32_t orig = micros();
    delay(usec / 1000UL);
    ESP_LOGD(TAG, "delay finished: %lu us asked, %lu us happened",usec / 1000UL * 1000UL, micros() - orig);
    orig = micros();
    delayMicroseconds(usec % 1000UL);
    ESP_LOGD(TAG, "delayMicro finished: %lu us asked, %lu us happened", usec % 1000UL, micros() - orig);
  }
```
which generated outputs like so when transmitting a space of 32666us:
`[00:45:37][D][remote_transmitter:066]: Sending remote code...`
`[00:45:37][D][helpers:183]: delay finished: 32000 us asked, 27 us happened`
`[00:45:37][D][helpers:186]: delayMicro finished: 666 us asked, 668 us happened`
`[00:45:37][D][helpers:183]: delay finished: 32000 us asked, 14 us happened`
`[00:45:37][D][helpers:186]: delayMicro finished: 666 us asked, 668 us happened`
As you can see, the delay() function does not do its job while the delayMicroseconds works as expected.

- So why does the `delay()` not work ?
I believe it is because we are pausing all other interrupts while transmitting. We are creating an [InterruputLock at line 72 in RemoteTransmitterComponent::send_internal](https://github.com/esphome/esphome/blob/5c86f332b269fd3e4bffcbdf3359a021419effdd/esphome/components/remote_transmitter/remote_transmitter_esp8266.cpp#L71). And the problem is that the `delay()` function internally uses interrupts....

If we rewrite delay_microseconds_accurate() like so:
```cpp
void delay_microseconds_accurate(uint32_t usec) {
  if (usec == 0)
    return;

  for (; usec > 16383UL; usec -= 16383UL)
    delayMicroseconds(16383UL);
  delayMicroseconds(usec);
}
```
This code works and I confirm all the timing to be correct!

The one thing I am not sure is: should we feed the watchdog in between the `delayMicroseconds()` calls? Otherwise we might wdt for long breaks...?

Happy to send a PR if you want me to

closed 2021-02-01 by glmnet