esphome/issues #1597: SIM800L Component - Home Assistant "esphome.send_sms" service doesn't understand HA templates

sunfos2019, 2020-11-05:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

HassOS 4.15
Supervisor 2020.11.0
Home Assistant 0.117.4

**ESP (ESP32/ESP8266, Board/Sonoff):**

D1 mini

**ESPHome version (latest production, beta, dev branch)**

1.15.3

**Affected component:**

SIM800L
https://esphome.io/components/sim800l.html

**Description of problem:**
I don't know if this problem concerns ESPHome or Home Assistant.

I have D1 mini with SIM800L attached and configured with ESPHome. Everything works fine - I can receive and send sms via Home Assistant.
When sending sms via Home Assistant service "esphome.send_sms" it sends all message as text and doesn't understand Home Assistant templates inside message like 'Phone Charge {{ states("sensor.phone_icloud_battery") }}%.' And phone receives exactly that message - **'Phone Charge {{ states("sensor.phone_icloud_battery") }}%.'**
But when sending such message for example via Home Assistant's Telegram integration I receive message like **Phone Charge 80%.**

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: d1_mini_11
  platform: ESP8266
  board: d1_mini

wifi:
  networks:
  - ssid: "xxx"
    password: "xxx"
  manual_ip:
    static_ip: 192.168.1.54
    subnet: 255.255.255.0
    gateway: 192.168.1.2

ota:

logger:
  baud_rate: 0

uart:
  tx_pin: D8
  rx_pin: D7
  baud_rate: 9600

text_sensor:
  - platform: template
    id: sms_sender
    name: "SMS_Sender"
  - platform: template
    id: sms_message
    name: "SMS_Message"

sim800l:
  on_sms_received:
  - lambda: |-
      id(sms_sender).publish_state(sender);
      id(sms_message).publish_state(message);

api:
  services:
  - service: send_sms
    variables:
      recipient: string
      message: string
    then:
    - sim800l.send_sms:
        recipient: !lambda 'return recipient;'
        message: !lambda 'return message;'
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

glmnet, 2020-11-05:
this is working for me, in HA automation action I have:

```yaml
data_template:
  message: 'Internet connection lost ({{ states.sensor.active_isp.state }})'
  recipient: '+xxxx'
service: esphome.eh_planta_alta_send_sms
```

HA complains about that `data_template` does not exists or something like that, I've read somewhere that now it should just work with `data` but this is old code which still works

glmnet, 2020-11-05:
Just tested with `data:` and still works as expected

sunfos2019, 2020-11-05:
> Just tested with `data:` and still works as expected

It's strange that it works for you, it doesn't work for me. I tried:
```yaml
data_template:
  message: 'TEST {{ states("sensor.phone_icloud_battery") }}%.'
  recipient: '+1234567'
```
and
```yaml
data:
  message: 'TEST {{ states("sensor.phone_icloud_battery") }}%.'
  recipient: '+1234567'
```
Home Assistant doesn't like it:
```
extra keys not allowed @ data['data_template']
extra keys not allowed @ data['data']
```

glmnet, 2020-11-05:
I see a difference, where I use: `states.sensor.active_isp.state` (which is a string) you use `states("sensor.phone_icloud_battery")` I'm not proficient in HA but could you try something like `states.sensor.phone_icloud_battery` or `states.sensor.phone_icloud_battery.state` may be in the jinga template editor

sunfos2019, 2020-11-05:
`{{ states("sensor.phone_icloud_battery") }}`
`{{ states('sensor.phone_icloud_battery') }}`
`{{ states.sensor.phone_icloud_battery.state }}`
all these give same result. And all rendered fine in template editor.
I think the problem is not about it. But thanks.

I wonder how does it work for you) My esp8266 ESPHome piece of code looks like:
```yaml
api:
  services:
  - service: send_sms
    variables:
      recipient: string
      message: string
    then:
    - sim800l.send_sms:
        recipient: !lambda 'return recipient;'
        message: !lambda 'return message;'
```

glmnet, 2020-11-05:
I believe is the same

I have public all my ESPHome config. Here is the node which has the sim800l

https://github.com/glmnet/esphome_devices/blob/master/my_home/EHPlantaAlta.yaml

closed 2021-03-19 by stale[bot]