esphome/issues #1600: ESP32 with Neopixelbus library stuck on scanning wifi

Probastian, 2020-11-05:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**ESPHome version (latest production, beta, dev branch)**

**Affected component:**

**Description of problem:**
ESP gets stuck on Scanning wifi when using neopixelbus light. It works in FastLED and without.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: teststrip
  platform: ESP32
  board: esp32dev

wifi:
  ssid: "SSID"
  password: "YouShallNot(ReadMy)Pass(word)"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Teststrip Fallback Hotspot"
    password: "ssBhVl7X1RhH"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "Robert17"

ota:
  password: "Robert17"

# Example configuration entry
light:
  - platform: neopixelbus
    variant: WS2812X
    type: GRBW
    pin: 2
    num_leds: 10
    name: "NeoPixel Light"
    effects:
      - strobe:
      - addressable_rainbow:
      - addressable_scan:
      - addressable_twinkle:
      - addressable_random_twinkle:
      - addressable_fireworks:
      - addressable_flicker:
```

**Logs (if applicable):**

```
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[21:09:15][D][wifi:324]: Starting scan...

```

**Additional information and things you've tried:**
Tried fast_connect option and a few variations and adjustements of the code and its parameters.
Also tried using FastLED, it works without a problem.

Probastian, 2020-11-05:
After setting the log level to DEBUG it somehow worked. Don't know why, but I'm glad it did ^^

Probastian, 2020-11-05:
The ESP restarted itself since no OTA attempt was made.
Now it doesn't work again. This is very weird...

Probastian, 2020-11-05:
I now tried using a static ip, which also didn't help.
On log level VERY_VERBOSE it looks like this:
```
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[22:09:12]i: wifi driver task: 3ffb5ae8, prio:23, stack:3584, core=0
[22:09:12]
[22:09:12][VV][esp-idf:000]: I (482) wifi: wifi firmware version: 7997e4b
[22:09:12]
[22:09:12][VV][esp-idf:000]: I (482) wifi: config NVS flash: enabled
[22:09:12]
[22:09:12][VV][esp-idf:000]: I (484) wifi: config nano formating: disabled
[22:09:12]
[22:09:12][VV][esp-idf:000]: I (491) wifi: Init dynamic tx buffer num: 32
[22:09:12]
[22:09:12][VV][esp-idf:000]: I (497) wifi: Init data frame dynamic rx buffer num: 32
[22:09:12]
[22:09:12][VV][esp-idf:000]: I (505) wifi: Init management frame dynamic rx buffer num: 32
[22:09:12]
[22:09:12][VV][esp-idf:000]: I (513) wifi: Init management short buffer num: 32
[22:09:12]
[22:09:12][VV][esp-idf:000]: I (521) wifi: Init static rx buffer size: 1600
[22:09:12]
[22:09:12][VV][esp-idf:000]: I (528) wifi: Init static rx buffer num: 16
[22:09:12]
[22:09:12][VV][esp-idf:000]: I (534) wifi: Init dynamic rx buffer num: 32
[22:09:12]
[22:09:12][VV][esp-idf:000]: I (609) wifi: mode : softAP (c4:4f:33:67:31:72)
[22:09:12]
[22:09:12][VV][esp-idf:000]: I (610) wifi: Total power save buffer number: 16
[22:09:12]
[22:09:12][VV][esp-idf:000]: I (612) wifi: Init max length of beacon: 752/752
[22:09:12]
[22:09:12][VV][esp-idf:000]: I (619) wifi: Init max length of beacon: 752/752
[22:09:12]
[22:09:12][D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[22:09:12][D][WiFiGeneric.cpp:337] _eventCallback(): Event: 14 - AP_STOP
[22:09:12][V][wifi_esp32:419]: Event: WiFi AP start
[22:09:12][V][wifi_esp32:352]: Event: WiFi ready
[22:09:12][VV][esp-idf:000]: I (647) wifi: mode : sta (c4:4f:33:67:31:71)
[22:09:12]
[22:09:12][D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[22:09:12][V][wifi_esp32:361]: Event: WiFi STA start
[22:09:12][D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[22:09:12][V][wifi_esp32:361]: Event: WiFi STA start
[22:09:12][VV][esp-idf:000]: I (674) wifi: Set ps type: 1
[22:09:12]
[22:09:12]
[22:09:12][D][wifi:324]: Starting scan...
[22:09:12][VV][scheduler:152]: Running interval '' with interval=5000 last_execution=4294961077 (now=369)
[22:09:12][VV][light.addressable:171]: Addressable Light 'NeoPixel Light' (effect_active=NO next_show=YES)
[22:09:12][VV][light.addressable:175]:   [ 0] Color: R=  0 G=  0 B=  0 W=  0
[22:09:12][VV][light.addressable:175]:   [ 1] Color: R=  0 G=  0 B=  0 W=  0
[22:09:12][VV][light.addressable:175]:   [ 2] Color: R=  0 G=  0 B=  0 W=  0
[22:09:12][VV][light.addressable:175]:   [ 3] Color: R=  0 G=  0 B=  0 W=  0
[22:09:12][VV][light.addressable:175]:   [ 4] Color: R=  0 G=  0 B=  0 W=  0
[22:09:12][VV][light.addressable:175]:   [ 5] Color: R=  0 G=  0 B=  0 W=  0
[22:09:12][VV][light.addressable:175]:   [ 6] Color: R=  0 G=  0 B=  0 W=  0
[22:09:12][VV][light.addressable:175]:   [ 7] Color: R=  0 G=  0 B=  0 W=  0
[22:09:12][VV][light.addressable:175]:   [ 8] Color: R=  0 G=  0 B=  0 W=  0
[22:09:12][VV][light.addressable:175]:   [ 9] Color: R=  0 G=  0 B=  0 W=  0
[22:09:12][VV][light.addressable:177]: 
[22:09:14][D][WiFiGeneric.cpp:337] _eventCallback(): Event: 1 - SCAN_DONE
[22:09:14][V][wifi_esp32:357]: Event: WiFi Scan Done status=0 number=7 scan_id=128
```

stefanheinrichsen, 2020-11-06:
Having the same issue here... I noticed some compiler warnings related to neopixelbus

```
.piolibdeps/co2ampel_unhb_demo/NeoPixelBus-esphome/src/internal/Esp32_i2s.c: In function 'i2sSetPins':
.piolibdeps/co2ampel_unhb_demo/NeoPixelBus-esphome/src/internal/Esp32_i2s.c:264:13: warning: 'i2sSignal' may be used uninitialized in this function [-Wmaybe-uninitialized]
             gpio_matrix_out(out, i2sSignal, invert, false);
             ^
.piolibdeps/co2ampel_unhb_demo/NeoPixelBus-esphome/src/internal/Esp32_i2s.c: In function 'i2sInit':
.piolibdeps/co2ampel_unhb_demo/NeoPixelBus-esphome/src/internal/Esp32_i2s.c:383:5: warning: 'i2sIntSource' may be used uninitialized in this function [-Wmaybe-uninitialized]
     esp_intr_alloc(i2sIntSource, ESP_INTR_FLAG_IRAM | ESP_INTR_FLAG_LEVEL1, &i2sDmaISR, &I2S[bus_num], &I2S[bus_num].isr_handle);
     ^
```
My LED part of the yaml file contains:

```
light:
  - platform: neopixelbus
    method: ESP32_I2S_1
    pin: GPIO4
    num_leds: 12
    name: "ws2812 ring"
    id: "ws2812"
    restore_mode: ALWAYS_OFF
```

Maybe someones has idea idea with the above details

Probastian, 2020-11-06:
I have those same compiler warnings. I'll try downgrading and hopefully pinpointing an exact version where it all broke. Maybe I can even find an fix for it.

mihaiblaga89, 2020-11-06:
had the same problem but managed to get around it by using 

```
- platform: fastled_clockless
    chipset: WS2812B
```

instead of  `neopixelbus`

pengelen, 2020-11-07:
this is a bug in the Neopixelbus 2.5.7, you can solve this with the following:
----------------
esphome:
  name: ESP12
  platform: ESP32
  board: nodemcu-32s

  # a workaround for the latest bug in the neopixel lib bug
  platformio_options:
    lib_deps: NeoPixelBus@2.5.6
-------------------

it also works with NeoPixelBus@2.6.0

Probastian, 2020-11-09:
I found another workaround (well @EuroTrash from the ESPHome Discord found it to be fair).
You can add neopixelbus as a custom_component and define a different version to be used. It will automatically use the custom component instead of the offical one.

Just **get [those files from the dev branch of esphome](https://github.com/esphome/esphome/tree/dev/esphome/components/neopixelbus)** and put them **in  <CONFIG_FOLDER>/custom_components/neopixelbus**. You will need to **change the last line of the file _lights.py_** to your desired version. (I tested 2.5.2, which was working fine. 2.5.6 should probably work too. I also tested 2.6.0 but that didn't compile). After doing that you probably want to **Clean Build Files**, but I guess that's **optional**.

peszkobartek, 2020-11-09:
[image]
 I doesn't work for me, I have the same issues as others

Probastian, 2020-11-09:
You can try my workaround in the comment above you. It's a little more complicated but gets the job done.

peszkobartek, 2020-11-10:
Just quick update. When I tried again the simple method it worked. After the initial test, everything works ok. During compiling file there was info about downloading the old NeoPixelBus, so maybe there was a problem with the connection to the source server.

Good Luck everyone ;)

stefanheinrichsen, 2020-11-10:
> esphome:
>...
>   # a workaround for the latest bug in the neopixel lib bug
>   platformio_options:
>     lib_deps: NeoPixelBus@2.5.6
> ```

Works here too but when using this option I have to manually define all other libs too. Else it gets confused and throws missing dependencies during compile.
Is this normal or did I forgot something else?

peszkobartek, 2020-11-10:
I don't have that problem, my code is as below, on the instance i am using, sensors, binary sensors, light, power supply, ledc

`esphome:
  name: esp32_bedroom
  platform: ESP32
  board: esp-wrover-kit

  # a workaround for the latest bug in the neopixel lib bug
  platformio_options:
    lib_deps: NeoPixelBus@2.5.6

wifi:
  ssid: "xxx"
  password: "xxx"
  fast_connect: true
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32 Bedroom Fallback Hotspot"
    password: "xxx"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

web_server:
  port: 80`

guardmedia, 2020-11-12:
Confirmed, issue is resolved with newer NeoPixelBus@2.6.0. Not sure if there is a way to add/modify lib_deps without losing default values though.

`esphome:
  name: $devicename
  platform: ESP32
  board: ttgo-t1
  platformio_options:
    lib_deps: NeoPixelBus@2.6.0`

pa1okz, 2021-01-19:
@OttoWinter , Would it be a good idea if you bump away from NeoPixelBus 2.5.7. as it doesn't work with anybody and including Platform IO options causes certain other dependencies missing...

emosenkis, 2021-04-04:
@glmnet can you please bump the NeoPixelBus dependency as described above?

closed 2021-08-22 by stale[bot]