esphome/issues #1602: MQTT over SSL won't connect with SNTP configured

kickouille, 2020-11-06:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 (different flavors, Adafruit Huzzah, Wemos Mini, Wemos Mini Pro)
**ESPHome version (latest production, beta, dev branch)**

Firmware: esphome v1.15.3 Nov 4 2020, 11:15:36
**Affected component:**

https://esphome.io/components/mqtt.html
https://esphome.io/components/time.html#sntp-configuration

**Description of problem:**

When MQTT over SSL is used in conjuction with SNTP, ESPhome starts connecting to MQTT server, but states it is disconnected some seconds after. Connection is never established.

If I remove SNTP configuration then MQTT over SSL works.

If I keep SNTP configuration and connect to an MQTT server without SSL it works.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: ***
  platform: ESP8266
  board: huzzah

wifi:
  networks:
  - ssid: "***"
    password: "***"
    priority: 50
  - ssid: "***"
    password: "***"
    priority: 40
  - ssid: "***"
    password: "***"
    priority: 60

logger:

mqtt:
  broker: ***
  port: ***
  username: ***
  password: ***
  ssl_fingerprints:
    - ***

ota:
  password: "***"

time:
  - platform: sntp
    id: sntp_time
    timezone: "Europe/Zurich"

status_led:
  pin:
    number: 2
    inverted: yes

```

**Logs (if applicable):**

```
[15:28:54][C][status_led:014]: Setting up Status LED...
[15:28:54][C][ota:366]: There have been 1 suspected unsuccessful boot attempts.
[15:28:54][I][app:029]: Running through setup()...
[15:28:54][C][wifi:033]: Setting up WiFi...
[15:28:54][D][wifi:324]: Starting scan...
[15:29:00][D][wifi:339]: Found networks:
[15:29:00][I][wifi:385]: - '***' (***) ▂▄▆█
[15:29:00][D][wifi:386]:     Channel: 1
[15:29:00][D][wifi:387]:     RSSI: -51 dB
[15:29:00][I][wifi:385]: - '***' (***) ▂▄▆█
[15:29:00][D][wifi:386]:     Channel: 6
[15:29:00][D][wifi:387]:     RSSI: -88 dB
[15:29:00][D][wifi:389]: - '***' (***) ▂▄▆█
[15:29:00][D][wifi:389]: - '***' (***) ▂▄▆█
[15:29:00][D][wifi:389]: - '***' (***) ▂▄▆█
[15:29:00][D][wifi:389]: - '' (***) ▂▄▆█
[15:29:00][D][wifi:389]: - '***' (***) ▂▄▆█
[15:29:00][D][wifi:389]: - '***' (***) ▂▄▆█
[15:29:00][D][wifi:389]: - '***' (***) ▂▄▆█
[15:29:00][I][wifi:194]: WiFi Connecting to '***'...
[15:29:02][I][wifi:457]: WiFi Connected!
[15:29:02][C][wifi:303]:   SSID: '***'
[15:29:02][C][wifi:304]:   IP Address: ***
[15:29:02][C][wifi:306]:   BSSID: ***
[15:29:02][C][wifi:307]:   Hostname: '***'
[15:29:02][C][wifi:311]:   Signal strength: -57 dB ▂▄▆█
[15:29:02][C][wifi:315]:   Channel: 1
[15:29:02][C][wifi:316]:   Subnet: ***
[15:29:02][C][wifi:317]:   Gateway: ***
[15:29:02][C][wifi:318]:   DNS1: ***
[15:29:02][C][wifi:319]:   DNS2: (IP unset)
[15:29:02][C][ota:029]: Over-The-Air Updates:
[15:29:02][C][ota:030]:   Address: ***.local:8266
[15:29:02][C][ota:032]:   Using Password.
[15:29:02][C][mqtt:025]: Setting up MQTT...
[15:29:02][D][mqtt:103]: Resolving MQTT broker IP address...
[15:29:02][D][mqtt:136]: Resolved broker IP address to ***
[15:29:02][I][mqtt:162]: Connecting to MQTT...
[15:30:02][I][mqtt:162]: Connecting to MQTT...
[15:30:32][W][mqtt:247]: MQTT Disconnected: TCP disconnected.
[15:30:32][D][mqtt:103]: Resolving MQTT broker IP address...
[15:30:32][D][mqtt:136]: Resolved broker IP address to ***
[15:30:32][I][mqtt:162]: Connecting to MQTT...
[15:30:54][I][ota:046]: Boot seems successful, resetting boot loop counter.

```

kickouille, 2020-11-12:
Hello,

Is anybody able to reproduce this issue?

Thank you for your help!

Regards,

Kick

closed 2021-03-27 by stale[bot]

NicoFR75, 2021-06-07:
Hello,

I was able to reproduce this issue, I'm using SNTP in my yaml files and never manage to get MQTT over SSL working.
By removing the "time" part from my yaml file, I can now connect to MQTT over SSL.

I'm using a personnal CA to issue my certificate (not self-signed) and my MQTT is "EMQ X Broker" v4.1.0

This is working :
[image]

This is not :
[image]