esphome/issues #1603: Invalid preamble and immediate disconnection when trying to connect

blackhack, 2020-11-06:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
pip

**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP32

**ESPHome version (latest production, beta, dev branch)**
1.15.3

**Affected component:**
...

**Description of problem:**
I'm following this https://esphome.io/guides/getting_started_command_line.html to setup the ESP32, but wherever I try to connect to the board via either toolrom.local:6053 or IP:6053, I get the following error in the browser plus an error in the log.

```
This page isn’t working
toolrom.local didn’t send any data.
ERR_EMPTY_RESPONSE
```
I get the same log error when trying to connect using the the android app, with the difference that the hostname doesn't work at all, so I have to use the ip.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: toolrom
  platform: ESP32
  board: esp32dev

wifi:
  ssid: "ABCD"
  password: "blablabla"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "ToolRom"
    password: "blablabla"

captive_portal:

# Enable logging
logger:
  level: VERBOSE

# Enable Home Assistant API
api:

ota:

switch:
  platform: gpio
  name: "TestSwitch"
  pin: 5
```

**Logs (if applicable):**
```
[15:40:54][C][wifi:443]: WiFi:
[15:40:54][C][wifi:303]:   SSID: 'ABCD'
[15:40:54][C][wifi:304]:   IP Address: 192.168.1.94
[15:40:54][C][wifi:306]:   BSSID: FC:BC:D1:E6:47:CC
[15:40:54][C][wifi:307]:   Hostname: 'toolrom'
[15:40:54][C][wifi:311]:   Signal strength: -59 dB ▂▄▆█
[15:40:54][V][wifi:313]:   Priority: 0.0
[15:40:54][C][wifi:315]:   Channel: 11
[15:40:54][C][wifi:316]:   Subnet: 255.255.255.0
[15:40:54][C][wifi:317]:   Gateway: 192.168.1.254
[15:40:54][C][wifi:318]:   DNS1: 1.1.1.1
[15:40:54][C][wifi:319]:   DNS2: 1.0.0.1
[15:40:54][C][switch.gpio:042]: GPIO Switch 'Living Room Dehumidifier'
[15:40:54][C][switch.gpio:043]:   Pin: GPIO5 (Mode: OUTPUT)
[15:40:54][C][switch.gpio:059]:   Restore Mode: Restore (Defaults to OFF)
[15:40:54][C][logger:185]: Logger:
[15:40:54][C][logger:186]:   Level: VERBOSE
[15:40:54][C][logger:187]:   Log Baud Rate: 115200
[15:40:54][C][logger:188]:   Hardware UART: UART0
[15:40:54][C][captive_portal:169]: Captive Portal:
[15:40:54][C][ota:029]: Over-The-Air Updates:
[15:40:54][C][ota:030]:   Address: toolrom.local:3232
[15:40:54][C][api:095]: API Server:
[15:40:54][C][api:096]:   Address: toolrom.local:6053

......

[15:59:12][W][api.connection:048]: Invalid preamble from 192.168.1.120
[15:59:12][V][api.connection:693]: Error: Disconnecting 192.168.1.120
[15:59:12][D][api:067]: Disconnecting 192.168.1.120
```

**Additional information and things you've tried:**
I tried using the latest github commit with the same result, also I tried with an esp32dev and Lolin32 boards.
I thought maybe it was my main router, so I tried using a separate router with openwrt but not luck.

randybb, 2020-11-06:
Android app for ESPHome API? Native API is used to communicate with ESPHome command line tool OR Home Assistant. Sure, you can write you own program...
With Home Assistant it is super easy. Just need to open your home assistant web, follow these [steps](https://www.home-assistant.io/integrations/esphome/) or just click on Notifications and you should be able to see your newly connected device with all defined entities.

blackhack, 2020-11-07:
> Home Assistant

I was referring to the android Home Assistant app.

randybb, 2020-11-07:
Home Assistant app communicates with Home Assistant and Home Assistant with ESPHome device

blackhack, 2020-11-07:
> Home Assistant app communicates with Home Assistant and Home Assistant with ESPHome device
> 
> For web interface you need to enable [web server component](https://esphome.io/components/web_server.html) that will provide [REST API](https://esphome.io/web-api/index.html) if it is what you are looking for.

Sorry, I was really confused, but the component you linked does exactly what I want.

Thank you, I'm closing now!

closed 2020-11-07 by blackhack