esphome/issues #1605: ESPHome fails generating code for just more than 20 sensors.

ogry, 2020-11-07:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Windows 10/PIP install

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 Developer board
(https://www.aliexpress.com/item/32842941195.html)

**ESPHome version (latest production, beta, dev branch)**

ESPHome v1.15.3 (may be Dev version, but is what gets with a simple 'pip install esphome')

**Affected component:**

Not sure, as it seems to be something related to generation proces.
https://esphome.io/components/sensor/index.html

**Description of problem:**
Error compiling from a yaml file with more than 20 sensors defined.

There is the full YAML file: https://pastebin.com/SW4sViCu

There are 155 rfreceiver sensors very simple, mostly like this one:
**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
binary_sensor:
  - id: botonmandoa01
    platform: remote_receiver
    name: 'Boton MandoA01'
    filters:
      - delayed_off: 250ms
    rc_switch_raw: 
      code: '010111000101000011000000'
    on_press:
     then:
       - switch.toggle: luz_oeste
       - delay: 20s
       - switch.turn_off: luz_oeste
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE
Archiving .pioenvs\testel32\libFrameworkArduino.a
src\main.cpp: In function 'void setup()':
src\main.cpp:802:6: note: variable tracking size limit exceeded with -fvar-tracking-assignments, retrying without
 void setup() {
      ^
Linking .pioenvs\testel32\firmware.elf
Retrieving maximum program size .pioenvs\testel32\firmware.elf
Building .pioenvs\testel32\firmware.bin
Checking size .pioenvs\testel32\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  14.1% (used 46208 bytes from 327680 bytes)
Flash: [======    ]  60.2% (used 1104118 bytes from 1835008 bytes)
esptool.py v2.6
================================================= [SUCCESS] Took 48.53 seconds =================================================
�[32mINFO Successfully compiled program.�[0m

```

**Additional information and things you've tried:**
Cutting of the YAML to exactly 20 sensors, all works again.

glmnet, 2020-11-09:
What is the failure?

glmnet, 2020-11-09:
Seems like you're confusing a note with a failure. Please explain what does not work and I'll reopen.

ogry, 2020-11-09:
No, this is a failure. The code being generated, compiled and even linked, but without a main, it fails with this trace:

```
Compiling .pioenvs\testel32\src\main.cpp.o
src\main.cpp: In function 'void setup()':
src\main.cpp:802:6: note: variable tracking size limit exceeded with -fvar-tracking-assignments, retrying without
 void setup() {
      ^
Linking .pioenvs\testel32\firmware.elf
Building .pioenvs\testel32\firmware.bin
Retrieving maximum program size .pioenvs\testel32\firmware.elf
Checking size .pioenvs\testel32\firmware.elf
esptool.py v2.6
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  14.1% (used 46208 bytes from 327680 bytes)
Flash: [======    ]  60.2% (used 1104118 bytes from 1835008 bytes)
============================================= [SUCCESS] Took 40.59 seconds =============================================
[32mINFO Successfully compiled program.[0m
[32mINFO Running:  platformio run -d testel32 -t upload --upload-port COM5[0m
Processing testel32 (board: esp32doit-devkit-v1; framework: arduino; platform: espressif32@1.12.4)
------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (1.12.4) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 3.10004.201016 (1.0.4)
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - tool-mkspiffs 2.230.0 (2.30)
 - toolchain-xtensa32 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Library Manager: Installing Hash
Library Manager: Already installed, built-in library
Found 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <AsyncTCP-esphome> 1.1.1
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <FS> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.7
|   |-- <AsyncTCP-esphome> 1.1.1
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|   |-- <ArduinoJson-esphomelib> 5.13.3
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <DNSServer> 1.1.0
|   |-- <WiFi> 1.0
|-- <Update> 1.0
|-- <WiFi> 1.0
Building in release mode
Retrieving maximum program size .pioenvs\testel32\firmware.elf
Checking size .pioenvs\testel32\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  14.1% (used 46208 bytes from 327680 bytes)
Flash: [======    ]  60.2% (used 1104118 bytes from 1835008 bytes)
Configuring upload protocol...
AVAILABLE: esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Use manually specified: COM5
Uploading .pioenvs\testel32\firmware.bin
esptool.py v2.6
Serial port COM5
Traceback (most recent call last):
  File "C:\Users\ogryd\.platformio\packages\tool-esptoolpy@1.20600.0\esptool.py", line 2959, in <module>
    _main()
  File "C:\Users\ogryd\.platformio\packages\tool-esptoolpy@1.20600.0\esptool.py", line 2952, in _main
    main()
  File "C:\Users\ogryd\.platformio\packages\tool-esptoolpy@1.20600.0\esptool.py", line 2652, in main
    esp = chip_class(each_port, initial_baud, args.trace)
  File "C:\Users\ogryd\.platformio\packages\tool-esptoolpy@1.20600.0\esptool.py", line 222, in __init__
    self._port = serial.serial_for_url(port)
  File "c:\python\lib\site-packages\serial\__init__.py", line 88, in serial_for_url
    instance.open()
  File "c:\python\lib\site-packages\serial\serialwin32.py", line 62, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port 'COM5': PermissionError(13, 'Acceso denegado.', None, 5)
*** [upload] Error 1
============================================== [FAILED] Took 4.60 seconds ==============================================
```

ogry, 2020-11-09:
When you tries that yaml, it compile, link and upload the code, and then it fails with this trace:

```
Writing at 0x0008c000... (86 %)
Writing at 0x00090000... (89 %)
Writing at 0x00094000... (91 %)
Writing at 0x00098000... (94 %)
Writing at 0x0009c000... (97 %)
Writing at 0x000a0000... (100 %)
Wrote 1104352 bytes (599642 compressed) at 0x00010000 in 13.7 seconds (effective 643.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
====================================================================== [SUCCESS] Took 21.99 seconds ======================================================================
[32mINFO Successfully uploaded program.[0m
[32mINFO Starting log output from COM5 with baud rate 115200[0m
[22:46:06]ets Jun  8 2016 00:22:57
[22:46:06]
[22:46:06]rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[22:46:06]configsip: 0, SPIWP:0xee
[22:46:06]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[22:46:06]mode:DIO, clock div:2
[22:46:06]load:0x3fff0018,len:4
[22:46:06]load:0x3fff001c,len:1044
[22:46:06]load:0x40078000,len:8896
[22:46:06]load:0x40080400,len:5828
[22:46:06]entry 0x400806ac
[22:46:07]ets Jun  8 2016 00:22:57
[22:46:07]
[22:46:07]rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[22:46:07]configsip: 0, SPIWP:0xee
[22:46:07]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[22:46:07]mode:DIO, clock div:2
[22:46:07]load:0x3fff0018,len:4
[22:46:07]load:0x3fff001c,len:1044
[22:46:07]load:0x40078000,len:8896
[22:46:07]load:0x40080400,len:5828
[22:46:07]entry 0x400806ac
[22:46:08]ets Jun  8 2016 00:22:57
[22:46:08]
```

glmnet, 2020-11-09:
Ok, I'm reopening this, your issue is related to a watchdog, not code generation, anyway, title says 20 sensors and your attached yaml has lot's (also you comment there are 155)

This kind of issues might be because there is a bug in esphome when handling lots of data (most people don't use lots of data) 

**Fix the title and carefully craft a minimum yaml which will create this issue** (otherwise anyone debugging this should do that and it's time consuming)

After providing new yaml I'll check again where might be the wdt triggering (rebooting the ESP continuously) and check if this can be solvable.

Keep in mind esphome is a software which makes stuff easy for end users like you, this comes with the tradeoff that complex stuff cannot be done quite easily (like handling 100+ remote codes blazing fast)

The way this thing works is it records remote signals and tests for each coincidence, this is time consuming for a microcontroller, another more optimal approach can be done, but again, this software has it's limitations.

ogry, 2020-11-09:
Ok. Sorry if I don't explain precisely and thank for the directions. I hope to do better next time...

This is the minimun YAML that runs ok:

https://pastebin.com/ZUWqyHHk

After adding one more sensor, it fails:

https://pastebin.com/PyG58vsW

with this trace:

```
Writing at 0x0008c000... (94 %)
Writing at 0x00090000... (97 %)
Writing at 0x00094000... (100 %)
Wrote 997392 bytes (549006 compressed) at 0x00010000 in 12.5 seconds (effective 638.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
====================================================================== [SUCCESS] Took 20.85 seconds ======================================================================
[32mINFO Successfully uploaded program.[0m
[32mINFO Starting log output from COM5 with baud rate 115200[0m
[23:30:41][I][logger:166]: Log initialized
[23:30:41][C][ota:366]: There have been 9 suspected unsuccessful boot attempts.
[23:30:41][I][app:029]: Running through setup()...
[23:30:41][C][remote_receiver.esp32:013]: Setting up Remote Receiver...
[23:30:41][C][remote_receiver.esp32:013]: Setting up Remote Receiver...
[23:30:41][W][remote_base:019]: Not enough RMT memory blocks avaiable, reduced to 3 blocks.
[23:30:41][C][wifi:033]: Setting up WiFi...
[23:30:42][D][wifi:324]: Starting scan...
[23:30:42]E (672) rmt: RMT[2] ERR
[23:30:42]E (672) rmt: status: 0x13080140
[23:30:43][D][wifi:339]: Found networks:
[23:30:43][I][wifi:385]: - 'testeles.guifi.net' (C0:25:E9:AE:38:AE) ▂▄▆█
[23:30:43][D][wifi:386]:     Channel: 1
[23:30:43][D][wifi:387]:     RSSI: -60 dB
[23:30:43][D][wifi:389]: - 'ogry24.guifi.net' (C0:3F:0E:12:9A:C8) ▂▄▆█
[23:30:43][D][wifi:389]: - 'Invitados' (C2:25:E9:AC:38:AE) ▂▄▆█
[23:30:43][D][wifi:389]: - 'ogry5.guifi.net' (DA:A7:56:1E:15:04) ▂▄▆█
[23:30:43][D][wifi:389]: - 'ogry.guifi.net' (D8:A7:56:1E:13:07) ▂▄▆█
[23:30:43][D][wifi:389]: - 'MOVISTAR_ADAD_EXT_R' (D8:47:32:6D:76:4B) ▂▄▆█
[23:30:43][D][wifi:389]: - 'MOVISTAR_6338' (E2:41:36:00:63:38) ▂▄▆█
[23:30:43][D][wifi:389]: - 'MOVIST_ADAD_EXT' (50:D4:F7:64:C4:52) ▂▄▆█
[23:30:43][I]***ERROR*** A stack overflow in task loopTask has been detected.
[23:30:43]abort() was called at PC 0x4008ddd8 on core 1
[23:30:43]
[23:30:43]Backtrace: 0x4008db90:0x3ffb0310 0x4008ddc1:0x3ffb0330 0x4008ddd8:0x3ffb0350 0x4008aee8:0x3ffb0370 0x4008cb08:0x3ffb0390 0x4008cabe:0x00000012
[33mWARNING Found stack trace! Trying to decode it[0m
[32mINFO Need to fetch platformio IDE-data, please stand by[0m
[32mINFO Running:  platformio run -d testel32test -t idedata[0m
[23:30:45]
[23:30:45]Rebooting...
[23:30:45]ets Jun  8 2016 00:22:57
[23:30:45]
[23:30:45]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[23:30:45]configsip: 0, SPIWP:0xee
[23:30:45]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[23:30:45]mode:DIO, clock div:2
[23:30:45]load:0x3fff0018,len:4
[23:30:45]load:0x3fff001c,len:1044
[23:30:45]load:0x40078000,len:8896
[23:30:45]load:0x40080400,len:5828
[23:30:45]entry 0x400806ac
[23:30:45][I][logger:166]: Log initialized
[23:30:45][C][ota:366]: There have been 10 suspected unsuccessful boot attempts.
[23:30:45][E][ota:371]: Boot loop detected. Proceeding to safe mode.
[23:30:45][I][app:029]: Running through setup()...
[23:30:45][C][wifi:033]: Setting up WiFi...
[... 23 lines cut ...]
[23:30:45]load:0x40080400,len:5828
[23:30:45]entry 0x400806ac
[23:30:45][I][logger:166]: Log initialized
```

The longest YAML without the _note_ in compiling phase is this:

https://pastebin.com/NX89EW9q

If you add a single more sensor, it fails with the _note_ in compiling fase and then fails after uploading with the trace of the first message:

```
Compiling .pioenvs\testel32test\FrameworkArduino\wiring_pulse.c.o
Compiling .pioenvs\testel32test\FrameworkArduino\wiring_shift.c.o
Archiving .pioenvs\testel32test\libFrameworkArduino.a
src\main.cpp: In function 'void setup()':
src\main.cpp:728:6: note: variable tracking size limit exceeded with -fvar-tracking-assignments, retrying without
 void setup() {
      ^
Linking .pioenvs\testel32test\firmware.elf
```

So, what if I change the subject to something like:

_ESPHome fails with more than 21 binary sensors declared_

glmnet, 2020-11-10:
Good info there...

Stack trace decoder is not working in Windows, I've sent a fix for that already, available in dev branch. 

> Debug exception reason: Stack canary watchpoint triggered (loopTask)

Anyway those exceptions are just confirming that the ESP32 is running out of memory. Stack memory to be more precise. I really don't know if this can be worked around easily or not (e.g. increasing stack size)

Try adding the debug component which will print out available resources on the esp node, make it work with the maximum qty of sensors before it fails and check available resources to see if it's running low on something.

ogry, 2020-11-10:
when I add the 'debug:' option, new information appears in the trace:

```
[14:50:32][C][ota:029]: Over-The-Air Updates:
[14:50:32][C][ota:030]:   Address: testel32test.local:3232
[14:50:32][C][api:095]: API Server:
[14:50:32][C][api:096]:   Address: testel32test.local:6053
[14:50:32][D][debug:023]: ESPHome version 1.15.3
[14:50:32][D][debug:025]: Free Heap Size: 166252 bytes
[14:50:32][D][debug:053]: Flash Chip: Size=4096kB Speed=40MHz Mode=DIO
[14:50:32][D][debug:086]: Chip: Model=ESP32, Features=WIFI_BGN,BLE,BT, Cores=2, Revision=1
[14:50:32][D][debug:088]: ESP-IDF Version: v3.2.3-14-gd3e562907
[14:50:32][D][debug:091]: EFuse MAC: 00008C21C8D108F0
[14:50:32][D][debug:143]: Reset Reason: Power On Reset
[14:50:32][D][debug:186]: Wakeup Reason: Unknown
[14:50:53][D][api.connection:604]: Client 'Home Assistant 0.117.4 (192.168.2.12)' connected successfully!
[14:50:58][D][api:067]: Disconnecting Home Assistant 0.117.4 (192.168.2.12)
[14:50:58][D][api.connection:604]: Client 'Home Assistant 0.117.4 (192.168.2.12)' connected successfully!
[14:51:09]E (42466) rmt: RMT[2] ERR
[14:51:09]E (42466) rmt: status: 0x13080140
[14:52:27][I][ota:046]: Boot seems successful, resetting boot loop counter.
```

glmnet, 2020-11-10:
yeah, may be it is related to the RMT which is a ESP32 only thingy, just in case did you try on ESP8266?

ogry, 2020-11-10:
No have any 8266 around... sorry. 

BTW, even with this error, the device works fine (when it works ;-) and receives and sends correctly RFCodes...

I'm couriuos abot the 40Mhz speed reported by debug. I assumed he would run at the promised 240Mhz.

glmnet, 2020-11-10:
> I'm couriuos abot the 40Mhz speed reported by debug. I assumed he would run at the promised 240Mhz.

thats the flash memory chip speed, not the core speed, it's normal.

timovos, 2020-11-21:
Same here (esp32doit-devkit-v1).

Adding one simple text_sensor results in error (remove it and it works):
`[18:57:27][D][wifi:304]: Starting scan...
[18:57:27]***ERROR*** A stack overflow in task wifi has been detected.
[18:57:27]abort() was called at PC 0x4008caf8 on core 0
[18:57:27]
[18:57:27]Backtrace: 0x4008c8b0:0x3ffdb810 0x4008cae1:0x3ffdb830 0x4008caf8:0x3ffdb850 0x40089c08:0x3ffdb870 0x4008b828:0x3ffdb890 0x4008b7de:0x3ffe440c
WARNING Found stack trace! Trying to decode it
WARNING Decoded 0x4008c8b0: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
WARNING Decoded 0x4008cae1: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
WARNING Decoded 0x4008caf8: vApplicationStackOverflowHook at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
WARNING Decoded 0x40089c08: vTaskSwitchContext at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3507
WARNING Decoded 0x4008b828: _frxt_dispatch at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/portasm.S:406
WARNING Decoded 0x4008b7de: _frxt_int_exit at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/portasm.S:206`

closed 2021-04-04 by stale[bot]