esphome/issues #1607: Unexpected error when compiling

hanneshier, 2020-11-08:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

macOS via pip3
**ESP (ESP32/ESP8266, Board/Sonoff):**

Several Boards, tried NodeMcu and gerneric
**ESPHome version (latest production, beta, dev branch)**

1.15.3

**Affected component:**

Several Components

**Description of problem:**
esphome crashes while compiling. See the logs for more information. This problem started yesterday all of a sudden, I don't know what has changed in my setup. It worked completely fine in one moment, but started crashing with all my Nodes in the next.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: dooropener
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: "MyWifif"
  password: "myPassword"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Dooropener Fallback Hotspot"
    password: "mzJHWmWvXAOr"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "xxx"

ota:
  password: "xxx"
```

**Logs (if applicable):**

```
INFO Reading configuration Documents/Programming/esphome/dooropener.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d Documents/Programming/esphome/dooropener
********************************************************************************
Obsolete PIO Core v4.3.4 is used (previous was 5.0.2)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/page/faq.html#multiple-pio-cores-in-a-system
********************************************************************************
Processing dooropener (board: d1_mini; framework: arduino; platform: espressif8266@2.6.2)
--------------------------------------------------------------------------------
Error: Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/platformio/__main__.py", line 109, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "/usr/local/lib/python3.9/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.9/site-packages/platformio/commands/__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "/usr/local/lib/python3.9/site-packages/click/core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.9/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.9/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/click/decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/platformio/commands/run/command.py", line 128, in cli
    process_env(
  File "/usr/local/lib/python3.9/site-packages/platformio/commands/run/command.py", line 170, in process_env
    result = {"env": name, "duration": time(), "succeeded": ep.process()}
  File "/usr/local/lib/python3.9/site-packages/platformio/commands/run/processor.py", line 79, in process
    result = p.run(build_vars, build_targets, self.silent, self.verbose, self.jobs)
  File "/usr/local/lib/python3.9/site-packages/platformio/managers/platform.py", line 414, in run
    self.install_packages(silent=True)
  File "/usr/local/lib/python3.9/site-packages/platformio/managers/platform.py", line 307, in install_packages
    self.pm.install(name, version, silent=silent, force=force)
  File "/usr/local/lib/python3.9/site-packages/platformio/managers/package.py", line 672, in install
    package_dir = self.get_package_dir(name, requirements, url)
  File "/usr/local/lib/python3.9/site-packages/platformio/managers/package.py", line 415, in get_package_dir
    manifest = self.get_package(name, requirements, url)
  File "/usr/local/lib/python3.9/site-packages/platformio/managers/package.py", line 384, in get_package
    for manifest in self.get_installed():
  File "/usr/local/lib/python3.9/site-packages/platformio/managers/package.py", line 374, in get_installed
    manifest = self.load_manifest(pkg_dir)
  File "/usr/local/lib/python3.9/site-packages/platformio/managers/package.py", line 347, in load_manifest
    manifest = ManifestParserFactory.new_from_file(manifest_path).as_dict()
  File "/usr/local/lib/python3.9/site-packages/platformio/package/manifest/parser.py", line 73, in new_from_file
    return ManifestParserFactory.new(
  File "/usr/local/lib/python3.9/site-packages/platformio/package/manifest/parser.py", line 123, in new
    return cls(contents, remote_url, package_dir)
  File "/usr/local/lib/python3.9/site-packages/platformio/package/manifest/parser.py", line 136, in __init__
    self._data = self.normalize_repository(self._data)
  File "/usr/local/lib/python3.9/site-packages/platformio/package/manifest/parser.py", line 174, in normalize_repository
    url = (data.get("repository") or {}).get("url")
AttributeError: 'str' object has no attribute 'get'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

```

**Additional information and things you've tried:**
I have reinstalled esphome and platformio

hanneshier, 2020-11-08:
This problem does not occur in the dev branch!

closed 2020-11-08 by hanneshier

jumika, 2020-11-12:
For me it was caused by a bad dependency. Running pip3 install esphome installed platformio 4.3.4.
After manually updating platformio to 5.0.2 everything works fine.
Altough at the end of installing platformio 5.0.2 i got a message saying this is not compatible with the current esphome version :S