esphome/issues #1609: Logger is broken in 1.16.0-dev

Anonym-tsk, 2020-11-08:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266, ESP32 (Wemos D1 Mini, Sonoff Basic, NodeMCU v3, Wroom ESP32 and others)

**ESPHome version (latest production, beta, dev branch)**

1.16.0-dev

**Affected component:**

logger

**Description of problem:**
There are no logs in dashboard.
```
INFO Reading configuration /config/co2mon.yaml...
INFO Starting log output from co2mon.local using esphome API
INFO Connecting to co2mon.local:6053 (192.168.0.18)
INFO Successfully connected to co2mon.local
```

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: co2mon
  platform: ESP8266
  board: d1_mini
  comment: Masterkit MT8057

sensor:
  - platform: zyaura
    clock_pin: D1
    data_pin: D2
    co2:
      name: "CO2Mon CO2"
      id: co2ppm
    temperature:
      name: "CO2Mon Temperature"
      id: temperature
    update_interval: 20s

status_led:
  pin:
    number: D4
    inverted: true

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: on
  ap:
    ssid: "CO2mon"
    password: "1234567890"

logger:

api:
  password: "esphome"
  reboot_timeout: 0s

ota:
  password: "esphome"
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**
Sometimes i can see log after few minutes. But very rare.

randybb, 2020-11-08:
Ok, so this weird behavior is not only on my site.

wisesokol, 2020-11-08:
i have this problem too.

lukasdietz, 2020-11-09:
Hey there, I have the same problem. After a downgrade to 1.15.3 everything is working like normal except for the xiaomi_ble part I want to use.

JoeyPoortman, 2020-11-09:
I am still on 1.15.3 and none of my devices that are powering up after ? yesterday i guess ? are connecting through the api?

glmnet, 2020-11-09:
Seems like a dashboard only issue. I can access logs via api from cli

JoeyPoortman, 2020-11-10:
My last 3 lights can't be controlled through HA because the esp doesn't connect to the api.

randybb, 2020-11-10:
@JoeyPoortman  But your problem is not related to the web logger. I don't have any problems with API connection at all, just the web logging is really slow.

glmnet, 2020-11-10:
It'd be nice if somebody could [bisect ](https://en.wikipedia.org/wiki/Bisection_(software_engineering)) this issue

glmnet, 2020-11-10:
May be this one? https://github.com/esphome/esphome/pull/1323

JoeyPoortman, 2020-11-10:
> @JoeyPoortman But your problem is not related to the web logger. I don't have any problems with API connection at all, just the web logging is really slow.

My web logger does nothing because it doesn't connect to HA ..

I have reinstalled Esphome add on, and removed all temp files in the folder from Esphome. Most things work now, but my last light won't connect, it only shows it is connected to the ip and thats it.

glmnet, 2020-11-10:
> May be this one? https://github.com/esphome/esphome/pull/1323

This is the regression commit. Question now is why

closed 2020-11-19 by glmnet