esphome/issues #1610: Needless severity order check and/or wrong order in LOG_LEVEL_SEVERITY array

clssn, 2020-11-08:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

any (root cause is independent of this, see below)

**ESP (ESP32/ESP8266, Board/Sonoff):**

any (root cause is independent of this, see below)

**ESPHome version (latest production, beta, dev branch)**

1.16.0-dev

**Affected component:**

https://esphome.io/components/logger.html#manual-tag-specific-log-levels

**Description of problem:**
During development of a component I want to generally log high-severity incidents (like WARNING), at the same time for my component I want to select a lower-severity, higher-verbosity level, like VERY_VERBOSE. From the documentation it looked like, this would be possible with the `logs` section of the `logger` config, but turns out, you can only select a lower-severity (higher verbosity) level as a general log level, and for specific components you can override this with high-severity (low verbosity) levels.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
logger:
  level: INFO
  logs:
    time_based.cover: VERY_VERBOSE
```

**Logs (if applicable):**

```
$ esphome mydevice.yaml logs
INFO Reading configuration mydevice.yaml...
ERROR Error while reading config: The local log level VERY_VERBOSE for time_based.cover must be less severe than the global log level INFO.
```

**Additional information and things you've tried:**

I think the term severity (German: *Schweregrad*) is wrongly interpreted in this implementation. For example an `ERROR` is a more severe incident than a `WARNING`. But the  `LOG_LEVEL_SEVERITY` (https://github.com/esphome/esphome/blob/dev/esphome/components/logger/__init__.py#L32) orders them high severity first. If the semantics of the array is actually severity, a higher index should imo indicate a higher severity, so that comparisons between severity variables can be naturally read.

In https://github.com/esphome/esphome/blob/dev/esphome/components/logger/__init__.py#L66 the order of severities has been constrained so that the general log severity needs to be lower (=more verbose) than the component specific ones defined in the `logs` section.

While I can imagine a use-case for this order (e.g. to silence individual "noisy" components while leaving the general verbosity higher), I think it shouldn't be constrained to this order as my requested use-case (generally have a low verbosity and choose a more verbose level for my component under development - to focus on the developed component) is valid as well.

Proposed solution: Remove the check for the order.
Optional: Fix (reverse) the `LOG_LEVEL_SEVERITY` array and change all uses in log level comparisons (only one present afaik).

clssn, 2020-11-08:
@napieraj, @OttoWinter, or someone of the owners, can you please re-open this ticket? I put quite some effort into explaining and the missing-template bot always closes my issues. I really want to contribute fixes and features to this great project, so please let me :-)

At the first attempt (#1608) I didn't get the template to fill in, so I thought there is none. Starting this issue, again, I didn't get a template to fill in, so I took it from the .github directory - and filled it in. Still no luck. Any clue what's wrong with github issues??

napieraj, 2020-11-08:
are you including the comment lines from the template?

the following at the beginning:
```
<!-- Thanks for reporting a bug for this project. READ THIS FIRST:

- Provide as many details as possible. Simply saying "X gives bug" or "X gives error" is not enough!
- Paste logs, configuration sample and code into the backticks (```).
- Read through the template carefully and fill out all missing details.
- Please also search for similar issues in this issue tracker first and read through the ESPHome FAQ.

DO NOT DELETE ANY TEXT from this template! Otherwise the issue may be closed without a comment.
-->
```
and this at the end:
``````

jesserockz, 2020-11-08:
I just added that line into the end

clssn, 2020-11-08:
Thanks a lot, @napieraj and @jesserockz!

clssn, 2020-11-08:
Well, really sorry for spamming here. Reading the docs to the end answered to me, why it is not possible to set individual log levels with the logs config. The global log level makes all lower severity messages being filtered out early. And then you can only constrain individual components further to log only higher severity. If it's like that for performance reasons, I won't have a chance to change this.
No prob. I'll grep locally...

closed 2020-11-08 by clssn