esphome/issues #1612: Hold state of GIPIO LOW during deep sleep mode

ribbles, 2020-11-09:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 TTGO T-Display

**ESPHome version (latest production, beta, dev branch)**

1.15.3

**Affected component:**

`deep_sleep`, `switch` & sensor `pmsx003`

**Description of problem:**
I would like to put the pmsx003 sensor to sleep when I put the ESP32 to sleep, to save batteries. I can put the pmsx003 to sleep, but when the ESP32 goes to sleep, the pmsx003 sensor turns back on, presumably because the PIN has been reverted to open. I used an RTC pin under the impression that they remain set during deep sleep. 

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: ttgo_sleep_bug
  platform: ESP32
  board: ttgo-lora32-v1

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      inverted: true
      mode: INPUT_PULLUP
    name: "${upper_devicename} Button S1"
    id: button0
    on_press:
      then:
        - switch.turn_off: pmsx003_sleep
        # so we can hear that it goes to sleep
        - delay: 5s
        - deep_sleep.enter: deep_sleep_off
    
    
switch:
  - platform: gpio
    pin: GPIO27
    name: "pmsx003 Sleep"
    id: pmsx003_sleep
    restore_mode: ALWAYS_ON

deep_sleep:
  id: deep_sleep_off

```

**Logs (if applicable):**

```
[19:00:46][I][app:105]: ESPHome version 1.15.3 compiled on Nov  8 2020, 18:58:23
[19:00:46][C][wifi:443]: WiFi:
[19:00:46][C][wifi:303]:   SSID: 

[redacted]
[19:00:46][C][wifi:304]:   IP Address: 10.0.0.140
[19:00:46][C][wifi:306]:   BSSID: 

[redacted]
[19:00:46][C][wifi:307]:   Hostname: 'ttgo_sleep_bug'
[19:00:46][C][wifi:311]:   Signal strength: -59 dB ▂▄▆█
[19:00:46][C][wifi:315]:   Channel: 11
[19:00:46][C][wifi:316]:   Subnet: 255.255.255.0
[19:00:46][C][wifi:317]:   Gateway: 10.0.0.1
[19:00:46][C][wifi:318]:   DNS1: 75.75.75.75
[19:00:46][C][wifi:319]:   DNS2: 75.75.76.76
[19:00:46][C][gpio.binary_sensor:015]: GPIO Binary Sensor '${upper_devicename} Button S1'
[19:00:46][C][gpio.binary_sensor:016]:   Pin: GPIO0 (Mode: INPUT_PULLUP, INVERTED)
[19:00:46][C][switch.gpio:042]: GPIO Switch 'pmsx003 Sleep'
[19:00:46][C][switch.gpio:043]:   Pin: GPIO27 (Mode: OUTPUT)
[19:00:46][C][switch.gpio:059]:   Restore Mode: Always ON
[19:00:46][C][logger:185]: Logger:
[19:00:46][C][logger:186]:   Level: DEBUG
[19:00:46][C][logger:187]:   Log Baud Rate: 115200
[19:00:46][C][logger:188]:   Hardware UART: UART0
[19:00:46][C][captive_portal:169]: Captive Portal:
[19:00:46][C][ota:029]: Over-The-Air Updates:
[19:00:46][C][ota:030]:   Address: ttgo_sleep_bug.local:3232
[19:00:46][C][api:095]: API Server:
[19:00:46][C][api:096]:   Address: ttgo_sleep_bug.local:6053
[19:00:46][C][deep_sleep:020]: Setting up Deep Sleep...
[19:00:56][D][api.connection:604]: Client 'Home Assistant 0.117.5 (10.0.0.168)' connected successfully!
[19:01:05][D][binary_sensor:036]: '${upper_devicename} Button S1': Sending state ON
[19:01:05][D][switch:025]: 'pmsx003 Sleep' Turning OFF.
[19:01:05][D][switch:045]: 'pmsx003 Sleep': Sending state OFF
[19:01:05][D][binary_sensor:036]: '${upper_devicename} Button S1': Sending state OFF
[19:01:10][I][deep_sleep:067]: Beginning Deep Sleep

```

**Additional information and things you've tried:**

rradar, 2020-11-14:
I'm not entirely sure but I think this can only achieved by hardware design.

ribbles, 2020-11-17:
Noob issue. 10k resistor to earth. Resolved.

closed 2020-11-17 by ribbles

adiman29, 2020-11-22:
i have the same problem and 10k resistor not working.
I using wemos d1 mini

dan-glass, 2020-11-23:
> i have the same problem and 10k resistor not working.
> 
> I using wemos d1 mini

I started at 10k, but ended up needing a 5k1 to got the voltage low enough. Try using a volt meter?