esphome/issues #1616: Multiple wi-fi networks change doesn't work

Spirituss, 2020-11-10:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant Core/Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**

Nodemcu, Blitzwolf plugs, Wemos D1 mini, ...  

**ESPHome version (latest production, beta, dev branch)**

1.16.0-dev

**Affected component:**

https://esphome.io/components/wifi.html?highlight=priority#wifi-networks

**Description of problem:**
Multiple wi-fi networks were defined in the firmware, all with different priority tags. In case the highest priority network in not available, ESP doesn't change the network to the next one with lower priority. ESP remains unavailable for HA, blinking by status led, which tells that there is no wi-fi connection.

**Problem-relevant YAML-configuration entries:**
```
wifi:  
  reboot_timeout: 0s      
  output_power: 17.5dB    
  networks:  
    - ssid: !secret wifi_ssid_1
      password: !secret wifi_pass_1
      hidden: true
      priority: 2.0   
    - ssid: !secret wifi_ssid_2
      password: !secret wifi_pass_2 
      hidden: true
      priority: 1.0      
    - ssid: !secret wifi_ssid_3
      password: !secret wifi_pass_3
      hidden: true
      priority: 0.0  
```

**Logs (if applicable):**

```
No log available since there is no connection to ESP at that moment.
```

**Additional information and things you've tried:**

Spirituss, 2021-03-10:
The problem still is actual. Such a configuration as above mentioned DOES'T work while should be according to esphome.io offsite.

randybb, 2021-03-10:
enable SSID broadcasting and it will work

Spirituss, 2021-03-11:
> enable SSID broadcasting and it will work

I don't ask how to use SSID, the issue is that the configuration above mentioned doesn't works, while offsite tells that ssid is the OPTIONAL additional filter. Either documentation should be updated or system behaviour be changed.
Moreover, "hidden" option doesn't matter for this problem.

Spirituss, 2021-06-03:
I have tested and found that in case bssid broadcasting is used the priority works incorrectly. In case the highest priority bssid network becomes unavailable, esphome switches to the lower priority network. But it doesn't **switch** back to the higher bssid network when is becomes available again.