esphome/issues #1618: Cant reconnect OTA after flashing via USB, but can connect with USB

hughesba, 2020-11-10:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant Core/Docker
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32-Cam dev module with OV2640 camera
**ESPHome version (latest production, beta, dev branch)**

ESP Home Version 1.15.3 and latest firmware 1.15.3 (I think, this is the version another ESP32S node is running, and both were updated at the same time). No firmware updates are pending.
**Affected component:**

https://esphome.io/guides/getting_started_hassio.html

**Description of problem:**
I recently had to re-flash a working ESP32-Cam dev module with OV2640 camera via USB due to the need to repartition (enabling Bluetooth LE). Up till then, everything was working fine, and the only change to the config was the addition of the "esp32_ble_tracker:" line. I have recently done the same procedure for an ESP32S with no issues, and my other ESP nodes are still working fine.

Now ESP Home won't recognise the node ESP-CAM. It responds to ping and port 3232 also responds. When I show logs, via OTA, it says 
[Errno 111] (Connection Refused), with the correct IP address for the node.

When I connect via USB and show logs, everything looks OK. It is capturing images every 10 secs as per the config file, and it is picking up ble tracker devices. It also says boot successful, resetting boot loop counter.

But I still can't connect via OTA. I have tried for a few days now - reflashing and deleting the device out of ESPHome and removing devices from Home Assistant, renaming the device, restarting ESPHome and Home Assistant etc. I have tried these several times and cleared out the relevant files for this node. 

Any help would be greatly appreciated.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE
esphome:
  name: cam
  platform: ESP32
  board: esp32cam

wifi:
  ssid: "My Wi-Fi"
  password: !secret wifi_password
  # Optional manual IP
  manual_ip:
    static_ip: x.y.z.34
    gateway: x.y.z.254
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32Cam Fallback Hotspot"
    password: "redacted"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

esp32_ble_tracker:

esp32_camera:
  name: Doorbell Camera
  external_clock:
    pin: GPIO0
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32
  resolution: 640x480
  jpeg_quality: 10 
# Flashlight CHECK PIN
output:
   - platform: gpio
     pin: GPIO4
     id: gpio_4 

light:
  - platform: binary
    output: gpio_4
    name: esp32cam light  

# The following can be omitted 
#  - platform: restart
#    name: esp32cam restart 

sensor:
  - platform: wifi_signal
    name: esp32cam wifi signal
    update_interval: 600s
 
  - platform: uptime
    name: esp32cam uptime
    
switch:
  - platform: restart
    name: "ESP32Cam Restart"
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```
INFO Reading configuration /config/esphome/esp32cam.yaml...
INFO Starting log output from x.y.z.34 using esphome API
INFO Connecting to x.y.z.34:6053 (x.y.z.34)
WARNING Initial connection failed. The ESP might not be connected to WiFi yet (Error connecting to x.y.z.34: [Errno 111] Connection refused). Re-Trying in 1 seconds

**Additional information and things you've tried:**
Reflashing via USB, deleting the device out of ESPHome and removing devices from Home Assistant, renaming the device, restarting ESPHome and Home Assistant etc. I have tried these several times and cleared out the relevant config files for this node from /config/esphome folder.

mcondren, 2020-11-11:
Same issue with 1.15.3.  pushed an upload via OTA and now when I try to connect with logging I get connection refused.

mcondren, 2020-11-11:
Rolling back to 1.15.1 resolved the issue.

hughesba, 2020-11-11:
> Rolling back to 1.15.1 resolved the issue.

Thanks @mcondren. I rolled back to v1.15.1 but still having the issue. I notice that port 6053 (the API port) is down on the node

hughesba, 2020-11-13:
Build another instance of HA and found same problem, so hardware or config issue. Tried change of board type in config and removed new configs. Issue resolved.

closed 2020-11-13 by hughesba