esphome/issues #1620: Calling servo.detach once on ESP32 renders further commands effectless - is this intended?

RAYs3T, 2020-11-11:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant on ProxMox

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 D1 mini

**ESPHome version (latest production, beta, dev branch)**

v1.15.3 (via Home Assistant)

**Affected component:**

https://esphome.io/components/servo.html

**Description of problem:**
So the idea is to detach the servo once the movement is done to avoid permanent micro adjustments.
Once you call `detach` on a servo, further commands can no longer control the servo.

Is this intended behavior? My expectation would be that the next time you call the `servo.write` it would attach again.

**Problem-relevant YAML-configuration entries:**
```yaml
      then:
        - logger.log: "Start servo change"
        - servo.write:
            id: ob_tilt
            level: !lambda 'return level / 100.0;'
        - delay: 2s
        - logger.log: "Done"
        - servo.detach: ob_tilt
         - logger.log: "Detached"

```

**Logs (if applicable):**

```

```

**Additional information and things you've tried:**
Once you remove the `detach` all further calls work fine.

ialex87, 2021-03-01:
Subscribing to this question/bug, I'm experiencing the same issue.  Is this intended ?

glmnet, 2021-03-01:
@ialex87 this is an old issue, which version are you running?

glmnet, 2021-03-01:
also, which device (esp32 / esp8266)

ialex87, 2021-03-01:
Ah sorry for not post this data, 
Wemos D1 mini ( esp8266) 
ESPhome 1.15.3 via Home Assistant

glmnet, 2021-03-01:
Can you update esphome? There had been updates on the servo. Specially an interesting one which auto detaches the servo after a time

ialex87, 2021-03-01:
Oh that's odd, It didn't show me there were updates for esphome, now I see I don't have the latest version. Thanks for heads up. I will give it a try. thanks!

closed 2021-03-29 by glmnet