esphome/issues #1621: Improving doc on contributing

uherting, 2020-11-11:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

any

**ESP (ESP32/ESP8266, Board/Sonoff):**

does not apply

**ESPHome version (latest production, beta, dev branch)**

ESPHome docs --- online version at ESPHome.io, URL: https://esphome.io/guides/contributing.html

**Affected component:**

Documentation on how to contribute to the ESPHome project as member of the esphome team / "ordinary" github user in regards of access rights to the esphome/esphome... repositories at GitHub.com

**Description of problem:**
I read the contribution page at ESPHome.io (URL see above) and from reading the _entire_ two sections of  I understood the following:
* https://esphome.io/guides/contributing.html#setting-up-development-environment is for project members (people able to commit / push to esphome/esphome) 
* https://esphome.io/guides/contributing.html#setting-up-git-environment is for people who are not project members but willing to contribute (forking and creating PRs)

I think the documentation should be more foward by addressing the access right topic in a way similar to this:
 if you are a member of the esphome team then do this to contribute: ....
 if you are an ordinary github user willing to contribute then do this: ...

**Problem-relevant YAML-configuration entries:**
```yaml
YAML does not apply in this issue

```

**Logs (if applicable):**

```
DEBUG LOG does not apply in this issue

```

**Additional information and things you've tried:**
I understood eventually what I as a github user have to do in order to contribute by reading and wondering what the information might mean. For the sake of any new contributors I would warmly welcome a straight forward approch aka "if this thatn that" if possible. Thank you.

rradar, 2020-11-14:
@uherting if you would be so kind send a pull request for review with the changes to the contributing docs you suggest :smiley:

uherting, 2020-11-15:
@rradar I am in the IT business for a wee while now and instead of selling my half hearted knowledge I would rather rely on people who are actively developing ESPHome to do a documentation which equals the product in the same way. ;-) 

I will happily document the extensions to ESPHome in  way I am used to as a professional in order that the extensions might be of help to anyone willing to use them.

closed 2021-03-15 by uherting