esphome/issues #1623: ESPHome (the addon) on Home Assistant doesn't work - fails at compile/upload

lancelon, 2020-11-12:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant 
**ESP (ESP32/ESP8266, Board/Sonoff):**
ESP32-POE

**ESPHome version (latest production, beta, dev branch)**
 version: 1.15.3

**Affected component:**

https://esphome.io/index.html

**Description of problem:**
[Looks like a fairly common bug](https://www.google.com/search?ei=63OtX6qfEpaP8gLhjJj4AQ&q=home+assistant+%22xtensa-esp32-elf%22+esphome&oq=home+assistant+%22xtensa-esp32-elf%22+esphome)

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
INFO Reading configuration /config/esphome/traffic.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/traffic
Processing traffic (board: esp32-poe; framework: arduino; platform: espressif32@1.12.4)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif32 3.10004.200129 (1.0.4) 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - toolchain-xtensa32 2.50200.80 (5.2.0)
Library Manager: Installing Hash
Library Manager: Already installed, built-in library
Dependency Graph
|-- <AsyncTCP-esphome> 1.1.1
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <FS> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.7
|   |-- <AsyncTCP-esphome> 1.1.1
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|-- <DNSServer> 1.1.0
|   |-- <WiFi> 1.0
|-- <Update> 1.0
|-- <WiFi> 1.0
Compiling /data/traffic/.pioenvs/traffic/src/esphome/components/api/api_connection.cpp.o
Compiling /data/traffic/.pioenvs/traffic/src/esphome/components/api/api_pb2.cpp.o
Compiling /data/traffic/.pioenvs/traffic/src/esphome/components/api/api_pb2_service.cpp.o
Compiling /data/traffic/.pioenvs/traffic/src/esphome/components/api/api_server.cpp.o
In file included from /root/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/random:42:0,
                 from /root/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/stl_algo.h:66,
                 from /root/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/algorithm:62,
                 from /root/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:142,
                 from src/esphome/core/component.h:5,
                 from src/esphome/components/api/api_connection.h:3,
                 from src/esphome/components/api/api_connection.cpp:1:
/root/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/limits:1598:7: internal compiler error: Segmentation fault
       min() _GLIBCXX_USE_NOEXCEPT { return __FLT_MIN__; }
       ^
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
*** [/data/traffic/.pioenvs/traffic/src/esphome/components/api/api_connection.cpp.o] Error 1
In file included from /root/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/random:42:0,
                 from /root/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/stl_algo.h:66,
                 from /root/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/algorithm:62,
                 from /root/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:142,
                 from src/esphome/core/component.h:5,
                 from src/esphome/components/api/proto.h:3,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_pb2.cpp:3:
/root/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/limits:1598:7: internal compiler error: Segmentation fault
       min() _GLIBCXX_USE_NOEXCEPT { return __FLT_MIN__; }
       ^
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
*** [/data/traffic/.pioenvs/traffic/src/esphome/components/api/api_pb2.cpp.o] Error 1
In file included from /root/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/random:42:0,
                 from /root/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/stl_algo.h:66,
                 from /root/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/algorithm:62,
                 from /root/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:142,
                 from src/esphome/core/component.h:5,
                 from src/esphome/components/api/proto.h:3,
                 from src/esphome/components/api/api_pb2.h:5,
                 from src/esphome/components/api/api_pb2_service.h:5,
                 from src/esphome/components/api/api_pb2_service.cpp:3:
/root/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/limits:1598:7: internal compiler error: Segmentation fault
       min() _GLIBCXX_USE_NOEXCEPT { return __FLT_MIN__; }
       ^
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
*** [/data/traffic/.pioenvs/traffic/src/esphome/components/api/api_pb2_service.cpp.o] Error 1
In file included from /root/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/random:42:0,
                 from /root/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/bits/stl_algo.h:66,
                 from /root/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/algorithm:62,
                 from /root/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:142,
                 from src/esphome/core/component.h:5,
                 from src/esphome/components/api/api_server.h:3,
                 from src/esphome/components/api/api_server.cpp:1:
/root/.platformio/packages/toolchain-xtensa32/xtensa-esp32-elf/include/c++/5.2.0/limits:1598:7: internal compiler error: Segmentation fault
       min() _GLIBCXX_USE_NOEXCEPT { return __FLT_MIN__; }
       ^
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
*** [/data/traffic/.pioenvs/traffic/src/esphome/components/api/api_server.cpp.o] Error 1
========================= [FAILED] Took 15.45 seconds =========================
```

**Additional information and things you've tried:**
Have tried running dev mode

randybb, 2020-11-12:
Issue where compiler is not available on unsupported platforms that is shown as "not found"  and your "segmentation fault" is not exactly the same.

Could you please show us your yaml file?

lancelon, 2020-11-12:
Sorry for delay. As you can see. Extremely vanilla.

```
esphome:
  name: traffic
  platform: ESP32
  board: esp32-poe

wifi:
  ssid: "[ssid]"
  password: "[x]"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Traffic Fallback Hotspot"
    password: "[x]"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "[x]"

ota:
  password: "[x]"
```

rradar, 2020-11-13:
> Operating environment/Installation (Hass.io/Docker/pip/etc.):
> 
> Home Assistant
> 
To fill out the presented template **completely** is always good. You will get more likely help and users don't need to ask for yaml/os/etc. with a new post.

If I remember right there were some one also failing with the xtensa package on aarch64 and bsd

lancelon, 2020-11-14:
updated more info about my HA instance

closed 2021-03-29 by stale[bot]