esphome/issues #1625: v1.14.5 falsely claims esp32-poe doesn't exist then refuses to upgrade

lancelon, 2020-11-13:
Same issue as #1427 with the added bug the upgrade-all doesn't work either

**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Mac OS X 
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32-POE
**ESPHome version (latest production, beta, dev branch)**

v1.14.5 
**Affected component:**

Wizard

**Description of problem:**
(board): esp32-poe      	
Sorry, I don't think the board "{}" exists.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```
$ esphome update-all
ERROR Missing configuration parameter, see esphome --help.
**Additional information and things you've tried:**

Going to try to upgrade to v1.15.0.

lancelon, 2020-11-13:
"Going to try to upgrade to v1.15.0."

That fixed it. Onwards

rradar, 2020-11-14:
Best is always to use latest stable release (at this moment 1.15.3) and if it doesn't work out than create a issue providing the full scope including your yaml & logs!

closed 2020-11-14 by rradar