esphome/issues #1632: ac_dimmer component doesn't dim correctly using RobotDyn AC dimmer

xanderdekok, 2020-11-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESPWROOM32
**ESPHome version (latest production, beta, dev branch)**

1.15.3
**Affected component:**

https://esphome.io/components/output/ac_dimmer.html?highlight=dimmer

**Description of problem:**
When using the ac_dimmer component in combination wit the RobotDyn AC dimmer the lamp has an irregular flicker. I use a dimmable LED lamp. Also a normal light bulb doesn't work. I live in the Netherlands so we have 230v at 50hz.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: esp_test
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: "***"
  password: "***"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp Test Fallback Hotspot"
    password: "***"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "ESPOTA&API"

ota:
  password: "ESPOTA&API"

# Enable Bluetooth scanning for this ESP32
# esp32_ble_tracker:

output:
  - platform: ac_dimmer
    id: dimmer1
    gate_pin: 22
    zero_cross_pin:
      number: 23
      mode: INPUT
      inverted: yes
    method: leading
    init_with_half_cycle: False
light:
  - platform: monochromatic
    output: dimmer1
    name: Dimmer test
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**
I have tried all the possible setting for the ac_dimmer component, but the shown configuration works best with the dimmable LED lamp. The normal light bulp has an even worse irregular flicker than the LED lamp using this configuration.

glmnet, 2020-11-16:
Well this came in a good time, as I just got a RobotDyn dimmer from @Svrbinek I'm just reproducing this issue, it seems a software issue indeed, I noticed the ZC spike is not very "square" and is more like a peak, other than that is not much different from my original test device.
I had to setup the ZC pin as INPUT_PULLUP (not just INPUT) otherwise the interrupt routine was missing some ZC events, use with default leading method. This kind of works, however not perfect.
Anyway the code is producing different times calculation and this causes flickering of course. Will have to connect the logic analyzer and add some debug pins there to see whats going on.

pterodaktil02, 2020-11-19:
Same problem 

**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**

`esphome:
  name: test_dimmer
  platform: ESP32
  board: esp32doit-devkit-v1`
**ESPHome version (latest production, beta, dev branch)**

ESPHome
Current version: 1.15.3
**Affected component:**

https://esphome.io/components/output/ac_dimmer.html

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```
output:
  - platform: ac_dimmer
    id: dimmer_light
    gate_pin: GPIO18
    zero_cross_pin:
      number: GPIO19
      mode: INPUT
      inverted: yes

light:
  - platform: monochromatic
    output: dimmer_light
    name: Dimmerized Light
    
sensor:
   - platform: duty_cycle
    pin:  GPIO21
    name:  GPIO21 
    internal: true 
    update_interval: 5s
  - platform: duty_cycle
    internal: true   
    pin:  GPIO18
    name:  GPIO18
    update_interval: 5s  

```

**Logs (if applicable):**

```
[08:29:56][D][duty_cycle:036]: 'GPIO21' Got duty cycle=0.0%
[08:29:56][D][sensor:092]: 'GPIO21': Sending state 0.00000 % with 1 decimals of accuracy
[08:29:56][D][duty_cycle:036]: 'GPIO18' Got duty cycle=100.0%
[08:29:56][D][sensor:092]: 'GPIO18': Sending state 100.00000 % with 1 decimals of accuracy

```

**Additional information and things you've tried:**
After  turn on Dimmerized Light  i can not  control it - it always   on with 100% power

glmnet, 2020-11-19:
Please test with mode INPUT_PULLUP

pterodaktil02, 2020-11-19:
> Please test with mode INPUT_PULLUP

It' work  but  we have  filcker   on  ordinaly lamp. is ther  way  change  frecudency of  cut

xanderdekok, 2020-11-19:
> Please test with mode INPUT_PULLUP

Also tested with INPUT_PULLUP. Unfortunately no difference...

pterodaktil02, 2020-11-20:
> > Please test with mode INPUT_PULLUP
> 
> Also tested with INPUT_PULLUP. Unfortunately no difference...

try   change pins

i use  this config

`output:
  - platform: ac_dimmer
    id: dimmer_light
    gate_pin: GPIO21
    zero_cross_pin:
      number: GPIO19
      mode: INPUT_PULLUP
      inverted: yes`

xanderdekok, 2020-11-27:
> > > Please test with mode INPUT_PULLUP
> > 
> > 
> > Also tested with INPUT_PULLUP. Unfortunately no difference...
> 
> try change pins
> 
> i use this config
> 
> `output:
> 
> * platform: ac_dimmer
>   id: dimmer_light
>   gate_pin: GPIO21
>   zero_cross_pin:
>   number: GPIO19
>   mode: INPUT_PULLUP
>   inverted: yes`

It seems to work, the lamp seems to be dimming stable now. Why does it work now?

But it has a real strange side affect. Everything is now inverted on my dashboard. So On is now Off... And 0% is full brightness, 100% is off.

mkleijn, 2020-11-30:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hass.io on PI3b+

**ESP (ESP32/ESP8266, Board/Sonoff):**
  platform: ESP32
  board: nodemcu-32s

**ESPHome version (latest production, beta, dev branch)**
Latest production (1.15.3)

I also had an issue with the RoboDyn dimmers (both with the single channel version as 4 channel version). Before switching to ESPHome I used my own code using this library: https://github.com/fabiuz7/Dimmable-Light-Arduino. The author of that library describes the same issue with the zero crossing signal not being 'digital' on the RoboDyn hardware: https://github.com/fabiuz7/Dimmable-Light-Arduino/wiki/Notes-about-dimmer-boards. This can be fixed on the hardware side by inserting a non-inverting Schmitt Trigger between the ZC signal from the optocoupler and the ESP. I ordered some SMD versions that I hoped i could solder in the jumper wires between the ESP32 and the board, but i don't trust my soldering skills too much, so until now I relied on some 'software debouncing' of the signal that the library provided. This worked reasonably well, but sometimes there was still some flickering, probably due to some spikes and surges on the net.

About two months ago I migrated to ESPHome because of the (fast) integration with Home Assistant and the fact I think this code has been tested a bit better than my own Arduino code :)

However, out of the box i wasn't able to get a smooth dimming on my (mostly) led lights.  But also a halogen bulb was flickering using the settings from the documentation. I tried all edge modes (raising, leading, leading pulse), with and without half cycle init. I eventually was able to get a reasonable result using the following settings:

` 
output:
  - platform: ac_dimmer
    id: dimmer1
    gate_pin: 16
    zero_cross_pin:
      number: 4
      mode: INPUT_PULLDOWN
    min_power: 35%

  - platform: ac_dimmer
    id: dimmer2
    gate_pin: 17
    zero_cross_pin:
      number: 4
      mode: INPUT_PULLDOWN

  - platform: ac_dimmer
    id: dimmer3
    gate_pin: 18
    zero_cross_pin:
      number: 4
      mode: INPUT_PULLDOWN

  - platform: ac_dimmer
    id: dimmer4
    gate_pin: 27
    zero_cross_pin:
      number: 4
      mode: INPUT_PULLDOWN

light:
  - platform: monochromatic
    output: dimmer1
    id: light_1
    name: Light Babyroom 
    default_transition_length: 50ms

  - platform: monochromatic
    output: dimmer2
    id: light_2
    name: Light Hallway
    default_transition_length: 50ms

  - platform: monochromatic
    output: dimmer3
    id: light_3
    name: Light Patio 
    default_transition_length: 50ms

  - platform: monochromatic
    output: dimmer4
    id: light_4
    name: Light Master bedroom 
    default_transition_length: 50ms
`

So instead of using 'inverted' and 'INPUT_PULLUP' on the zero crossing pin as advised in the documentation and this thread I use normal and INPUT_PULLDOWN. It kinda works but it doesn't sit well with me that I go against the advised settings. Probably i'm not detecting the zero crossing of the first half of the cycle, but of the second half (so 180 degree phase shift). Is that a bad thing? Or doesnt it really matter that much?

I still see some flickering with the lights at certain values, and it is also sensitive to noise on the net: I notice for instance that when my laptop adapter is plugged in without any load (so not connected to my laptop) this causes some led bulbs connected to the same phase to flicker. Other led lights using the same ESP / dimmer board have no visual flickering.

glmnet, 2020-11-30:
There is a big warning at the top of the docs saying this component is not fully tested yet https://esphome.io/components/output/ac_dimmer.html so I really don't understand why you expect to work flawlessly

Otto did the initial work here without any hardware at all, that was amazing, I took over the PR and did a substantial amount of testing but I did not have available the popular RobotDyn module, I had another one and I had very good result with that module, that's why I released this component as is now. You can find more details in the PR https://github.com/esphome/esphome/pull/880

I now have the RobotDyn so I'll test with pull down to see what happens, but as I stated before INPUT_PULLUP does works better than nothing in my setup but still has issues at some brightness levels.

mkleijn, 2020-11-30:
Sorry? Where do I state that i expect it to work flawlessly? I thought this was a good place to write down my experiences so others might be able to try my non-inverted / pulldown method. And supply the original developers with a link to another library / wiki page that describes this issue with the RoboDyn analog ZC signal and how they combated it. 

I actually wanted to drop my experience in PR 880 as I was invited by that big warning at the top of the docs you mention, but the comments there are closed so I posted it here.

Anyway, kudos to both Otto and you for the amazing work on ESPHome, and apologies if I offended you in any way. 

Out of interest, what dimmer module you had good experience with? It might be the easiest solution to swap that (if it fits my casing).

glmnet, 2020-11-30:
> and the fact I think this code has been tested a bit better than my own Arduino code :)

Maybe I took this comment specific to this code, but you meant all ESPHome, no worries I'm not offended is just that the warning is there.

Anyway, your post is indeed useful. 
In the long 880 thread there is a picture of the hardware I used for development, looks very much like the one used by the library you posted, a blue one, mine is 4 channel not 8 though.

One key difference between that library (and other libraries) I've seen around and the one in ESPHome is that the ISR is setup on the ZC pin as `RISING` whereas in ESPHome it is `FALLING` [see here](https://github.com/esphome/esphome/blob/7ff3f752e23b0c11aa2123209d7b2e296cbfdf63/esphome/components/ac_dimmer/ac_dimmer.cpp#L174) this might be related, but I need to test that better.

mkleijn, 2020-12-01:
Ah ok @glmnet I see how that line you quoted could be seen as me being sarcastic about the code maturity of this component or ESPHome in general, but that was not how I intended it. 

Interesting to hear about your tests with FALLING vs RAISING on the ZC interrupt. I'll see if I can do some testing of my own with this.

But in the end it might be impossible to get a complete flicker-free implementation without some hardware adjustments like adding a Schmitt trigger or replacing the 4N25 optocoupler by one with an integrated Schmitt trigger like the H1L11 (http://www.bristolwatch.com/ele2/zcnew.htm). 
The 'official' library supplied by RoboDyn (https://github.com/RobotDynOfficial/RBDDimmer) which I started out with using gave far worse results so maybe it's just the best possible result with these boards as they are.

Svrbinek, 2020-12-01:
Hello. If necessary, I can help with testing. I am glad that this project is moving in the right direction.

Hendrikdenhond, 2021-01-01:
I have the dimmer connected to some halogen lights (resistive load) and also experienced a lot of flickering.
I tried all combinations of code but nothing worked.
BUT I found the origin of my problem.  I was powering the esp with a external 5V powersupply soldered to the 5V and GND pin and had flickering.  When the ESP was connected over and USB cable with powerplug the flickering was gone.
So I suspect that the filtering on the powersupply was not sufficient enough and messed up the zero cross signal. Did not tried LED lamps yet.
For info: I tested on belgium powernet, 240AC 50Hz.

xanderdekok, 2021-01-01:
I did some more testing and I got the dimmer also working with a Sonoff 4CH. You can see my configuration below.
I did a few modifications to the Sonoff 4CH:

1. I soldered pin headers onto the board to flash it and to use with the dimmer for the zero cross pin
2. I soldered a wire to the GPIO 10 pin at 3rd button for the gate pin
3. I soldered 2 wires to the 4th button, to be able to use an external switch to turn on relay 4.

I did notice some side affects

1. The dimmer icon on the Lovelace dashboard is still inverted, like I also experienced before.  See my previous comment.
2. There was interference between the wires for the dimmer and the external switch. This caused button 4 to be triggered irregularly.  It also caused the dimmed lights to drop out for about 0.5s at a time at irregular intervals.

I was able to solve the interference by making sure the wires for the dimmer and the external switch are no longer close together. So apparently when using this dimmer it is prone to signal interference... 
This might be why some people are experiencing a flicker. I too had a flicker with my original setup for my ESP32 board and after switching pins it miraculously worked.

```yaml
esphome:
  name: tuinverlichting
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: "#3_NoT"
  password: "SmartThingsThatShouldNotPhoneHome"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "T Verlichting Fallback Hotspot"
    password: "xxx"

captive_portal:

# Enable Web server
web_server:
  port: 80

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "xxx"

ota:
  password: "xxx"

interval:
  - interval: 1s
    then:
      if:
        condition:
          wifi.connected:
        then:
          - output.turn_on: blue_led
        else:
          - output.turn_off: blue_led

binary_sensor:
- platform: gpio
  pin:
    number: GPIO0
    mode: INPUT_PULLUP
    inverted: True
  name: "Tuinverlichting Button 1"
  on_press:
    - switch.toggle: relay_1
- platform: gpio
  pin:
    number: GPIO9
    mode: INPUT_PULLUP
    inverted: True
  name: "Tuinverlichting Button 2"
  on_press:
    - switch.toggle: relay_2
#- platform: gpio
#  pin:
#    number: GPIO10
#    mode: INPUT_PULLUP
#    inverted: True
#  name: "Tuinverlichting Button 3"
#  on_press:
#    - switch.toggle: relay_3
- platform: gpio
  pin:
    number: GPIO14
    mode: INPUT_PULLUP
    inverted: True
  name: "Tuinverlichting Button 4"
  on_press:
    - switch.toggle: relay_4
- platform: status
  name: "Tuinverlichting Status"

switch:
- platform: gpio
  name: "Tuinverlichting Relay 1"
  pin: GPIO12
  id: relay_1
- platform: gpio
  name: "Tuinverlichting Relay 2"
  pin: GPIO5
  id: relay_2
- platform: gpio
  name: "Tuinverlichting Relay 3"
  pin: GPIO4
  id: relay_3
- platform: gpio
  name: "Tuinverlichting Relay 4"
  pin: GPIO15
  id: relay_4

output:
- platform: esp8266_pwm
  id: blue_led
  pin: GPIO13
  inverted: True

- platform: ac_dimmer
  id: dimmer_tuinverlichting
  gate_pin: GPIO10
  zero_cross_pin:
    number: GPIO2
    mode: INPUT_PULLUP
  inverted: yes

light:
  - platform: monochromatic
    output: dimmer_tuinverlichting
    name: Dimmer Veranda

```

Hendrikdenhond, 2021-01-02:
Turned out i still had a little of interference as well.  Installed a screened wire for the Zero-cross signal and now it seems to be working.

glmnet, 2021-01-02:
That's interesting. Another option could be using a stronger pull up in the zc line, i.e. a 4.7k ohm resistor to +3.3v

xanderdekok, 2021-01-02:
No idea... I'm no electrical engineer ;)

But I'm still wondering why the controls of this dimmer are inverted...???

Hendrikdenhond, 2021-01-02:
> No idea... I'm no electrical engineer ;)
> 
> But I'm still wondering why the controls of this dimmer are inverted...???

You are missing 2 spaces before inverted in your code i think....
 
This i my code:

 - platform: ac_dimmer
    id: dimmer1
    gate_pin: GPIO14
    zero_cross_pin:
      number: GPIO15
      mode: INPUT
      inverted: yes
    min_power: 0.25
    max_power: 0.9

xanderdekok, 2021-01-21:
> > No idea... I'm no electrical engineer ;)
> > But I'm still wondering why the controls of this dimmer are inverted...???
> 
> You are missing 2 spaces before inverted in your code i think....

Thanks!

Indeed those 2 spaces were the culprit. I'm not a yaml expert, but I did not expect that an alignment issue would have an effect like this. Even the validator did not show an error.

wojciechczyz, 2021-03-06:
Seems more people are using this component now, thats great. I believe there is small software problem when control is between 78% and 99%, I wonder if others observe the same and hopefully somebody with osciloscope may see how actually output of RobotDyn AC dimmer looks like.

I have used it together with RobotDyn AC dimmer to control small AC 220V 50 Hz ventilator, adding also snubber circuit ( Capacitor 0.15uF and Resistor 300 Ohm ) to protect the Triac.

 It seems it works well at 100% and then 65%-78%. 

To see if it is related to motor and the snubberr circuit I did following test:

Keeping input of 220V 50Hz I have disconnected output - ventilator and snubber circuit and still see that in range 78% - 99% small LED on RobotDyn AC dimmer board is not being lit correctly. This LED is PWM output driven. So only after change from 99% to 100% it glows much stronger. Between 65% till 99% changes do not change light intensity of the LED.

My configuration is:

output:
  - platform: ac_dimmer
    id: dimmer1
    gate_pin: GPIO33
    zero_cross_pin:
      number: GPIO34
      mode: INPUT
      inverted: yes

light:
  - platform: monochromatic
    output: dimmer1
    name: "Ventilator"

glmnet, 2021-03-06:
Check you don’t get electrical interferences. This thing switches ac with inductive loads and that’s noisy and confuses the algorithm.
Try shielding wires if possible

wojciechczyz, 2021-03-07:
Thanks for suggestion. I thought the same and so performed test which was disconnecting load. Yet, I can see that between 65% till 99% indicator LED on the RobotDyn board does not change intensity, only after switching from 99% to 100% it gets much brigther. It kind of smells that calculation gets "stuck" in this range. This indicator LED, D4 is showing "intensity" of PWM input:

https://robotdyn.com/pub/media/0G-00005677==Mod-Dimmer-5A-1L/DOCS/Schematic==0G-00005677==Mod-Dimmer-5A-1L.pdf

xanderdekok, 2021-03-07:
@wojciechczyz 
To start off with I want to say that I'm no electrical engineer.

To test if dimmer actually dimmed in the high ranges, I used a power meter to measure the absorbed power of my lamp that I wanted to dim. I did this because the lamp did not seem to dim in the high ranges. I could not notice any drop of the light intensity with the naked eye.
When testing the high ranges, I did see a drop in the absorbed power though.

Although I haven't specifically tested the 87-99% range, I can't remember the issue you're describing. 

Unfortunately don't have a test setup at hand, but maybe you can do this test yourself?
And maybe the added snubber circuit causes this problem?

wojciechczyz, 2021-03-07:
Thank you for your suggestion, I actually removed all load including snubber, so it seems - only estimating indicator PWM LED instensity of the controller board that something goes strange in 87-99% range - there is no change in intensity. Then there is big change from 99% to 100%.

fraxinas, 2021-03-14:
I'm seeing very erratic behavior when dimming a purely resistive load (light bulb) below ~75% with this circuitry: https://github.com/fraxinas/Geckocontroller/blob/main/hardware/Geckocontroller.pdf
I'm going to try switching my ZC detection with a schmitt trigger IC because the signal does indeed not look very digital and then try again.

4xvince, 2021-04-18:
Hi,
I'm new in using home assistant, and esphome.
I want to use an ESP8266 and the Robodyn AC dimmer (dual output).
But when I try to use the ac_dimmer module, the AC output flickers a lot (depends on the % command).

This is my  YAML:
output:
  - platform: ac_dimmer
    id: dimmer1
    gate_pin: GPIO12
    method: leading pulse
    zero_cross_pin:
      number: GPIO14
      mode: INPUT
      inverted: yes
  - platform: ac_dimmer
    id: dimmer2
    gate_pin: GPIO15
    zero_cross_pin:
      number: GPIO14
      mode: INPUT
      inverted: yes

light:
  - platform: monochromatic
    output: dimmer1
    name: "ECS power1"
    id: ecs_power1
  - platform: monochromatic
    output: dimmer2
    name: "ECS power2"
    id: ecs_power2

I'm an french hardware and embedded software engineer :-) , so I have a PicoScope with me.
First I watch the ZeroCross signal, with INPUT configuration
[image]
, and with INPUT_PULLUP
[image]
Even the signal is better, it not seems to be the cause of the flicker output....

Then I watch the dimmer output, the pulse with is 12µs 
This is a capture with in blue the dimmer output and in red the AC current
[image]
The AC current starts after 20µs

This is another capture with in blue the dimmer output and in red the AC current, with a lot a flicker
[image]
This capture shows that, even the dimmer output is present, the AC current does not start.

The flicker cause seams to be the short time of the dimmer output activation.
The propagation time of the optocoupler, and triac is near 20µs
=> I would test a pulse width of dimmer near 30 or 50µs

How can I do this to confirm my analysis?

glmnet, 2021-04-18:
Thanks for the detail in the analysis!! This is an interesting find!
Easiest to test is if you fork esphome and run from command line (or use vscode dev container)

The value you want to change is here:
https://github.com/esphome/esphome/blob/bf1885af3f5045a43dce38339485326678fc4b44/esphome/components/ac_dimmer/ac_dimmer.cpp#L17-L20

4xvince, 2021-04-18:
Ok, I could test it
but have you got a manual or tutorial to "fork esphome and run from command line"?

glmnet, 2021-04-19:
You might not be using a local installation 

Run 

`pip install -e .`

In the path where setup.python is

4xvince, 2021-04-20:
Thanks,
I lost some time with conflict versions of python on my w10 machine :-(
but that works fine now!

I try with GATE_ENABLE_TIME = 50;
[image]

No more flicker with this value :-)
I try also with 100, with same results.

mkleijn, 2021-04-20:
That sounds encouraging @4xvince ! I'll try to find some time to try this with my setup with 3x single Robodyn and 1x 4 channel Robodyn dimmers

@glmnet is it possible to change these source files in the ESPHome addon of Home Assistant as well? Or is this only possible with local installation & cmd line?

glmnet, 2021-04-20:
Here is a PR for you to test, install dev version of the add on and go to options and enter `ac-dimmer-longer-triac-gate`  in **esphome_version** field

[image]

4xvince, 2021-06-26:
Hello @glmnet,

So, whats next step to integrate this correction?

wojciechczyz, 2021-06-26:
Is this option to stay activated for all installations in future or would it be configurable? If so, possibly configuration option would be better than separate build.

glmnet, 2021-06-27:
> Is this option to stay activated for all installations in future or would it be configurable? If so, possibly configuration option would be better than separate build.

I can add the configuration option. So far one user reported this is not making any difference in his setup. So unless somebody says this is indeed useful I don't find a reason to add it.

4xvince, 2021-06-27:
Hi @glmnet,
I am very surprised of your reply...
I found a bug, find the root cause, find a solution, and it will not be corrected?

The timing in the code make no sens, and the comment in the code do not take account of the Optocoupler.
The Triac need a minimal pulse, but do not care about the length...
You can use any value, but not too short ! This works well with 50µs, 100µs or 1ms...

Stay with a value under 50µs makes the output flickering, I have showed you the proof...

A pulse larger than 50µs is mandatory, but not sufficient for proper behavior
=> the fact that one user has reported this is not making any difference is not a valid argument for not implement this correction.

Please consider including this bug fix for everyone because:
1-the user with a working setup, still have a working setup
2-the user with flickering ouptut, will have a working setup
3-the user with a bad setup, will have a bad setup but with correct driving of the Triac :-)

jougs, 2021-06-28:
@glmnet, @4xvince: Being the user who [reported](https://github.com/esphome/esphome/pull/1708#issuecomment-856684241) that the fix does not make a difference, I thought I quickly chip in.

My take is that if this fix helps at least some users, it should be merged by all means. Even if it does not resolve the issue for some, it will probably still help narrowing down the problems as @4xvince [pointed out](https://github.com/esphome/issues/issues/1632#issuecomment-869225273).

mkleijn, 2021-06-28:
@glmnet I never came round to reporting back on my tests, sorry about that. I tested using the your PR in the DEV version of the ESPHome HA integration and didn't get a much better result than using my current config using a non-inverted zero-crossing with pull-down:

` 
output:
  - platform: ac_dimmer
    id: dimmer1
    gate_pin: 16
    zero_cross_pin:
      number: 4
      mode: INPUT_PULLDOWN
    min_power: 35%

  - platform: ac_dimmer
    id: dimmer2
    gate_pin: 17
    zero_cross_pin:
      number: 4
      mode: INPUT_PULLDOWN

  - platform: ac_dimmer
    id: dimmer3
    gate_pin: 18
    zero_cross_pin:
      number: 4
      mode: INPUT_PULLDOWN

  - platform: ac_dimmer
    id: dimmer4
    gate_pin: 27
    zero_cross_pin:
      number: 4
      mode: INPUT_PULLDOWN
`

In the end I reverted to latest stable as I experienced an additional issue that HA seemed to briefly lose connection to the ESP device every 20 tot 30 seconds, and then come back up after 1 second or so. I don't think it has to do with this particular code change but probably something else in the dev branch. 

That being said I agree with @jougs and @4xvince that if it helped others it is worth merging it.

glmnet, 2021-06-28:
@4xvince sorry, I'm just not following up this properly, I'm not dropping the PR, in fact will merge it, I guess I was just waiting for you or somebody else to (double) check this is working ok now? I did ask for feedback on the PR thread but got no replies.

I guess the comment should be updated to mention that the `GATE_ENABLE_TIME` is larger because of optocoupler / triac driver propagation time? Does that make sense?

is 50uS ok? I can quickly change the PR to 100 if somebody else want's to do more testing this way

4xvince, 2021-07-01:
Hello, 

First, 50µs is OK, but to give margin due to capacity and length wire, 100µs should be fine.

Yes, comment should be updated, like you write:
To take account of the optocoupler and triac propagation time, the GATE_ENABLE_TIME is set to 100µs.

TarheelGrad1998, 2021-07-17:
I've been lurking for a while, and have tried all the above configs with little impact on the flicker.  I'd be willing to test this PR but no idea how.

> install dev version of the add on

Is that OS only then (re: "add on")?  I'm running Container and also ESPHome in a container.  
Or maybe just be a little more patient?  Sounds like it may be on it's way.

TarheelGrad1998, 2021-07-18:
Well, that's what I get for volunteering.  I tried working with the dimmer today and now I cannot get it to provide any power, either with the esp32 or an arduino.  Somehow I might have blown it.  :(

If someone can provide instructions and I can get it to work again, I'll be glad to give it a try.  Otherwise...

4xvince, 2021-07-21:
Hi, I could help you to get it works...
First, if you did not connect big AC load, it is difficult to damage the  AC dimmer module.
Please provide us schematics, or at least wiring of you setup.

TarheelGrad1998, 2021-07-30:
Thanks.  I ended up buying another, but I haven't had time to try it yet.  I'll report back.  Mostly I lack knowledge as to how to test this PR.

The original one, I had working against ESPHome but had the flicker when dimmed.  I wanted to see how Arduino compared, so I tried with an UNO using [their example here](https://github.com/RobotDynOfficial/Lib-RBD-Dimmer-for-Mega-UNO-Leonardo/tree/master/libraries/RBDdimmer/examples/SerialMonitorDim).  But once I wired that up, it would do nothing, even when going back to wiring to the ESP32.  So anyway, yeah, decided to throw a few more bucks into this project.  :)

Possible I had something wrong.  In which case, I have an extra now, lol.

TarheelGrad1998, 2021-07-31:
Ok, I'm bowing out.  I spent quite a bit of time today working on the dimmer.  On ESPHome it works but the flicker is intolerable, and I don't know enough to try this fix.  For whatever reason, I can't get the dimmer to do anything when connected to Arduino.  The UNO responds find but not even a blip from the light.  Frustrating.  

Into the drawer it goes.  Maybe I'll try again if/when the kinks get worked out.

eg321, 2021-09-14:
Just note: can't get ac_dimmer working on the esp8266 (nodemcuv2), but all is quite good at esp32 board with the same hw (4x channels robodyne).

It works as on/off only at esp8266.. there are no other voltage levels (checked with "input", "input_pullup").

closed 2021-09-15 by glmnet