esphome/issues #1633: climate_ir_lg Heat mode does not work

korvad, 2020-11-15:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):** **Hass.io**

**ESP (ESP32/ESP8266, Board/Sonoff):** **ESP8266**

**ESPHome version (latest production, beta, dev branch)**: **1.16.0-dev**

**Affected component:** https://esphome.io/components/climate/ir_climate.html

**Description of problem:**
Hello

I'm using this component https://esphome.io/components/climate/ir_climate.html to control my LG AC

Everything works fine except **heat** mode
When I try to use this mode it does not work well

In esphome log (see below) I see that IR code send to AC but it does switch to heat mode
It seems to me that in send implementation of ir component for lg AC here:
https://github.com/esphome/esphome/blob/dev/esphome/components/climate_ir_lg/climate_ir_lg.cpp

there is no implementation for **CLIMATE_MODE_HEAT** in LgIrClimate::transmit_state() method
Am I right concerning this issue or some configuration tweak needed from my side?
Below my esphome configuration

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: lounge_ir_conditioner
  platform: ESP8266
  board: nodemcuv2

api:

wifi:
  ssid: 'ssid'
  password: 'password'
  manual_ip:
    static_ip: IP
    gateway: GW
    subnet: NET

climate:
  - platform: climate_ir_lg
    name: "Lounge conditioner"
    sensor: lounge_temperature

sensor:
  - platform: homeassistant
    id: lounge_temperature
    name: "Lounge Temperature"
    internal: true
    entity_id: sensor.lounge_temperature

logger:
  level: DEBUG

remote_transmitter:
  pin: GPIO4
  carrier_duty_percent: 50%

ota:

```

**Logs (if applicable):**

```
...
[18:32:43][D][climate:010]: 'Lounge conditioner' - Setting
[18:32:43][D][climate:014]:   Mode: HEAT
[18:32:43][D][climate.climate_ir_lg:072]: climate_lg_ir mode code: 0x03
[18:32:43][D][climate.climate_ir_lg:174]: Sending climate_lg_ir code: 0x88C000C
[18:32:43][D][remote_transmitter:066]: Sending remote code...
[18:32:43][D][climate:262]: 'Lounge conditioner' - Sending state:
[18:32:43][D][climate:265]:   Mode: HEAT
[18:32:43][D][climate:270]:   Fan Mode: AUTO
[18:32:43][D][climate:273]:   Swing Mode: OFF
[18:32:43][D][climate:276]:   Current Temperature: 28.20°C
[18:32:43][D][climate:282]:   Target Temperature: 24.00°C
[18:32:43][D][climate:262]: 'Lounge conditioner' - Sending state:
[18:32:43][D][climate:265]:   Mode: HEAT
[18:32:43][D][climate:270]:   Fan Mode: AUTO
[18:32:43][D][climate:273]:   Swing Mode: OFF
[18:32:43][D][climate:276]:   Current Temperature: 28.20°C
[18:32:43][D][climate:282]:   Target Temperature: 24.00°C
...

```

**Additional information and things you've tried:**

dbeltman, 2021-03-12:
I also seem to have an issue related to / with this. My AC will go into cooling mode when i call the heat command.

dbeltman, 2021-03-16:
My issues with heating were fixed by using the development version which has this PR https://github.com/esphome/esphome/pull/1555 in it to support heating for LG AC.

korvad, 2021-03-29:
It's works to me also from develop branch. I suppose that issue can be close. Thx.

gastonc, 2021-07-09:
> I also seem to have an issue related to / with this. My AC will go into cooling mode when i call the heat command.

> I also seem to have an issue related to / with this. My AC will go into cooling mode when i call the heat command.

hi, i have the same problem with climate_ir_lg.  always turn on in cool mode.  did you find a solution?  Thank you

danieldabate, 2022-03-16:
> hi, i have the same problem with climate_ir_lg. always turn on in cool mode. did you find a solution? Thank you

Just saw your message, I modified a few months ago the current implementation to support all the "on" modes and also the "FAN ONLY" mode that is missing.

I created this issue to ask for help testing the implementation, but received no answers whatsoever, so didn't bother uploading any of the code: https://github.com/esphome/feature-requests/issues/1302 

If you are interested I can upload my changes for you to test. I'm hesitant of sending a pull request, because I don't know if every LG AC supports "FAN ONLY" mode and I don't want to break anything for others.

Let me know.

KenMacD, 2022-05-21:
@danieldabate I'd be interested, if you wouldn't mind pushing them to your fork. I won't be able to test for another month or so though.

danieldabate, 2022-05-22:
> @danieldabate I'd be interested, if you wouldn't mind pushing them to your fork. I won't be able to test for another month or so though.

Hey, just pushed the changes to my fork: https://github.com/danieldabate/esphome/tree/dev

As I said, it adds support for "Fan Only" mode + all on modes to turn on the air conditioner in any of the modes.

The rest should work as usual. I can't test the "swing" modes, because my air conditioner doesn't support it, but I haven't touched that part of the code.

Take your time to test it, hope it works for you.