esphome/issues #1634: Please clarify proto file license

maruel, 2020-11-16:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
N/A

**ESP (ESP32/ESP8266, Board/Sonoff):**
N/A

**ESPHome version (latest production, beta, dev branch)**
N/A

**Affected component:**
proto files

**Description of problem:**
The [license](https://github.com/esphome/esphome/blob/dev/LICENSE) calls out that the C++ code is under a GPLv3 license and that the python and all other parts are under MIT (paraphrasing).

Both [api.proto](https://github.com/esphome/esphome/blob/dev/esphome/components/api/api.proto) (6 authors) and [api_options.proto](https://github.com/esphome/esphome/blob/dev/esphome/components/api/api_options.proto) (1 author) are inside the C++ source tree.

Given the nature of GPLv3, I'd like clarity on which license they fall under because this can affect my capacity to contribute.

Optional nice to have in increasing order of complexity:

- Document this explicitly in the license file. 
- Add an short license header to the proto files.
- Move the proto file in a separate directory for more clarity.
- Given the multiple licenses used in this single repository, use a tool like [addlicense](https://github.com/google/addlicense) to insert a short 3 lines copyright+license header to every source file.
- Enforce addlicense check in github actions on PRs.

**Logs (if applicable):**
N/A

**Additional information and things you've tried:**
N/A

jesserockz, 2020-11-17:
@OttoWinter As you are the original author I think this falls to you?

maruel, 2021-03-17:
@jesserockz would you be open to a PR that moves the proto files in a separate directory to call out that it's the API layer and not implementation?

ssieb, 2021-03-17:
The license seems clear to me.  .proto files are not C++ (or in the specific list of included file extensions), so they would come under the "other parts" clause.  There is no real separation anyway, Python files are mixed in with the C++ files.  How would it affect your ability to contribute either way?  If you modify the .proto files, you're going to have modify other ones as well.
But regardless, the .proto files are also in the aioesphomeapi repo which is explicitly MIT only.

jesserockz, 2021-03-17:
> If you modify the .proto files, you're going to have modify other ones as well.

This is true as you will have to regenerate the python and c++ files to match therefore touching both license types

maruel, 2021-03-17:
I'd like to play with a Go implementation for a few things like Camera, motion detection, etc which means that I would copy the two proto files (with attribution) to a new project but nothing in python or C++. Hence the importance for me to have clarity for the proto files (api.proto and api_options.proto) licence.

https://github.com/esphome/aioesphomeapi is a vetted precedent by the esphome organization, contains the two proto files and is licensed as MIT. I think it answers my question. Thanks!

closed 2021-07-29 by stale[bot]