esphome/issues #1638: ESP-07 flash configuration not correct?

pe1rmg, 2020-11-18:
ESP 07 module is configured as 4MB in the ESPHome framework, while it is a 1MB module. The ESP 07**S** is the 4MB module however...

**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

ESPHome running on a Mac.

**ESP (ESP32/ESP8266, Board/Sonoff):**

Generic ESP 07 module

**ESPHome version (latest production, beta, dev branch)**

ESPHome v1.15.3
**Affected component:**

Core component.

**Description of problem:**

When using the board type "esp07" the configuration used is for a board with 4MB of flash. This causes OTA upload errors etc. Changing the board config to "esp01-1m" solves this issue. This workaround is ok, but for new users of ESPHome this can be confusing. If the reason behind it is that the esp-07**S** has the 4MB, maybe another configuration option can be added for the **S** module.
This was brought up in a previous, now closed issue, but the current version of ESPHome still exhibits this issue. 

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: test
  platform: ESP8266
  board: esp07
```

**Additional information and things you've tried:**

randybb, 2020-11-18:
Probably you are not aware how it works. ESPHome just passes these parameters to PlatformIO, so in case you feel a board definition is wrong, then you need raise an issue in [PlatformIO project](https://github.com/platformio/platformio-core/issues) and ideally fix it :) Once done, we can fix flash size definition in https://github.com/esphome/esphome/blob/dev/esphome/pins.py

Here is PlatformIO definition for [ESP-07](https://docs.platformio.org/en/latest/boards/espressif8266/esp07.html) that has been created in this [commit](https://github.com/platformio/platformio-core/commit/2c4fbda5e3fd5ac312a8514f7cffbc6dc93882e2) and based on [esp8266](https://www.esp8266.com/wiki/doku.php?id=esp8266-module-family#summary_table) wiki they made a mistake.

rbuehlma, 2021-01-22:
This has been fixed on the develop branch: https://github.com/platformio/platform-espressif8266/commit/468a6250fad7f77e1fe6a169b927de29358cfa8d#diff-322cabb954198713e5a7c1d31309d4cd7f11210031fda49806d294239f6745ad
Does it need to get on the master first?

sergeolkhovik, 2021-01-22:
Hmm.. here're my settings for ESP-07-based device:

```
esphome:
  name: d1
  platform: ESP8266
  board: esp07
  #board: esp01_1m
```

And those compiled well, running on 1.15.3.

closed 2021-06-05 by stale[bot]