esphome/issues #1639: Custom component with multiply types of sensors

geelenbert, 2020-11-18:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32
**ESPHome version (latest production, beta, dev branch)**

v1.15.3
**Affected component:**

Custom Component

**Description of problem:**
I have build a custom component to retrieve status and data from my WHR930 ventilation unit.
This device has many sensors (temperature, speeds,operating hours, etc...)

Now i'm having trouble with passing the sensor data to Home assistant. 
If i only use sensors, it works great. but i wanted to add a binary_sensor, but now i don't know how to configure this.

I tries in the added yaml file, but it does not work. 
How should this be done ?

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: whr930
  platform: ESP32
  board: esp32-evb
  includes:
    - whr930_component.h

wifi:
  ssid: "************"
  password: "************"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Whr930 Fallback Hotspot"
    password: "zJeRqpzZofNn"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

binary_sensor:
  - platform: gpio
    name: "S1"
    pin:
      number: 34
      inverted: True
      mode: INPUT_PULLUP
    on_click:
      min_length: 50ms
      max_length: 350ms
      then:
      - logger.log: "Click"      

uart:
  id: custom_uart_bus
  tx_pin: GPIO4
  rx_pin: GPIO36
  baud_rate: 9600

sensor:
- platform: custom
  lambda: |-
    auto whr930_sensors = new WHR930Component(id(custom_uart_bus));
    App.register_component(whr930_sensors);
    return {whr930_sensors->ComfortTemp_sensor,whr930_sensors->OutsideAirTemp_sensor,whr930_sensors->SupplyAirTemp_sensor,whr930_sensors->ReturnAirTemp_sensor,whr930_sensors->ExhaustAirTemp_sensor,whr930_sensors->FilterStatus_sensor};
    
  sensors:
  - name: "Comfort Temperature"
    unit_of_measurement: °C
    accuracy_decimals: 1
  - name: "Outside Air Temperature"
    unit_of_measurement: °C
    accuracy_decimals: 1
  - name: "Supply Air Temperature"
    unit_of_measurement: °C
    accuracy_decimals: 1   
  - name: "Return Air Temperature"
    unit_of_measurement: °C
    accuracy_decimals: 1   
  - name: "Exhaust Air Temperature"
    unit_of_measurement: °C
    accuracy_decimals: 1      
  
  binary_sensor:
  - name: "Filter Status"

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

heckler, 2020-12-02:
Just as a suggestion, for these types of queries it might be more appropriate to ask on the [ESPHome Facebook group](https://www.facebook.com/groups/2431447663796200), which has lots of users willing to comment.  This issues repo is more appropriate for bug reports.

With that out of the way, I wonder if you have the pin number setup correctly?    You can see below an example that I know works, this is on a ESP8266 board (a Sonoff Basic RF):

```yaml
binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    name: "Fireplace Light Button"
    on_press:
      - switch.toggle: relay
```

If you can make it work; if not, I suggest you post it on the FB group, you'll certainly have a quicker turnaround.

Either way, it is probably best to close this issue here, since this is not related to a bug in ESPHome.

geelenbert, 2020-12-03:
I figured it out on another way.

I made a custom component.

And just when i was about done with it, i found another project.... 
https://github.com/wichers/esphome-comfoair

closed 2020-12-03 by geelenbert