esphome/issues #1640: ESP32

DonGar, 2020-11-18:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io on an Intel NUC.

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32-DevKitC-32E

**ESPHome version (latest production, beta, dev branch)**

Latest Production (0.117.6)

**Affected component:**

ADC Sensor
https://esphome.io/components/sensor/adc.html

**Description of problem:**
When attenuation is set, the sensor will read the max allows voltage, or 0, but nothing in between with most voltages reading as max voltage.

If attenuation is not set and I use a voltage divider, the ESP32 reads expected voltage values.

When not using the voltage divider, the circuit is:

board 3.3v -> variable resister -> GPIO36
  Will always read 3.9v, even with 1000k Ohms of resistence.

board GND -> GPIO36
  Will always read 0.0v.

Disconnected -> GPIO36
  Will float the voltage around randomly.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: test
  platform: ESP32
  board: nodemcu-32s

sensor:
  - platform: adc
    pin: GPIO36
    name: "Flex 0"
    update_interval: 1s
    attenuation: 11db
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

Hoelli4C, 2020-11-21:
Same here: ESP32-Wroom-32E with ESPHome version 1.15.3-

I tried using GPIO32 and GPIO34 so far. I measure the attached battery at 3.29v but with attenuation set to 11db I get 3.9v as output. If I connect GND to my GPIO, I get 0v, if I disconnect it, I get floating low 0.0x voltages.

One Example: 
```yaml
[13:51:45][C][adc:026]:   Unit of Measurement: 'V'
[13:51:45][C][adc:026]:   Accuracy Decimals: 2
[13:51:45][C][adc:026]:   Icon: 'mdi:flash'
[13:51:45][C][adc:035]:   Pin: 34
[13:51:45][C][adc:047]:  Attenuation: 11db (max 3.9V)
[13:51:45][C][adc:051]:   Update Interval: 60.0s
[13:56:10][D][adc:056]: 'esp32-test-battery': Got voltage=3.90V
[13:56:10][D][sensor:092]: 'esp32-test-battery': Sending state 3.90000 V with 2 decimals of accuracy
```

glmnet, 2020-11-21:
> When not using the voltage divider, the circuit is:
> 
> board 3.3v -> variable resister -> GPIO36
> Will always read 3.9v, even with 1000k Ohms of resistence.

Can you be more specific here? if you just put a resistor in series with a voltage it will not cause the voltage to drop

DonGar, 2020-11-21:
Voltage does drop across a resister. There are some types of sensors which create a variable amount of resistance, and the way to read the sensor is to read the output voltage, which is what I'm trying to do.

General background on voltage drop across a resistor:

https://www.swtc.edu/Ag_Power/electrical/lecture/series_circuits.htm#:~:text=%224.,drop%20%3D%20Current%20times%20Resistor%20size.

DonGar, 2020-11-21:
Also, as a note, if I measure the difference between GPIO36 and GND with a meter, I get values in line with what I'm expecting. This leads me to think it's an ADC issue.

closed 2021-04-04 by stale[bot]