esphome/issues #1643: Devices permanently disconnected from wifi

jpgimenez, 2020-11-20:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass.io

**ESP (ESP32/ESP8266, Board/Sonoff):**

  * ESP32
  * Sonoff

**ESPHome version (latest production, beta, dev branch)**

1.15.3

**Affected component:**

https://esphome.io/components/wifi.html

**Description of problem:**
I was cleaning the config of some of my devices, and enabling ap/captive portal at the same time. In the process I saw some of the devices disappear from the network, I first suspect from the wifi settings, but the captive was not showing neither, I found issue #1629, but in my case the credentials are ok, other devices are using those credentials without issues, the main difference with working devices is that I removed the `platform: gpio` from `binary_sensor` section, because I wasn't using the button.

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: bedroom
  platform: ESP8266
  board: esp01_1m

wifi:
  ssid: !secret wifi-ssid
  password: !secret wifi-password

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

# Example configuration entry
web_server:
  port: 80

binary_sensor:
  - platform: status
    name: "System Status"

# Example configuration entry
sensor:
  - platform: uptime
    name: "System Uptime"
  - platform: wifi_signal
    name: "WiFi Signal Sensor"
    update_interval: 60s

switch:
  - platform: gpio
    name: "Relay"
    pin: GPIO12
    id: relay
    restore_mode: ALWAYS_ON

output:
  - platform: esp8266_pwm
    id: blue_led
    pin:
      number: GPIO13
      inverted: True
```

```diff
diff --git a/esphome/bedroom.yaml b/esphome/bedroom.yaml
index df961b2..ef7a20e 100644
--- a/esphome/bedroom.yaml
+++ b/esphome/bedroom.yaml
@@ -4,11 +4,8 @@ esphome:
   board: esp01_1m
 
 wifi:
-  networks:
-  - ssid: !secret wifi-ssid-int
-    password: !secret wifi-password-int
-  - ssid: !secret wifi-ssid-ext
-    password: !secret wifi-password-ext
+  ssid: !secret wifi-ssid
+  password: !secret wifi-password
 
 # Enable logging
 logger:
@@ -25,34 +22,6 @@ web_server:
 binary_sensor:
   - platform: status
     name: "System Status"
-  - platform: gpio
-    pin:
-      number: GPIO14
-      mode: INPUT_PULLUP
-      inverted: True
-    name: "Button"
-    on_multi_click:
-    - timing:
-        - ON for at most 1s
-        - OFF for at most 1s
-        - ON for at most 1s
-        - OFF for at least 0.2s
-      then:
-        - logger.log: "Double Clicked"
-    - timing:
-        - ON for 1s to 2s
-        - OFF for at least 0.5s
-      then:
-        - logger.log: "Single Long Clicked"
-    - timing:
-        - ON for at most 1s
-        - OFF for at least 0.5s
-      then:
-        - logger.log: "Single Short Clicked"
-        - homeassistant.service:
-            service: automation.trigger
-            data:
-              entity_id: "automation.light_toggle_bedroom"
 
 # Example configuration entry
 sensor:
```

**Logs (if applicable):**

I don't have logs.
 
**Additional information and things you've tried:**

I tried restarting, but nothing, the devices didn't try to connect to the router, no intent are show on the router logs.

blkqi, 2020-11-20:
I have encountered a similar issue with my esp01_1m boards.

alexyao2015, 2020-11-20:
You should post serial logs

sergeolkhovik, 2020-11-20:
Not sure but I might had already something similar - device becomes unavailable in HA after some time of work (I had ESP-07 which is ESP-01 + external antenna). Also I found that device can be pinged! And only after checking serial logs I found that there was an exception related to memory. The reason of that was I added remote_receiver component. So I removed some functionality (extra page with weather forecast) and got all working stable.

jpgimenez, 2020-11-21:
this is weird, if I connect to serial port it connects, but as I soon as I switch to power line it fails, not even try to connect to wifi... I can post the logs from the serial connection, but is a normal log of a device connecting to the wifi...

jpgimenez, 2020-11-22:
if I try an OTA this is logged:
```
[D][ota:243]: OTA in progress: 0.2%

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 3584, room 16 
tail 0
chksum 0x30
csum 0x30
csum err
ets_main.c .??
```

jpgimenez, 2020-11-22:
I got it working... I flashed over serial and adding the gpio section, now it connects to the wifi without issues.

jpgimenez, 2020-11-23:
Also, I was reviewing my config files and the docs, and I found that I have `board: esp01_1m` maybe from the beginning, but the docs now says `board: esp8285`
why? I configured something wrong there? was it changed? could I be facing some issues because of that?

alexyao2015, 2020-11-23:
Try to narrow down the issue by testing configs until you have a bare minimal config with the issue. Then try to determine which component/line its coming from.

closed 2020-12-27 by jpgimenez