esphome/issues #1646: Wrong arch image pulled when installing via hassio addon.

thedesp, 2020-11-21:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant Supervised 0.117.6 running on Ubuntu 20.10 for Raspberry Pi 4 (aarch64)

**ESP (ESP32/ESP8266, Board/Sonoff):**

NodeMCU, Sonoff Basic
**ESPHome version (latest production, beta, dev branch)**

1.15.3
**Affected component:**

**Description of problem:**
When installing esphome hassio addon from Home Assistant Supervisor Dashboard, armv7 image was pulled from docker hub instead of aarch64.
This causes compilation to result in the same error as in #1386.  

When installing standalone via docker command line "docker run --rm -v "${PWD}":/config -it esphome/esphome"
The image pulled was the correct esphome-aarch64:latest and I successfully compiled the same yaml.

**Problem-relevant YAML-configuration entries:**

```yaml
Irrelervant to the problem.
```

**Logs (if applicable):**

```
Home Assistant supervisor log
20-11-21 07:11:11 INFO (MainThread) [supervisor.addons] Creating Home Assistant add-on data folder /data/addons/data/15ef4d2f_esphome
20-11-21 07:11:11 INFO (SyncWorker_4) [supervisor.docker.interface] Downloading docker image esphome/esphome-hassio-armv7 with tag 1.15.3.

esphome addon startup log
-----------------------------------------------------------
 Hass.io Add-on: ESPHome
 ESPHome Hass.io add-on for intelligently managing all your ESP8266/ESP32 devices.
-----------------------------------------------------------
 Add-on version: 1.15.3
 You are running the latest version of this add-on.
 System: Ubuntu 20.10  (armv7 / raspberrypi4)
 Home Assistant Core: 0.117.6
 Home Assistant Supervisor: 2020.11.0
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
```

**Additional information and things you've tried:**

thedesp, 2020-12-08:
Never mind.

I just noticed that I was running home assistant armhf base imaage on aarch64 host OS. 
After changing home assistant installation to aarch64 image, esphome addon was pulling the correct aarch64 image too.

closed 2020-12-08 by thedesp

Royrdan, 2021-04-14:
Hi @thedesp . Im having the exact same issue except the oposite. When installing from the UI it installs aarch64 version but when installing from the hass cli in terminal it installs the arm7 version fine. I cannot work out how you fixed this in home assistant core as I can see in Home Assistant core I am setup for arm7. Any help would be greatly appreciated.
Thanks

thedesp, 2021-04-14:
> Hi @thedesp . Im having the exact same issue except the oposite. When installing from the UI it installs aarch64 version but when installing from the hass cli in terminal it installs the arm7 [...]

I did a new clean supervised install and specify machine type with aarch64 architecture then the UI pull the correct arch when installing add-on.
I suppose you can try specifying machine type with arm7 arch to force the add-on to be also on arm7.