esphome/issues #1653: Problem with DS18b20 sensors

HouseSimOn, 2020-11-24:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

I use Hass.io.
**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP8266 D1 mini
**ESPHome version (latest production, beta, dev branch)**

ESPHome version 1.15.3

**Affected component:**

https://esphome.io/components/sensor/dallas.html

**Description of problem:**
I have 10 DS18B20 sensors connected to the ESP8266 (D2 pin). Everything works fine till at least one sensors is warmer than 55 (about) degree Celsius. During this time I'm getting 0 degree Celsius from all sensors. When sensor is going to be colder everything is working fine again. I'm not sure if it is a problem with software (probably not) but I don't know where is the problem.
**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: co_cwu
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: "-"
  password: "-"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "-"
    password: "-"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

# Example configuration entry
dallas:
  - pin: D2
    update_interval: 2s

# Individual sensors
sensor:
  - platform: dallas
    address: 0x880301A27976XXXX
    name: "s1"
    filters:
    - sliding_window_moving_average:
        window_size: 8
        send_every: 8
# and the same for next 9 sensors

  - platform: wifi_signal
    name: "WiFi"
    update_interval: 60s
```

**Logs (if applicable):**

```
[17:33:20][W][dallas.sensor:140]: 's1' - Reseting bus for read failed!
[17:33:20][W][dallas.sensor:140]: 's2' - Reseting bus for read failed!
[17:33:20][W][dallas.sensor:140]: 's3' - Reseting bus for read failed!
[17:33:20][W][dallas.sensor:140]: 's7' - Reseting bus for read failed!
[17:33:20][W][dallas.sensor:140]: 's8' - Reseting bus for read failed!
[17:33:20][W][dallas.sensor:140]: 's4' - Reseting bus for read failed!
[17:33:20][W][dallas.sensor:140]: 's5' - Reseting bus for read failed!
[17:33:20][W][dallas.sensor:140]: 's6' - Reseting bus for read failed!
[17:33:20][W][dallas.sensor:140]: 's9' - Reseting bus for read failed!
[17:33:23][D][dallas.sensor:153]: 's0': Got Temperature=0.0°C
[17:33:23][D][sensor:092]: 's0': Sending state 0.00000 °C with 1 decimals of accuracy
[17:33:23][W][dallas.sensor:140]: 's1' - Reseting bus for read failed!
[17:33:23][D][sensor:092]: 's1': Sending state 3.80469 °C with 1 decimals of accuracy
[17:33:23][W][dallas.sensor:140]: 's7' - Reseting bus for read failed!
[17:33:23][D][sensor:092]: 's7': Sending state 0.00000 °C with 1 decimals of accuracy
[17:33:23][W][dallas.sensor:140]: 's8' - Reseting bus for read failed!
[17:33:23][D][sensor:092]: 's8': Sending state 16.64062 °C with 1 decimals of accuracy
[17:33:23][W][dallas.sensor:140]: 's2' - Reseting bus for read failed!
[17:33:23][D][sensor:092]: 's2': Sending state 23.21875 °C with 1 decimals of accuracy
[17:33:23][W][dallas.sensor:140]: 's3' - Reseting bus for read failed!
[17:33:23][D][sensor:092]: 's3': Sending state 5.04688 °C with 1 decimals of accuracy
[17:33:23][W][dallas.sensor:140]: 's6' - Reseting bus for read failed!
[17:33:23][D][sensor:092]: 's6': Sending state 6.71875 °C with 1 decimals of accuracy
[17:33:23][W][dallas.sensor:140]: 's5' - Reseting bus for read failed!
[17:33:23][D][sensor:092]: 's5': Sending state 4.15625 °C with 1 decimals of accuracy
[17:33:23][D][dallas.sensor:153]: 's4': Got Temperature=0.0°C
[17:33:23][D][sensor:092]: 's4': Sending state 12.34375 °C with 1 decimals of accuracy
[17:33:23][D][dallas.sensor:153]: 's9': Got Temperature=0.0°C
[17:33:23][D][sensor:092]: 's9': Sending state 6.07812 °C with 1 decimals of accuracy
[17:33:24][D][dallas.sensor:153]: 's0': Got Temperature=0.0°C
[17:33:24][W][dallas.sensor:140]: 's1' - Reseting bus for read failed!
[17:33:24][W][dallas.sensor:140]: 's2' - Reseting bus for read failed!
[17:33:24][W][dallas.sensor:140]: 's3' - Reseting bus for read failed!
[17:33:24][W][dallas.sensor:140]: 's4' - Reseting bus for read failed!
[17:33:24][W][dallas.sensor:140]: 's5' - Reseting bus for read failed!
[17:33:24][W][dallas.sensor:140]: 's6' - Reseting bus for read failed!
[17:33:24][W][dallas.sensor:140]: 's7' - Reseting bus for read failed!
[17:33:24][W][dallas.sensor:140]: 's8' - Reseting bus for read failed!
[17:33:24][W][dallas.sensor:140]: 's9' - Reseting bus for read failed!
[17:33:26][E][dallas.sensor:126]: Requesting conversion failed
[17:33:28][D][sensor:092]: 'WiFi': Sending state -69.00000 dB with 0 decimals of accuracy
[17:33:28][W][dallas.sensor:140]: 's1' - Reseting bus for read failed!
[17:33:28][W][dallas.sensor:140]: 's2' - Reseting bus for read failed!
[17:33:28][W][dallas.sensor:140]: 's3' - Reseting bus for read failed!
[17:33:28][W][dallas.sensor:140]: 's4' - Reseting bus for read failed!
[17:33:28][W][dallas.sensor:140]: 's5' - Reseting bus for read failed!
[17:33:28][W][dallas.sensor:140]: 's6' - Reseting bus for read failed!
[17:33:28][W][dallas.sensor:140]: 's9' - Reseting bus for read failed!
[17:33:28][W][dallas.sensor:140]: 's0' - Reseting bus for read failed!
[17:33:28][W][dallas.sensor:140]: 's8' - Reseting bus for read failed!
[17:33:28][W][dallas.sensor:140]: 's7' - Reseting bus for read failed!
[17:33:30][W][dallas.sensor:140]: 's0' - Reseting bus for read failed!
[17:33:30][W][dallas.sensor:140]: 's1' - Reseting bus for read failed!
[17:33:30][W][dallas.sensor:140]: 's2' - Reseting bus for read failed!
[17:33:30][W][dallas.sensor:140]: 's3' - Reseting bus for read failed!
[17:33:30][W][dallas.sensor:140]: 's4' - Reseting bus for read failed!
[17:33:30][W][dallas.sensor:140]: 's5' - Reseting bus for read failed!
[17:33:30][W][dallas.sensor:140]: 's6' - Reseting bus for read failed!
[17:33:30][W][dallas.sensor:140]: 's7' - Reseting bus for read failed!
[17:33:30][W][dallas.sensor:140]: 's8' - Reseting bus for read failed!
[17:33:30][W][dallas.sensor:140]: 's9' - Reseting bus for read failed!
[17:33:32][E][dallas.sensor:126]: Requesting conversion failed
[17:33:34][W][dallas.sensor:140]: 's0' - Reseting bus for read failed!
[17:33:34][W][dallas.sensor:140]: 's1' - Reseting bus for read failed!
[17:33:34][W][dallas.sensor:140]: 's2' - Reseting bus for read failed!
[17:33:34][W][dallas.sensor:140]: 's3' - Reseting bus for read failed!
[17:33:34][W][dallas.sensor:140]: 's4' - Reseting bus for read failed!
[17:33:34][W][dallas.sensor:140]: 's5' - Reseting bus for read failed!
[17:33:34][W][dallas.sensor:140]: 's6' - Reseting bus for read failed!
[17:33:34][W][dallas.sensor:140]: 's7' - Reseting bus for read failed!
[17:33:34][W][dallas.sensor:140]: 's8' - Reseting bus for read failed!
[17:33:34][W][dallas.sensor:140]: 's9' - Reseting bus for read failed!
[17:33:36][E][dallas.sensor:126]: Requesting conversion failed
[17:33:38][E][dallas.sensor:126]: Requesting conversion failed
[17:33:40][W][dallas.sensor:140]: 's0' - Reseting bus for read failed!
[17:33:40][W][dallas.sensor:140]: 's1' - Reseting bus for read failed!
[17:33:40][W][dallas.sensor:140]: 's2' - Reseting bus for read failed!
[17:33:40][W][dallas.sensor:140]: 's3' - Reseting bus for read failed!
[17:33:40][W][dallas.sensor:140]: 's4' - Reseting bus for read failed!
[17:33:40][W][dallas.sensor:140]: 's5' - Reseting bus for read failed!
[17:33:40][W][dallas.sensor:140]: 's6' - Reseting bus for read failed!
[17:33:40][W][dallas.sensor:140]: 's7' - Reseting bus for read failed!
[17:33:40][W][dallas.sensor:140]: 's8' - Reseting bus for read failed!
[17:33:40][W][dallas.sensor:140]: 's9' - Reseting bus for read failed!
[17:33:42][E][dallas.sensor:126]: Requesting conversion failed
[17:33:44][E][dallas.sensor:126]: Requesting conversion failed
[17:33:46][E][dallas.sensor:126]: Requesting conversion failed
[17:33:48][E][dallas.sensor:126]: Requesting conversion failed
[17:33:50][E][dallas.sensor:126]: Requesting conversion failed

```

**Additional information and things you've tried:**
Comment for log: Real temperature for each sensor was about 25 degree Celsius. Only one sensor was warmer than 55.
I've changed sensors to another ones - without success.

randybb, 2020-11-25:
Interesting... My heating controller currently uses 7 DS18B20 sensors monitoring temperature up to 80 °C without any issues. These are digital sensors and are responding only to commands from master, it could be that the one sensor causes a short circuit that takes the whole bus down.

Have you tried to heat it up and then just disconnect it?

cameronbunce, 2020-11-25:
Based on the extensive work of Chris Petrich, https://github.com/cpetrich/counterfeit_DS18B20, it looks like the actual sensor you're using is not a "real" one and could indeed be inducing shorts or other little gremlins. 

You don't mention it, so I have to ask: Do you have the 4.7 Ohm bus resistor in place here?

notwithstanding,
I would try each individual sensor and see if the bus failure is present on each. 

And then I would try a different source of the sensor to see if you've got a strange characteristic in your sensors.

HouseSimOn, 2020-12-01:
Sorry for late response but I was testing several options. I have several dozen sensors from different markets. Each sensor with original 1 meter cable breaks when I heat it up over 55-90 degree. Sensors without original cable work very well. I tested it with 4,7k pull-up resistor.

Then I thought that I try to change value of pull-up resistor (from 4,7k to 1k). Everything works fine with 1k resistor. I've been testing it for 6 days.

Thank You for Your help and sorry that I created SW issue.

Regards

closed 2020-12-01 by HouseSimOn