esphome/issues #1654: pn532_spi fails since introduction of pn532_i2c

apeeters, 2020-11-24:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**
Wemos D1 Mini (ESP8266)

**ESPHome version (latest production, beta, dev branch)**
1.16-dev commit 0059a6d

**Affected component:**
https://esphome.io/components/binary_sensor/pn532.html

**Description of problem:**

Since the introduction of i2c support for pn532 the spi communication fails. The following config works before but not after commit 0059a6d.

**Problem-relevant YAML-configuration entries:**
```yaml
spi:
  clk_pin: D5
  miso_pin: D6
  mosi_pin: D7

pn532_spi:
  cs_pin: D4
```

**Logs (if applicable):**

```
[0;32m[I][app:029]: Running through setup()...[0m
[0;35m[C][spi:022]: Setting up SPI bus...[0m
[0;32m[I][pn532_spi:015]: PN532Spi setup started![0m
[0;32m[I][pn532_spi:020]: SPI setup finished![0m
[0;35m[C][pn532:027]: Setting up PN532...[0m
[1;31m[E][pn532:031]: Error sending version command[0m
[1;31m[E][component:092]: Component was marked as failed.[0m
[0;32m[I][app:059]: setup() finished successfully![0m
[0;32m[I][app:105]: ESPHome version 1.16.0-dev compiled on Nov 24 2020, 19:50:29[0m
[0;35m[C][spi:081]: SPI bus:[0m
[0;35m[C][spi:082]:   CLK Pin: GPIO14 (Mode: OUTPUT)[0m
[0;35m[C][spi:083]:   MISO Pin: GPIO12 (Mode: INPUT)[0m
[0;35m[C][spi:084]:   MOSI Pin: GPIO13 (Mode: OUTPUT)[0m
[0;35m[C][spi:085]:   Using HW SPI: YES[0m
[0;35m[C][pn532:321]: PN532:[0m
[0;35m[C][pn532:333]:   Update Interval: 1.0s[0m
[0;35m[C][pn532_spi:065]:   CS Pin: GPIO2 (Mode: OUTPUT)[0m
```

I did notice this timeout while running on commit 22e1758, but tags read correctly:
```
[0;32m[I][app:029]: Running through setup()...[0m
[0;35m[C][spi:022]: Setting up SPI bus...[0m
[0;35m[C][pn532:025]: Setting up PN532...[0m
[0;32m[I][app:059]: setup() finished successfully![0m
[1;31m[E][pn532:346]: Timed out waiting for readiness from PN532![0m
[0;33m[W][pn532:103]: Requesting tag read failed![0m
[0;32m[I][app:105]: ESPHome version 1.16.0-dev compiled on Nov 24 2020, 19:54:32[0m
  ...
 [0;36m[D][pn532:156]: Found new tag '<snip>'[0m

```

I see the same log entries on 1.15.3.

**Additional information and things you've tried:**

Also tried latest build (7ff3f75), same result.

nmaggioni, 2021-02-03:
PN532 via SPI is broken as of [v1.16.0](https://github.com/esphome/esphome/releases/tag/v1.16.0), it doesn't react to any tag. Reverting to v1.15.3 works fine as a temporary fix.

jesserockz, 2021-02-04:
@apeeters @nmaggioni Can you please try esphome/esphome#1511 and confirm it is working for you with that code?

nmaggioni, 2021-02-04:
@jesserockz Applying those changes on top of either `dev` or `v1.16.0` works, tags are now read correctly. Thanks a lot!

apeeters, 2021-02-04:
@jesserockz Tag reading works again! Wifi also seems to be more stable (to be confirmed), great work! 

Tag reading now reports a Mifare authentication issue that I didn't see on 1.15.3, tag reads find though.

```
[0;36m[D][pn532:267]: Mifare classic[0m
[1;31m[E][pn532.mifare_classic:093]: Authentication failed - Block 0x04[0m
[0;36m[D][pn532:149]: Found new tag '<snip>'[0m
[0;36m[D][pn532:282]: Waiting to read next tag[0m
```

jesserockz, 2021-02-04:
Ah yes, I might need to change the logging messages. It just means your tag does not have NDEF formatted data (which was added this release too)

closed 2021-02-05 by jesserockz

jesserockz, 2021-02-14:
The fix is released in `1.16.1`

degrashopper, 2021-07-30:
I am using the pn532_i2c component on ESPHOME v 1.20.2 and getting this error. Is this reintroduced? 

[E][pn532.mifare_classic:093]: Authentication failed - Block 0x04
[D][text_sensor:015]: 'Druppel input': Sending state '45-4D-85-89'
[D][pn532:159]: Found new tag '45-4D-85-89'
[D][pn532:292]: Waiting to read next tag