esphome/issues #1656: Using !secret tag in packages subdirectory not using correct path for secrets.yaml

shbatm, 2020-11-25:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hassos VM with ESPHome Add-on (dev)

**ESP (ESP32/ESP8266, Board/Sonoff):**

Any

**ESPHome version (latest production, beta, dev branch)**

v1.16.0-dev

**Affected component:**

[packages](https://esphome.io/guides/configuration-types.html#packages), but also maybe any include

**Description of problem:**
When including a package file located in a subdirectory (e.g. config/common/) like in the [packages](https://esphome.io/guides/configuration-types.html#packages) example, using `!secret` will look for a `secrets.yaml` file in that directory, rather than the root directory.

**Problem-relevant YAML-configuration entries:**
```yaml
# In device config yaml file:
packages:
  device_base: !include common/device_config.yaml

# In common/device_config.yaml:
wifi:
  ssid: !secret iot_wifi_ssid
  password: !secret iot_wifi_password

```

**Logs (if applicable):**

```
ERROR Error while reading config: Invalid YAML syntax:

Error reading file /config/esphome/common/secrets.yaml: [Errno 2] No such file or directory: '/config/esphome/common/secrets.yaml'
```

**Additional information and things you've tried:**
Flattening the structure or putting a copy of the secrets file in the subdirectory work, but the `!secret` tag

ax42, 2020-11-27:
Found the same -- workaround is in the FAQ https://esphome.io/guides/faq.html#how-do-i-use-my-home-assistant-secrets-yaml

> If you want to keep all your secrets in one place, make a secrets.yaml file in the esphome directory with these contents (so it pulls in the contents of your main Home Assistant secrets.yaml file from one directory higher):

`<<: !include ../secrets.yaml`

shbatm, 2020-11-28:
@ax42 thanks for the lead on the workaround.

My concern is just within esphome, it should only be looking for its secrets.yaml in the root esphome folder (which if you want to direct to HA you can use the `<<:` trick.

Right now you'd need to put a secrets.yaml file in every subdirectory within esphome. If you compare this to how Home Assistant's processor works, you can use `!secret` in any subdirectory and it references the file in /config.

closed 2021-04-11 by stale[bot]