esphome/issues #1657: PZEM-004T V3 bug

mttstt, 2020-11-26:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

n/a

**ESP (ESP32/ESP8266, Board/Sonoff):**

NodeMcu & WeMos D1 mini

**ESPHome version (latest production, beta, dev branch)**

ESPHome v1.15.2

**Affected component:**

https://esphome.io/components/sensor/pzemac.html

**Description of problem:**
With Tasmota v9.1, pzem data are read correctly (see logs), while with EspHome data are not read correctly. I'm not able decode the reply of pzem in EspHome (see log).

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: pzem
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: !secret Wifi_SSID_1
  password: !secret Wifi_PWD_1
  # Optional manual IP
  manual_ip:
    static_ip: 192.168.1.11
    gateway: 192.168.1.1
    subnet: 255.255.255.0
    dns1: 192.168.1.1
    dns2: 8.8.8.8
 # This will cause ESPHome to only enable the access point when no connection to the WiFi router can be made.
  ap:
    ssid: pzemRecovery

# Enable captive portal
captive_portal:

web_server:
   port: 80

# Enable logging
logger:
#  baud_rate: 0
  level: VERY_VERBOSE

# Enable ota
ota:

# Enable Home Assistant API
api:

time:
  - platform: homeassistant
    id: homeassistant_time

binary_sensor:
  - platform: status
    name: " Stato sensore"

text_sensor:
  - platform: wifi_info
    ip_address:
      name: Address_IP
    ssid:
      name: SSID
    bssid:
      name: BSSID
  - platform: version
    name: "ESPHome Version"

uart:
  rx_pin: D2
  tx_pin: D1
  baud_rate: 9600
  stop_bits: 2

sensor:
  - platform: pzemac
    address: 1
    current:
      name: "PZEM-004T V3 Current(1)"
    voltage:
      name: "PZEM-004T V3 Voltage(1)"
    energy:
      name: "PZEM-004T V3 Energy(1)"
    power:
      name: "PZEM-004T V3 Power(1)"
    frequency:
      name: "PZEM-004T V3 Frequency(1)"
    power_factor:
      name: "PZEM-004T V3 Power Factor(1)"
    update_interval: 60s

```

**Logs (if applicable):**

```
Logs ESPHOME:
....
[20:13:42][VV][uart_esp8266:111]:     Wrote 0b00000001 (0x01)
[20:13:42][VV][uart_esp8266:111]:     Wrote 0b00000100 (0x04)
[20:13:42][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[20:13:42][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[20:13:42][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[20:13:42][VV][uart_esp8266:111]:     Wrote 0b00001010 (0x0A)
[20:13:42][VV][uart_esp8266:111]:     Wrote 0b01110000 (0x70)
[20:13:42][VV][uart_esp8266:111]:     Wrote 0b00001101 (0x0D)
[20:13:42][VV][uart_esp8266:132]:     Read 0b00000001 (0x01)
[20:13:42][VV][uart_esp8266:132]:     Read 0b00010000 (0x10)
[20:13:42][VV][uart_esp8266:132]:     Read 0b10010100 (0x94)
[20:13:42][VV][uart_esp8266:132]:     Read 0b00000111 (0x07)
[20:13:42][VV][uart_esp8266:132]:     Read 0b00000000 (0x00)
[20:13:42][VV][uart_esp8266:132]:     Read 0b00000000 (0x00)
[20:13:42][VV][uart_esp8266:132]:     Read 0b00000000 (0x00)
[20:13:42][VV][uart_esp8266:132]:     Read 0b00000000 (0x00)
[20:13:42][VV][uart_esp8266:132]:     Read 0b10101000 (0xA8)
[20:13:42][VV][uart_esp8266:132]:     Read 0b00000010 (0x02)
[20:13:42][VV][uart_esp8266:132]:     Read 0b00000000 (0x00)
[20:13:42][VV][uart_esp8266:132]:     Read 0b01000000 (0x40)
[20:13:42][VV][uart_esp8266:132]:     Read 0b00011111 (0x1F)
[20:13:42][VV][uart_esp8266:132]:     Read 0b00000000 (0x00)
[20:13:42][VV][uart_esp8266:132]:     Read 0b00000000 (0x00)
[20:13:42][VV][uart_esp8266:132]:     Read 0b00101101 (0x2D)
[20:13:42][VV][uart_esp8266:132]:     Read 0b11111100 (0xFC)
....

Logs Tasmota are OK:
...
20:18:04 DMP: 01 04 14 09 3E 00 00 00 00 00 00 00 00 08 25 00 00 01 F4 00 00 00 00 FB 8B
20:18:05 DMP: 01 04 14 09 3A 00 00 00 00 00 00 00 00 08 25 00 00 01 F4 00 00 00 00 0A BB
20:18:06 DMP: 01 04 14 09 39 00 00 00 00 00 00 00 00 08 25 00 00 01 F4 00 00 00 00 4E FF
...

```

**Additional information and things you've tried:**
I tried with WeMos D1 mini and Nodemcu. Same results.
The pzem reply in EspHome are strange...

randybb, 2020-11-26:
I don't have any issue with my 3 pzem sensors connected to one bus, but i are explicitly defining my _uart_ as _modbus_.

````yaml
modbus:
  uart_id: uart2

uart:
  - id: uart2
    rx_pin: GPIO16
    tx_pin: GPIO17
    baud_rate: 9600
    stop_bits: 2
````

mttstt, 2020-11-27:
Thanks randybb.

I modified YAML configuration but the problem persists on WeMos d1 mini. 
With Tasmota, the 2 pzem work correctly. Any idea?

**YAML:**
```
esphome:
  name: pzem
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: !secret Wifi_SSID_1
  password: !secret Wifi_PWD_1
  # Optional manual IP
  manual_ip:
    static_ip: 192.168.1.11
    gateway: 192.168.1.1
    subnet: 255.255.255.0
    dns1: 192.168.1.1
    dns2: 8.8.8.8
 # This will cause ESPHome to only enable the access point when no connection to the WiFi router can be made.
  ap:
    ssid: pzemRecovery

# Enable captive portal
captive_portal:

web_server:
   port: 80

# Enable logging
logger:
#  baud_rate: 0
  level: VERY_VERBOSE

# Enable ota
ota:

# Enable Home Assistant API
api:

time:
  - platform: homeassistant
    id: homeassistant_time

binary_sensor:
  - platform: status
    name: " Stato sensore"

text_sensor:
  - platform: wifi_info
    ip_address:
      name: Address_IP
    ssid:
      name: SSID
    bssid:
      name: BSSID
  - platform: version
    name: "ESPHome Version"

modbus:
      uart_id: uartpzem

uart:
  - id: uartpzem
    rx_pin: D2
    tx_pin: D1
    baud_rate: 9600
    stop_bits: 2

sensor:
  - platform: pzemac
    address: 1
    current:
      name: "PZEM-004T V3 Current(1)"
    voltage:
      name: "PZEM-004T V3 Voltage(1)"
    energy:
      name: "PZEM-004T V3 Energy(1)"
    power:
      name: "PZEM-004T V3 Power(1)"
    frequency:
      name: "PZEM-004T V3 Frequency(1)"
    power_factor:
      name: "PZEM-004T V3 Power Factor(1)"
    update_interval: 60s

  - platform: pzemac
    address: 2
    current:
      name: "PZEM-004T V3 Current (2)"
    voltage:
      name: "PZEM-004T V3 Voltage(2)"
    energy:
      name: "PZEM-004T V3 Energy(2)"
    power:
      name: "PZEM-004T V3 Power(2)"
    frequency:
      name: "PZEM-004T V3 Frequency(2)"
    power_factor:
      name: "PZEM-004T V3 Power Factor(2)"
    update_interval: 60s

  - platform: uptime
    name: Uptime
    icon: mdi:clock-outline
    update_interval: 300s
    unit_of_measurement: days
    filters:
      - multiply: 0.000011574
  - platform: wifi_signal
    name: "Segnale wifi sensore"
    update_interval: 60s
```

**New Logs:**
[18:50:36][VV][uart_esp8266:111]:     Wrote 0b00000001 (0x01)
[18:50:36][VV][uart_esp8266:111]:     Wrote 0b00000100 (0x04)
[18:50:36][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[18:50:36][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[18:50:36][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[18:50:36][VV][uart_esp8266:111]:     Wrote 0b00001010 (0x0A)
[18:50:36][VV][uart_esp8266:111]:     Wrote 0b01110000 (0x70)
[18:50:36][VV][uart_esp8266:111]:     Wrote 0b00001101 (0x0D)
[18:50:36][VV][uart_esp8266:132]:     Read 0b00000001 (0x01)
[18:50:36][VV][uart_esp8266:132]:     Read 0b00010000 (0x10)
[18:50:36][VV][uart_esp8266:132]:     Read 0b10010100 (0x94)
[18:50:36][VV][uart_esp8266:132]:     Read 0b00110100 (0x34)
[18:50:36][VV][uart_esp8266:132]:     Read 0b00000000 (0x00)
[18:50:36][VV][uart_esp8266:132]:     Read 0b00000000 (0x00)
[18:50:36][VV][uart_esp8266:132]:     Read 0b00000000 (0x00)
[18:50:36][VV][uart_esp8266:132]:     Read 0b00000000 (0x00)
[18:50:36][VV][uart_esp8266:132]:     Read 0b01000110 (0x46)
[18:50:36][VV][uart_esp8266:132]:     Read 0b00000011 (0x03)
[18:50:36][VV][uart_esp8266:132]:     Read 0b00000101 (0x05)
[18:50:36][VV][uart_esp8266:132]:     Read 0b01000000 (0x40)
[18:50:36][VV][uart_esp8266:132]:     Read 0b00011111 (0x1F)
[18:50:36][VV][uart_esp8266:132]:     Read 0b00000000 (0x00)
[18:50:36][VV][uart_esp8266:132]:     Read 0b00000000 (0x00)
[18:50:36][VV][uart_esp8266:132]:     Read 0b00011011 (0x1B)
[18:50:36][VV][uart_esp8266:132]:     Read 0b11111111 (0xFF)
[18:50:41][VV][api.service:220]: on_ping_request: PingRequest {}
[18:50:41][VV][api.service:032]: send_ping_response: PingResponse {}
[18:50:43][VV][scheduler:152]: Running interval '' with interval=10000 last_execution=46924 (now=56924)
[18:50:45][VV][scheduler:152]: Running interval 'update' with interval=60000 last_execution=4294966484 (now=59190)
[18:50:45][VV][uart_esp8266:111]:     Wrote 0b00000010 (0x02)
[18:50:45][VV][uart_esp8266:111]:     Wrote 0b00000100 (0x04)
[18:50:45][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[18:50:45][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[18:50:45][VV][uart_esp8266:111]:     Wrote 0b00000000 (0x00)
[18:50:45][VV][uart_esp8266:111]:     Wrote 0b00001010 (0x0A)
[18:50:45][VV][uart_esp8266:111]:     Wrote 0b01110000 (0x70)
[18:50:45][VV][uart_esp8266:111]:     Wrote 0b00111110 (0x3E)
[18:50:45][VV][uart_esp8266:132]:     Read 0b00000010 (0x02)
[18:50:45][VV][uart_esp8266:132]:     Read 0b00010000 (0x10)
[18:50:45][VV][uart_esp8266:132]:     Read 0b10010100 (0x94)
[18:50:45][VV][uart_esp8266:132]:     Read 0b00110110 (0x36)
[18:50:45][VV][uart_esp8266:132]:     Read 0b00000000 (0x00)
[18:50:45][VV][uart_esp8266:132]:     Read 0b00000000 (0x00)
[18:50:45][VV][uart_esp8266:132]:     Read 0b00000000 (0x00)
[18:50:45][VV][uart_esp8266:132]:     Read 0b00000000 (0x00)
[18:50:45][VV][uart_esp8266:132]:     Read 0b10101000 (0xA8)
[18:50:45][VV][uart_esp8266:132]:     Read 0b00000010 (0x02)
[18:50:45][VV][uart_esp8266:132]:     Read 0b00000000 (0x00)
[18:50:45][VV][uart_esp8266:132]:     Read 0b01000000 (0x40)
[18:50:45][VV][uart_esp8266:132]:     Read 0b00011111 (0x1F)
[18:50:45][VV][uart_esp8266:132]:     Read 0b00000000 (0x00)
[18:50:45][VV][uart_esp8266:132]:     Read 0b00000000 (0x00)
[18:50:45][VV][uart_esp8266:132]:     Read 0b01010011 (0x53)
[18:50:45][VV][uart_esp8266:132]:     Read 0b11111000 (0xF8)

k-bronowski, 2020-12-01:
Change stop_bits propery of uart from 2 to 1.

gabrielbatir, 2020-12-16:
It worked for me too after I changed stop_bits to 1 even if I get this in the logs:
`[19:05:42][E][uart:054]:   Invalid stop bits: Integration requested stop_bits 2 but you have 1!`

closed 2021-04-30 by stale[bot]