esphome/issues #1658: ESPHome unable to maintain connection to WiFi

lanrat, 2020-11-27:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

esphome and homeassistant in Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**

 ESP32: nodemcu-32s

**ESPHome version (latest production, beta, dev branch)**

ESPHome 1.15.3

**Affected component:**

WIFI: https://esphome.io/components/wifi.html

**Description of problem:**
My ESPHome sensor appears to have the network connection die every 7-9 minutes. When it is dead, it does not report back to Home Assistant, and is not reachable via ping. However it does come back on its own usually 6-8 minutes later, which means this cycle repeats every 15 minutes. Even when the networking is not working, the ESP appears to still function locally by displaying sensor data on the attached display.

I have spent a lot of time debugging this, and it appears to be an incompatibility between the ESP32 and my Router (Ruckus Unleashed AP). When ESPHome seems to no longer be reachable on the network, or when it later becomes visible again (via ping) there does not appear to be anything useful printed to the debug logs, even with increased verbosity.

[image]

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: tricoder
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "WIFINAME"
  password: "PASSWORD"
  power_save_mode: none
  ap:
    ssid: "AP"
    password: "PASSWORD"

captive_portal:

logger:

api:
  password: "PASSWORD"

web_server:
   port: 80

i2c:
  - id: id_i2c_sensor
    sda: GPIO21
    scl: GPIO22
  - id: id_i2c_screen
    sda: GPIO18
    scl: GPIO19

uart:
 rx_pin: GPIO26
 id: id_uart_ppm
 baud_rate: 9600

font:
  - file: "fonts/arial.ttf"
    id: font_16
    size: 16
  - file: "fonts/arial.ttf"
    id: font_12
    size: 12
  - file: "fonts/arial.ttf"
    id: font_10
    size: 10

globals:
   - id: temp_f
     type: float
     restore_value: no
     initial_value: 'false'
   - id: temp_f2
     type: float
     restore_value: no
     initial_value: 'false'

display:
  - platform: ssd1306_i2c
    model: "SSD1306 128x64"
    address: 0x3C
    i2c_id: id_i2c_screen
    update_interval: 1s
    id: id_screen
    lambda: |-
      // title text
      it.printf(it.get_width()/2, 0, id(font_16), TextAlign::TOP_CENTER, "TRICODER");

      // top line
      it.line(0, 16, it.get_width(), 16);
      // bottom line
      it.line(0, 48, it.get_width(), 48);
      // middle line
      it.line(it.get_width()/2, 16, it.get_width()/2, 48);

      // Print humidity
      if (id(id_humidity).has_state()) {
        it.printf(1, it.get_height(), id(font_10), TextAlign::BOTTOM_LEFT, "%2.1f %2.1f%s", id(id_humidity).state, id(id_hum2).state, id(id_humidity).get_unit_of_measurement().c_str());
      }
      // Print temp
      if (id(id_temp).has_state()) {
        it.printf(it.get_width()-1, it.get_height(), id(font_10), TextAlign::BOTTOM_RIGHT , "%2.1f %2.1f°F", id(temp_f), id(temp_f2));
      }

      // Print ppm
      // ppm 10
      if (id(id_ppm_10).has_state()) {
         it.printf(1,18, id(font_16), TextAlign::TOP_LEFT , "%3.0f %s", id(id_ppm_10).state, id(id_ppm_10).get_unit_of_measurement().c_str());
      }
      // ppm 2.5 & 1
      if (id(id_ppm_1).has_state()) {
         it.printf(1,34, id(font_10), TextAlign::TOP_LEFT , "%3.0f/2.5  %3.0f/1", id(id_ppm_25).state, id(id_ppm_1).state);
      }

      // print gas and pressure
      if (id(id_gas).has_state()) {
        it.printf(it.get_width()-1, 18, id(font_12), TextAlign::TOP_RIGHT , "%5.0f %s", id(id_gas).state, id(id_gas).get_unit_of_measurement().c_str());
      }
      if (id(id_pressure).has_state()) {
        it.printf(it.get_width()-1, 33, id(font_12), TextAlign::TOP_RIGHT , "%4.1f %s", id(id_pressure).state, id(id_pressure).get_unit_of_measurement().c_str());
      }

sensor:
  - platform: dht
    pin: GPIO23
    model: AM2302
    temperature:
      name: "DHT Temperature"
      id: id_temp2
      on_value:
        then:
          - lambda: id(temp_f2) = x * (9.0/5.0) + 32.0; # convert C -> F
    humidity:
      name: "DHT Humidity"
      id: id_hum2
    update_interval: 10s

  - platform: pmsx003
    uart_id: id_uart_ppm
    type: PMSX003
    pm_1_0:
      name: "Particulate Matter <1.0µm Concentration"
      id: id_ppm_1
      filters:
        - sliding_window_moving_average:
            window_size: 30
            send_every: 10
    pm_2_5:
      name: "Particulate Matter <2.5µm Concentration"
      id: id_ppm_25
      filters:
        - sliding_window_moving_average:
            window_size: 30
            send_every: 10
    pm_10_0:
      name: "Particulate Matter <10.0µm Concentration"
      id: id_ppm_10
      filters:
        - sliding_window_moving_average:
            window_size: 30
            send_every: 10

  - platform: bme680
    i2c_id: id_i2c_sensor
    temperature:
      name: "Temperature"
      id: id_temp
      on_value:
        then:
          - lambda: id(temp_f) = x * (9.0/5.0) + 32.0; # convert C -> F
    pressure:
      name: "Pressure"
      id: id_pressure
    humidity:
      name: "Humidity"
      id: id_humidity
    gas_resistance:
      name: "Gas Resistance"
      id: id_gas
    address: 0x77
    update_interval: 30s
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

- I tried setting a timeout with an HTTP get every 10s to see if that would keep the connection alive
- I hard-coded my AP's MAC address in the yaml
- I used a more powerful power supply
- changed wifi `reboot_timer` to a smaller value (5min)
- replaced the ESP32 with another ESP32
- Tried adjusting my Ruckus AP's settings
    - Disabled 802.11r & 802.11k
    - Disabled "Application Visibility" setting
    - Disabled inactivity timeout
    - and many others...

Eventually I got out my [OpenWRT travel router](https://www.gl-inet.com/products/gl-ar750s/) and had the ESP connect to that and it worked flawlessly. So it appears to be some incompatibility between ESPHome and my Ruckus AP.

[image]

I've tried yet again adjusting every setting I could think of in the Ruckus AP to no effect.

Are there any particular settings ESPHome needs to maintain a stable WiFi connection? The ESP32 with ESPHome is the only device I have that has difficulty connecting to the network.

randybb, 2020-11-28:
It is WiFi, you can't just compare two devices as there are many factors in settings, AP position and then in reflected signal,...

Normally I would propose too add [RSSI sensor](https://esphome.io/components/sensor/wifi_signal.html) to your configuration to know if it is not caused just by bad reception. In your case it is dying regularly, so probably you have enabled Band steering that is trying to force it to 5GHz just by ignoring few association requests and it is doing it over and over again.

lanrat, 2020-11-30:
Thanks for the response. My ESP has always been stationary within 1m of the APs so I doubt signal is an issue.

I created another virtual network with most settings disavled on my Ruckus AP and had the ESP connect to it and it worked perfectly. Then I slowly adjusted the new network, one setting at a time, to be identical to the old one. After each change I waited a few hours to see if the ESP would disconnect. It never did. 

Eventually the only difference between the new ESP network I created and the old one was the name... So I disabled my old network and renamed the new one to the old one, which resulted on all my devices connecting to the new one. Its been a day so far and no disconnects.

Its working now, but I have no idea why it was not working before and why it is working now. As far as I can tell the only difference between the networks is the virtual MAC address assigned to each one.

lanrat, 2020-12-02:
My ESP32 has been able to maintain a stable connection for the past 3 days. At this point I have no idea what caused the issue I've seen and am no longer able to reproduce the problem. So I'll close this issue.

closed 2020-12-02 by lanrat