esphome/issues #1659: New version of Xiaomi CGG1 temperature & humidity sensor not supported ?

LoicFr, 2020-11-28:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Supervisor 2020.11.0
Docker 18.09.6
Home Assistant 0.118.4

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 DevKitC WROOM-32U

**ESPHome version (latest production, beta, dev branch)**

ESPHome 1.15.3

**Affected component:**

https://esphome.io/components/sensor/xiaomi_ble.html#cgg1

**Description of problem:**
Hello,

I have just bought a new Xiaomi CGG1 sensor and I can't manage to retrieve its values with ESPHome. It also does not work with the last version of ble_monitor https://github.com/custom-components/ble_monitor and even with the Xiaomi Home app on Android (it can't be associated). I hope this is not a new version with a new unsupported firmware... The only thing I achieved to do is to pair it with the official app Quingping+ and the values can be read in this app. I can also see that the firmware version is 1.50. I have an older CGG1 I bought last year which perfectly works everwhere, ESPHome included, but the firmware version is different: 1.1.2_0036. Does anyone achieve to read values of CGG1 sensor with firmware version 1.50 ? I checked that it is not a CGG1H version (Home Kit version), because it is written CGG1 on the box. ESPHome can see the sensor during a BLE scan, but its values are not read. RSSI values can be read by ESPHome.

Any help would be greatly appreciated :) Maybe firmware version shown in Quingping+ app can be a good indicator to evaluate the support of CGG1 sensor version in ESPHome ?

**Problem-relevant YAML-configuration entries:**
```yaml
sensor:
  - platform: xiaomi_cgg1
    mac_address: "58:2D:34:10:XX:XX"
    temperature:
      name: "Temperature0"
    humidity:
      name: "Humidity0"
    battery_level:
      name: "Battery0"
```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

Flameeyes, 2020-11-30:
I'm affected by this — it appears this version needs a bindkey, probably using the same encryption as CGD1 and similar. I have the bindkey for one of them, I'll try to get some time to test this soon.

LoicFr, 2020-12-01:
Hi,

Thanks for your reply. This is bad luck, I chose this model because it was simple to use it, without bindkey...
How did you manage to get the bindkey of this new CGG1 ? It seems it can't be paired with Xiaomi Home app, and I think that this is a mandatory step to get the bindkey.

Regards,
Loïc

Flameeyes, 2020-12-01:
Yeah I got myself a few CGG1 a couple of weeks ago, and they probably were 1.12 versions, they worked fine without bindkey. The ones I received yesterday don't show up at all, and they are definitely 1.50 versions. The other difference is that the old ones were named "ClearGrass Temp & RH" and the new ones "Qingping Temp & RH M" (Mijia?)

The new ones do pair with the latest (European) version of Xiaomi Home app, but you can't sniff the session because it's protected now with a security policy. I did find a reference (https://github.com/custom-components/ble_monitor/issues/7#issuecomment-595874419) to a modded app that logs the pairings to a file. That turned out to pair the old sensors too, but I haven't actually completed those.

Note that the esphome cgg1 driver doesn't implement bindkey at all. I think I'll just try copying the code from the cgd1 and see if that works, but I have not had the time to sit down at the right computer for that today.

LoicFr, 2020-12-01:
Yes, CGG1 "M" versions are for Mijia and "H" versions for HomeKit. I am afraid to have another version very strangely named "Goose". I can't pair it with Xiaomi Home app and I try with a China server and a European server. On the package of the sensor there is no "works with Mijia" logo (I have one with the old version). Maybe I have a H version ? On the package it is just written CGG1. I think I will try to find another version, because even if you achieve to make yours to work with ESPHome, it will not help me because I can't pair it to get the bindkey.

Flameeyes, 2020-12-01:
Both of mine have had "Works with Mijia" on the outside, only physical difference in packaging is that in the second lot the battery was _not_ fitted inside with plastic holding it, but rather sitting next to the mounting bracket. And the second lot had "qingping" embossed on the back.

FTR the first lot was coming from "Mi Yeelight Store" and the second from "Xiaomi-Chain Store", if anyone else is finding this later.

LoicFr, 2020-12-01:
My bad version is coming from "Mi smartHome Store". I was confident because every pictures in people's comments show that they received a version with the "works with mijia" logo, in contrary with the one I received. So I think it is very difficult to be sure to receive the right version because the stock can be mixed with different versions.

Anyway, thank you for helping me clarify this. Maybe we can close this issue because yours is quite different since you have another version than mine ?

jorpand, 2020-12-02:
@LoicFr same problem here, got the Goose one, the only way to make it work with another device is with the Qingping+ application...

jmtemmos, 2020-12-04:
Hi. Same problem here. Bought several and they work in Xiami Home app but not in hassio/esphome.
Log outputs:

Qingping Temp & RH M

Too bad....

Flameeyes, 2020-12-06:
So I can confirm that the encrypted CGG1 I have use exactly the same protocol as the CGD1 and other devices.

I've sent a pull request to add an optional bindkey configuration. I'm about to update the docs as well.

jmtemmos, 2020-12-06:
@Flameeyes Thanks a lot ! I'll wait for the merge then.

LoicFr, 2020-12-06:
Good news, thank you for solving this ! I give up on "Goose" version and will try to get the new encrypted version you have.

closed 2020-12-06 by LoicFr

tonypascal, 2021-01-17:
Hello,
did anyone tried / found a standalone implementation of the decrypting function on ESP32 / Arduino IDE ?
I guess this it the decrypt_xiaomi_payload function, but I can't make it work...
any help ?
Tony

rogodra, 2021-01-23:
Hello, in what state the implementation?
I have two CGG1 Qingping, I got the bindkey, I'm waiting to be able to use them.
Thanks!

jorpand, 2021-01-23:
Hi! How did you manage to get the bindkey?

KR

tonypascal, 2021-01-23:
Personnaly, I used the custom Adnroid app method. Work well:
link here<https://github.com/custom-components/ble_monitor/blob/master/faq.md#my-sensors-ble-advertisements-are-encrypted-how-can-i-get-the-key>

Download<https://www.kapiba.ru/2017/11/mi-home.html> MiHome mod and install the MiHome mod apk on your android phone. Use google translate to translate the website, as the website is only available in Russian. At the bottom, you will find the download link to the latest version. MiHome mod is basically just MiHome with some translatons from chinese to English and Russian, but also with the possibility to save the encryption key.

  *   During setup of the app, give access to the local device storage.
  *   Add the sensor device to the MiHome app.
  *   After setting up, a file pairings.txt will be created in /devicestorage/vevs/logs/misc/pairings.txt. Open this file. The encryption key you need is called Bindkey. Also make a note of the corresponding Mac. If the pairings.txt file isn't created, try an older version of MiHome mod.

jorpand, 2021-01-23:
Thank you very much!

It's really appreciated!

Jordi

rogodra, 2021-01-23:
> 
> […](#)

In the way that tonypascal comments, with the application of andorid. If you already have it linked, you must first unlink it and then link back.

Now you need to know how it can be used with esphome.

tonypascal, 2021-01-23:
If anyone is much better than me in programming, is there a way to get the Bluetooth decoding and description functions work in the Arduino IDE ?

TheGroundZero, 2021-02-06:
> > Personnaly, I used the custom Android app method. Work well: link here<https://github.com/custom-components/ble_monitor/blob/master/faq.md#my-sensors-ble-advertisements-are-encrypted-how-can-i-get-the-key> Download<https://www.kapiba.ru/2017/11/mi-home.html>

Which app version of the MiHome Mod did you use?
On my tablet, the `/devicestorage/vevs/` folder has no `logs` folder and the other folders (`devices`, `automations`, ...) are empty.

blackbile71, 2021-02-08:
@TheGroundZero
You have to manually make the /logs folder!!!
After that delete the device pair again, and there will be the files in the logs directory!!!

TheGroundZero, 2021-02-08:
> 
> 
> @TheGroundZero
> You have to manually make the /logs folder!!!
> After that delete the device pair again, and there will be the files in the logs directory.
> The apk has permission to write to the internal memory!

Got it :)

Now how do I configure it in ESPHome (in Home Assistant)?

blackbile71, 2021-02-09:
Sorry, my latest CGG1 which I received yesterday from Aliexpress, has "Goose" on a sticker on the box, BUT the fw version not 1.5 only 1.1.xxx

So I don't have the bindkey to work...

TheGroundZero, 2021-02-09:
> 
> 
> Sorry, my latest CGG1 which I received yesterday from Aliexpress, has "Goose" on a sticker on the box, BUT the fw version not 1.5 only 1.1.xxx
> 
> So I don't have the bindkey to work...

Got it to work by using `esphome_version: 'Flameeyes:cgg1-enc'`
https://github.com/esphome/esphome-docs/pull/876

Patrryk, 2021-02-14:
> 
> 
> > Sorry, my latest CGG1 which I received yesterday from Aliexpress, has "Goose" on a sticker on the box, BUT the fw version not 1.5 only 1.1.xxx
> > So I don't have the bindkey to work...
> 
> Got it to work by using `esphome_version: 'Flameeyes:cgg1-enc'`
> [esphome/esphome-docs#876](https://github.com/esphome/esphome-docs/pull/876)

Hi,

Where or in which file I can change this parameter `esphome_version: 'Flameeyes:cgg1-enc'`

TheGroundZero, 2021-02-14:
> > > Sorry, my latest CGG1 which I received yesterday from Aliexpress, has "Goose" on a sticker on the box, BUT the fw version not 1.5 only 1.1.xxx
> > > So I don't have the bindkey to work...
> > 
> > 
> > Got it to work by using `esphome_version: 'Flameeyes:cgg1-enc'`
> > [esphome/esphome-docs#876](https://github.com/esphome/esphome-docs/pull/876)
> 
> Hi,
> 
> Where or in which file I can change this parameter `esphome_version: 'Flameeyes:cgg1-enc'` ? I cannot find it.

You need to set this in the add-on configuration in Home Assistant:  
https://sequr.be/blog/2021/01/home-automation-room-temperature-monitoring-using-xiaomi-temperature-sensor-esp32-and-esphome/#esphome-1

Patrryk, 2021-02-14:
> You need to set this in the add-on configuration in Home Assistant: [...]

Hi, unfortunately there is no field in my addon. I have the latest version installed (1.16.1)

[image]

TheGroundZero, 2021-02-14:
> Hi, unfortunately there is no field in my addon. I have the latest version installed (1.16.1)
>
> [image]

This is what I see
[image]

[image]

Perhaps it depends on how you installed HA/ESPHome?
I installed HassOS as a Proxmox VM.

Try using the yaml?

```yaml
ssl: true
certfile: fullchain.pem
keyfile: privkey.pem
esphome_version: 'Flameeyes:cgg1-enc'
relative_url: ''
```

Patrryk, 2021-02-14:
I have it installed on VM but on Virtualization Station on QNAP. All other addons have this possibility of configuration. Could You inform me where the configuration yaml file for ESPhome:

1) should be localized ?
2) what should be the name of the file?
3) what is the exact content o f Your file?

I will make my own separate file, I hope it ESPHome will read it.

TheGroundZero, 2021-02-14:
> I have it installed on VM but on Virtualization Station on QNAP. All other addons have this possibility of configuration. Could You inform me where the configuration yaml file for ESPhome:
> 
>     1. should be localized ?
> 
>     2. what should be the name of the file?
> 
>     3. what is the exact content of Your file?
> 
> 
> I will make my own separate file, I hope ESPHome will read it.

Can't help you with that, I'm afraid.
It's probably in one of the "higher" level folders used by/shared with the ESPHome container.

I suppose you don't have this menu option at the top-right?
[image]

Patrryk, 2021-02-14:
I have installed (dev) version of ESPHome, now the Configuration Tab is visible.

tonypascal, 2021-03-17:
I installed you version. nice !
it seems to be working pretty wel 
[image]
I now have 4 CGG1 encrypted temp / humidity sensors and 1 not encrypted working together !
good job

mario-tux, 2021-07-08:
Sorry for hijacking the report but I got a simple question: is the CGG1H (HomeKit) version supported in any way by esphome? I got one from Amazon (asking for a CGG1) but I'm wondering if I got any change to use it.