esphome/issues #1660: TTGO t-camera ESP32 WROVER - web/api/ota stops working after enabling esp32_ble_tracker

PiotrPolewczak, 2020-11-29:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**
Hass.io on docker

**ESP (ESP32/ESP8266, Board/Sonoff):**
TTGO t-camera ESP32 WROVER

**ESPHome version (latest production, beta, dev branch)**
ESPHome version like v1.15.3 and 1.16-dev

**Affected component:**
web/ota/api

**Description of problem:**
web/api/ota does not start after enabling esp32_ble_tracker

in debug we have : 

[E][AsyncTCP.cpp:1273] begin(): failed to start task

**Problem-relevant YAML-configuration entries:**
```yaml
substitutions:
  devicename: ttgocam
  friendly_name: ttgocam

esphome:
  name: $devicename
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_psk

web_server:
  port: 80

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

binary_sensor:
  - platform: gpio
    pin: GPIO33
    name: $friendly_name PIR
    device_class: motion

  - platform: gpio
    pin:
      number: GPIO34
      mode: INPUT_PULLUP
      inverted: True
    name: $friendly_name Button

  - platform: status
    name: $friendly_name Status

esp32_ble_tracker:

sensor:
  - platform: wifi_signal
    name: $friendly_name WiFi Signal
    update_interval: 10s
  - platform: uptime
    name: $friendly_name Uptime
    
[... 181 lines cut ...]
    largest_free_block 4696 alloc_blocks 60 free_blocks 3 total_blocks 63
  At 0x3ffe4350 len 113840 free 52336 allocated 61440 min_free 47212
    largest_free_block 52336 alloc_blocks 1 free_blocks 1 total_blocks 2
  Totals:
    free 4251640 allocated 185164 min_free 4241772 largest_free_block 4194252

after api task creating 
	
Heap summary for capabilities 0x00000004:
  At 0x3f800000 len 4194303 free 4194252 allocated 0 min_free 4194252
    largest_free_block 4194252 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x3ffaff10 len 240 free 0 allocated 120 min_free 0
    largest_free_block 0 alloc_blocks 5 free_blocks 1 total_blocks 6
  At 0x3ffb6388 len 7288 free 0 allocated 6920 min_free 0
    largest_free_block 0 alloc_blocks 21 free_blocks 0 total_blocks 21
  At 0x3ffb9a20 len 16648 free 16 allocated 15308 min_free 16
    largest_free_block 0 alloc_blocks 79 free_blocks 4 total_blocks 83
  At 0x3ffc8a90 len 95600 free 80 allocated 92244 min_free 56
    largest_free_block 12 alloc_blocks 199 free_blocks 12 total_blocks 211
  At 0x3ffe0440 len 15072 free 4584 allocated 9488 min_free 236
    largest_free_block 4540 alloc_blocks 59 free_blocks 3 total_blocks 62
  At 0x3ffe4350 len 113840 free 35928 allocated 77832 min_free 35928
    largest_free_block 35928 alloc_blocks 2 free_blocks 1 total_blocks 3
  Totals:
    free 4234860 allocated 201912 min_free 4230488 largest_free_block 4194252

brumela, 2020-12-20:
I can confirm this. My setup is AI Thinker ESP32-CAM;
board = esp32cam  
ESPHome version: 1.15.3

```
esphome:
  name: $devicename
  platform: ESP32
  board: esp32cam

api:
ota:

#esp32_camera:
#   external_clock:
#     pin: GPIO0
#     frequency: 20MHz
#   i2c_pins:
#     sda: GPIO26
#     scl: GPIO27
#   data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
#   vsync_pin: GPIO25
#   href_pin: GPIO23
#   pixel_clock_pin: GPIO22
#   power_down_pin: GPIO32
#   name: $devicename
#   vertical_flip: false
#   horizontal_mirror: false

esp32_ble_tracker:

sensor:
  - platform: ble_rssi
    mac_address: $flora2Mac
    name: "${flora2} RSSI value"

  - platform: xiaomi_hhccjcy01
    mac_address: $flora2Mac
    temperature:
      name: "${flora2} Temperature"
    moisture:
      name: "${flora2} Moisture"
    illuminance:
      name: "${flora2} Illuminance"
    conductivity:
      name: "${flora2} Soil Conductivity"

```

It works either _**with esp32_camera:**_ or _**esp32_ble_tracker:**_ configuration, but not with both. If both are enabled I get 
`[E][AsyncTCP.cpp:709] connect(): failed to start task`

buxtronix, 2021-03-23:
Not enough ram on the device to support both BLE and camera I'm afraid.

closed 2021-08-04 by stale[bot]