esphome/issues #1665: problem uploading esp8266

alecarma2002, 2020-12-02:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESP (ESP32/ESP8266, Board/Sonoff):**

**ESPHome version (latest production, beta, dev branch)**

**Affected component:**

**I can't upload my YAML code into the nodeMCU bord. Sometimes nothing goes wrong and the everything works fine but after a bit the node goes offline and ESPhome can't comunicate no more with the board:**

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
INFO Successfully compiled program.
INFO Running:  esptool.py --before default_reset --after hard_reset --baud 460800 --chip esp8266 --port /dev/ttyUSB0 write_flash 0x0 /data/sdfghj/.pioenvs/sdfghj/firmware.bin
esptool.py v2.8
Serial port /dev/ttyUSB0
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP8266: Timed out waiting for packet header
INFO Upload with baud rate 460800 failed. Trying again with baud rate 115200.
INFO Running:  esptool.py --before default_reset --after hard_reset --baud 115200 --chip esp8266 --port /dev/ttyUSB0 write_flash 0x0 /data/sdfghj/.pioenvs/sdfghj/firmware.bin
esptool.py v2.8
Serial port /dev/ttyUSB0
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP8266: Timed out waiting for packet header

```

**Additional information and things you've tried:
I tried to ad a static IP after the few succesfull first uploads via usb connection to the raspberry but after a few minutes running a got stuck with the same problem every time.**

Erriez, 2020-12-26:
@alecarma2002, Can you try to upload with GPIO15 pin (D0) floating?

raidnet-ms, 2021-01-06:
Same problem after latest esphome update.

I'm able to update with OTA while updating with USB on raspberry gives me same error.

If i compile the binary, download it on windows and flash it with esphome flasher, in this case the node won't connect to wifi.

All my previous node are updating fine with ota.

closed 2021-05-21 by stale[bot]