esphome/issues #1668: ESP32 Deep Sleep 1.6mA consumption

Papadoma, 2020-12-03:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Hass in docker

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 WRoom dev board with ch304g USB2serial

**ESPHome version (latest production, beta, dev branch)**

ESPHome version 1.15.3

**Affected component:**

ESP core

**Description of problem:**

I am trying to make a bed persence sensor that wakes up after an opamp triggers the ESP32, which is in deep sleep
The ESP32 is set to wake up periodically to update the battery state on HASS (currently set every minute but will be reduced)
The sensor is powered by a 18650 shield and also has an INA219 attached to it, which is powered down during deep sleep.

I have monitored the consumption of the ESP32 during deep sleep and it is 1.6mA. The total consumption of the sensor is 2.5mA in deep sleep. The opamp with the surounding components spends 0.7mA. The ESP32 with only the 18650 shield attached is 1.8mA (so the shield itself contributes only 0.2mA to the overall consumption). The sensor is powered via the 3.3V rail, so no current is caused by and 5V to 3.3V conversion.

I am trying to figure out why the ESP32 module is wasting 1.6mA instead of a few uA. I have reached the conclusion that the ESPhome does not power down certain modules within the ESP32

**Problem-relevant YAML-configuration entries:**
```yaml

esphome:
  name: bed_presence
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid:  
  password: 
  fast_connect: True
  power_save_mode: HIGH
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Bed Presence Fallback Hotspot"
    password:  
  manual_ip:
    static_ip:  
    gateway:  
    subnet:  
captive_portal:

# Enable logging
logger:

ota:

deep_sleep:
  id: deep_sleep_1  
  wakeup_pin: 
    number: 39
    inverted: True
  wakeup_pin_mode: INVERT_WAKEUP
  run_duration: 1s
  sleep_duration: 60s

switch:
  - platform: gpio
    name: "Power Sensor enable"
    pin: 5
    internal: true
    id: weight_sensor
    #inverted: true
    restore_mode: ALWAYS_ON 
 
i2c:
  sda: 21
  scl: 22 
  scan: False
  frequency: 800kHz
    
sensor:
  - platform: ina219
    address: 0x40
    shunt_resistance: 0.1 ohm
    bus_voltage:
      name: "INA219 Bus Voltage"
      accuracy_decimals: 2
    max_voltage: 4.5V
    max_current: 0.5A
    update_interval: 0.3s
  - platform: adc
    pin: 36
    attenuation: 11db
    name: "Weight"
    update_interval: 0.3s

binary_sensor:
  - platform: gpio
    pin: 39
    name: "Bed Occupancy"
    device_class: occupancy
    filters:
      - invert:

mqtt:
  broker:  
  username:  
  password:  
  birth_message:
    topic: bedroom/status
    payload: online
  will_message:
    topic: bedroom/status
    payload: online  
  on_message:
    - topic: bedroom/ota_mode
      payload: 'ON'
      then:
        - deep_sleep.prevent: deep_sleep_1
    - topic: bedroom/ota_mode
      payload: 'OFF'
      then:
        - deep_sleep.enter: deep_sleep_1  

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

I have tried to isolate the consumption by removing stuff. First I removed everything apart from the 18650 shield and the ESP32. I measured consumption at the battery and at the 3.3V pin of the ESP32 and found them to be 1.8mA and 1.6mA respectively. Then I added the INA219 sensor which didn't contribute anything as during deep sleep is powered down. Then I added the opamp which brought the consumption up to 2.5mA, thus concluding that the opamp requires 0.7mA

I then focused on the ESP32 board. I've removed the shield to try putting a pullup resistor on the memory (I read somewhere that another board had this design flaw and thought it might be the case). I tried removing the LDO (the 3.3V was backfeeding on the 5V rail so I thought removing the LDO might fix it). I tried removing the USB2serial (someone else was complaining that the USB2serial wasn't sleeping). In every case, I didn't see any reduction. 

Therefore I deduced that something's wrong with the deep sleep implementation of ESPhome for the ESP32.

Papadoma, 2021-04-09:
Don't mark it as stale. It hasn't been resolved yet

jbrepogmailcom, 2021-04-10:
Apparently the ```wakeup_pin``` in ```deep_sleep:``` is the problem. After removing it the deep sleep is 11 microAmpers as it shoud be.

Papadoma, 2021-04-10:
> Apparently the `wakeup_pin` in `deep_sleep:` is the problem. After removing it the deep sleep is 11 microAmpers as it shoud be.

Indeed, yet this is only useful if you want an ESP that wakes upon regular time intervals
If instead you want to wake up during an external interrupt, it's too wasteful

Papadoma, 2021-05-13:
Not sure if this is being worked on, but I did find that there is a function for ESP32 which allows for the ULP to do the ADC, adc1_ulp_enable(). Searching around I found indications that when the ADC is enabled before sleep, it causes extra consumption. Some suggested that putting adc_power_off() before going to sleep, reduces consumption dramatically. I assume that the ADC needs to be on for the wakepin detection to work. 

Perhaps this is an indication towards a solution, as the ULP will require much less power than the main processors.

drpetersen, 2021-05-13:
With the fix suggested [here](https://github.com/espressif/arduino-esp32/issues/1113#issuecomment-805273242) (i.e., call `adc_power_off` before entering deep sleep), I consistently measure a deep sleep current of less than 0.1 mA (unfortunately, that is as accuracte as my cheap multimeter gets) on (1) a Firebeetle ESP32, (2) a ESP32-Cam with LDO soldered out, and (3) a DevKitC V2 with LDO soldered out and CP2102 deactivated by disconnecting pins 6 and 7.

The `wakeup_pin` still works even then, which seems to make sense since it does not require an ADC (you can even use pins which do not have ADC capability for wakeup). I suspect that `adc1_ulp_enable` would only be needed if you wanted to use the ADC in ULP code, which, as far as I understand, is not supported by ESPHome anyway.

Papadoma, 2021-05-13:
> The `wakeup_pin` still works even then, which seems to make sense since it does not require an ADC (you can even use pins which do not have ADC capability for wakeup). I suspect that [...]

It may not supported, but you can still write custom components and call ESP32 functions, so I guess it's not unrealistic to use the ULP in ESPhome. I haven't tried it though.

e-tinkers, 2021-05-22:
`adc_power_off()` is [depreciated](https://github.com/espressif/esp-idf/blob/1067b28707e527f177752741e3aa08b5dc64a4d7/components/driver/include/driver/adc_common.h#L155-L162). It is recommended to use adc_power_release().

drpetersen, 2021-05-23:
Thank you, @e-tinkers, I wasn't aware of that. Still, it doesn't help me turning the (admittedly) quick and dirty fix into a proper solution. Calling `adc_power_release()` should obviously be done by (or after checking with) the very component that had called `adc_power_acquire()` in the first place. I'm at a complete loss figuring out which component that might be, though. None of the `adc_power_*` functions are used anywhere in ESPHome, as far as I can see.

```
~ $ grep -ri adc_power .local/lib/python3.9/site-packages/esphome/
.local/lib/python3.9/site-packages/esphome/components/deep_sleep/deep_sleep_component.cpp:  adc_power_off();
```

The only line found was patched in by myself.

Of course I can simply replace that line with `adc_power_release()`. But for one thing, I couldn't be sure that ADC power will actually be switched off then, because the usage counter might still remain positive. And what is more, this defeats the whole purpose of ensuring that you're supposed to only switch ADC power off once it's no longer needed.

So (as you might already have noticed :rofl:), I have no clue how to fix this properly.

e-tinkers, 2021-05-24:
@drpetersen, Sorry my bad, I should made it clearly. Personally I was using `adc_power_off` as suggested by [#1113](https://github.com/espressif/arduino-esp32/issues/1113#issuecomment-805273242) to address the issue of higher than normal current when going to deep sleep after using WiFi, and so far it works well. It is only recently when I upgrade the ESP32 firmware, I noticed the compiler warming notice that `adc_power_off` is deprecated. I guess you still can use `adc_power_off` if it is works for you. I have not yet test the `adc_power_release()` yet.

Papadoma, 2021-05-26:
I confirm that adding the `adc_power_off` brings the consumption down to 1.2mA total for my case
Since the opamp requires 0.7mA and the shield 0.2mA, that leaves 0.3mA for the ESP32, which is probably because the board has still its regulator and UART chip connected.

krikk, 2021-07-11:
can someone give me a hint how to integrate this `adc_power_off` fix into a normal esphome project, perhaps someone can share his code... would be very nice, because i'm only a beginner in esphome :)

dekiesel, 2021-08-11:
> can someone give me a hint how to integrate this `adc_power_off` fix into a normal esphome project, perhaps someone can share his code... would be very nice, because i'm only a beginner in esphome :)

Second this, this would be really nice to have!

drpetersen, 2021-08-12:
I have simply patched this into my `~/.local/lib/python3.9/site-packages/esphome/components/deep_sleep/deep_sleep_component.cpp`, like so:

```
#include "deep_sleep_component.h"
#include "esphome/core/log.h"
#include "esphome/core/application.h"

// ADDED THE FOLLOWING LINE
#include "driver/adc.h"

namespace esphome {
namespace deep_sleep {

…

  App.run_safe_shutdown_hooks();

#ifdef ARDUINO_ARCH_ESP32
  if (this->sleep_duration_.has_value())
    esp_sleep_enable_timer_wakeup(*this->sleep_duration_);
  if (this->wakeup_pin_.has_value()) {
    bool level = !(*this->wakeup_pin_)->is_inverted();
    if (this->wakeup_pin_mode_ == WAKEUP_PIN_MODE_INVERT_WAKEUP && (*this->wakeup_pin_)->digital_read())
      level = !level;
    esp_sleep_enable_ext0_wakeup(gpio_num_t((*this->wakeup_pin_)->get_pin()), level);
  }
  if (this->ext1_wakeup_.has_value()) {
    esp_sleep_enable_ext1_wakeup(this->ext1_wakeup_->mask, this->ext1_wakeup_->wakeup_mode);
  }
  // ADDED THE FOLLOWING LINE
  adc_power_off();
  esp_deep_sleep_start();

…
```

(Of course this only works for ESPHome installed with `pip3`. Also, you might need to find the location of your installed esphome package with `python -c 'import esphome; print(esphome.__path__)'` and adjust the path of the file to be patched accordingly. With a docker install, you would need to do something similar inside the container.)

The drawback of this approach is that you have to re-apply the patch after every ESPHome upgrade. It might be easier to put the needed invocations into the shutdown code of your `.yaml` config file instead, like so:

```
esphome:

…

  on_shutdown:
    then:
      - lambda: "adc_power_off();"
```

Or, if you want to use the newer `adc_power_acquire` / `adc_power_release` semantics instead of the deprecated `adc_power_off`, you would insert:

```
esphome:

…

  on_boot:
    then:
      - lambda: "adc_power_acquire();"
  on_shutdown:
    then:
      - lambda: "adc_power_release();"
```

Haven't actually tried this latter aproach myself yet, though.

Micha70, 2022-04-18:
Hello, I have a question. How can I use adc_power_acquire/adc_power_release? If I try to use in my ESP home I get following error: 
`
Compiling /data/gartensensor/.pioenvs/gartensensor/src/main.cpp.o

/config/esphome/gartensensor.yaml: In lambda function:

/config/esphome/gartensensor.yaml:13:7: error: 'adc_power_acquire' was not declared in this scope

   13 |       lambda: 'adc_power_acquire();'
      |       ^~~~~~~~~~~~~~~~~

/config/esphome/gartensensor.yaml: In lambda function:

/config/esphome/gartensensor.yaml:17:7: error: 'adc_power_release' was not declared in this scope

   17 |       lambda: 'adc_power_release();'
      |       ^~~~~~~~~~~~~~~~~

*** [/data/gartensensor/.pioenvs/gartensensor/src/main.cpp.o] Error 1`

Do I need to install / declare something in addition? I'm using ESPhome with homeassistant.

justind000, 2023-02-16:
I ran into this issue as well. I tried the lambda approach as suggested above, but couldn't figure out the proper way to include the needed includes. 

I put adc_power_acquire();/adc_power_release(); in a custom componenet which worked to compile, but still gave me the same current use. I then put adc_power_off(); which gave me the depracation errors, but dropped current use to the expected level.

simonjowett, 2023-02-17:
Same issue here too,  I get the same error when trying to place the commands in YAML.

Gabriel-Gardin, 2023-03-20:
Hello, 
I have also tested using adc_power_release() and adc_power_off(), and only adc_power_off() worked, droping the current consumption by 1.6mA during deep sleep.

noeldtz, 2023-05-14:
@Gabriel-Gardin could you please share your code to achieve including the adc_power_off() in a custom component? I'm also running into the problem of high power usage (1.3 mA) in deep sleep, using a firebeetle DRF0654.

maframan, 2023-06-22:
Hi, can you share the final yaml code because I receive the same above error? Thanks a lot

Micha70, 2023-06-24:
Hello @maframan ,

Im using on shutdown:
```

  on_shutdown:
    then:
      - logger.log: Shutdown called
      - lambda: |-
          pinMode(4, OUTPUT);
          digitalWrite(4, HIGH);
          gpio_deep_sleep_hold_en();
          adc_power_release(); 
      - output.set_level:
          id: esp32_birdbox_light
          level: "100%"

```

This was working for me reaching the lowest possible sleep current. But I mean was more than 1.6mA...

maframan, 2023-06-24:
> Hello @maframan ,
>
> Im using on shutdown:
>
> ```
>
>   on_shutdown:
>     then:
>       - logger.log: Shutdown called
>       - lambda: |-
>           pinMode(4, OUTPUT);
>           digitalWrite(4, [...]

Thanks a lot for your help!!!

maframan, 2023-06-24:
> This was working for me reaching the lowest possible sleep current. But I mean was more than 1.6mA...

Hi, i've added your suggested configuration but the power consumption is too high (about 50mA nut the green led blink continuos), without your confiuration the power was abot 5,9mA.
What's not correct in my configuration?

esphome:
  name: power-test
  friendly_name: Power Test
  on_shutdown:
    then:
      - logger.log: Shutdown called
      - lambda: |-
          pinMode(4, OUTPUT);
          digitalWrite(4, HIGH);
          gpio_deep_sleep_hold_en();
          adc_power_release(); 
          
esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino

# Enable logging
logger:
   level: DEBUG
# Enable Home Assistant API
api:
  encryption:
    key: "xxxxxxxxxxxxxxxxxxxxxxxxxxx"

ota:
  password: "xxxxxxxxxxxxxxxxxxxxxxxxx"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Power-Test Fallback Hotspot"
    password: "xxxxxxxxxxxxxxxxxx"

captive_portal:

deep_sleep:
  run_duration: 10sec
  wakeup_pin:
    number: GPIO04
    inverted: False
    mode: INPUT_PULLUP
  wakeup_pin_mode: KEEP_AWAKE
  id: deep_sleep_1

sensor:
  - platform: adc
    pin: GPIO01
    name: "Battery"
#    update_interval: 5s
    attenuation: 0db
    raw: true
    filters:
      - multiply: 0.00095238

web_server:
  port: 80
  auth:
    username: admin
    password: 'Andale@79'

Micha70, 2023-06-25:
Hello @maframan ,
you have something connected on GPIO4?
This looks for me it goes never to sleep.

I'm using also different board.
```
esphome:
  name: "esp32-birdbox"
  platform: ESP32
  board: esp32dev

output:

  - platform: ledc
    pin: GPIO4
    channel: 2 # channel 1 is used for esp32_camera
    id: esp32_birdbox_light
    inverted: false

```

To be honest 5mA is already a quite good value I mean. I don't believe I have less. Unfortunately I can not measure. Is not accessible. But is working on my side since about 1 year, powered by a small Lipo and changed by solar.

[image]