esphome/issues #1669: ESPHome v1.15.3: ESP32 DOIT devkit v1 - Flashing with ESPHome --> no network: Auth Expired

XcaliburJeroen, 2020-12-04:
ESPHome: v1.15.3
Platform: ESP32
Board: esp32doit-devkit-v1

Installation on Raspberry Pi 4 (4Gb), ESPHome installed as docker container (no relation to my home-assistant install).
Affected component: wifi
See [https://esphome.io/components/wifi.html?highlight=wifi](https://esphome.io/components/wifi.html?highlight=wifi) for details

**Context:**
I have some ESP32 DOIT DevKit v1's which I programmed with Arduino IDE.Networking, LEDs, DHT22, MQTT, webserver, ... all goes fluent. With my arduino code, the connection is established almost immediately.

As the programming in Arduino IDE is very low level, I wanted to tryout the ESPHome way of working. I have a raspberry pi 4 (4Gb), equipped with Ubuntu Server 20.04, docker, docker-compose. Home Assistant is installed as a docker container. As there is no supervisor (no HASSIO install) available, I had to install the addons separately. So I also installed the esphome/esphome container.The ESPHome interface works fine. Editing and validating the yaml configurations goes fine.Compiling and flashing the configuration goes fine; at least I expect it to be fine as I don't get errors and the SUCCESS message is printed.

When I use a default setup with only the wifi setup, after flashing, the ESP32 starts up, but in the console log, I get

- initialising wifi
- scanning of wifi networks --> fineAll networks are shown, including the proper MAC addresses
- Credentials with which the connection is tried to be established are shown and are correct
- When the connection is actually made; I get an AUTH EXPIRED exception; almost immediately.
- ESP32 immediately tries to reconnect, but almost instantly gives back the same error AUTH_EXPIRED
- The loop to (re)scan the wifi network takes over; scanning goes fine, all data is correct; but the STA_CONNECTED fails.
- After a while the ESP32 goes to AP mode. In purple, it shows the configuration of the AP mode (fallback SSID and password); but immediately the ESP32 starts the STA mode again==> doesn't stay in the AP mode...
- On my router (which should get the DHCP requests and/or the wifi-session visible) NO indication at all is shown of my ESP32.
==> I expect that the wifi fails all together.

I enabled the very_verbose logging, but this doesn't give any further information except 2 additional/underlying error codes
I got error codes 5 - STA_DISCONNECTED and 2 - AUTH_EXPIRE

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: testtest
  platform: ESP32
  board: esp32doit-devkit-v1
#  platformio_options:
#    platform: espressif32@1.9.0

wifi:
  ssid: "MyWifiNetwork"
  password: "testtest"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Testtest Fallback Hotspot"
    password: "testtest"

captive_portal:

# Enable logging
logger:
  level: VERY_VERBOSE

# Enable Home Assistant API
api:
  password: "testtest"

ota:
  password: "testtest"
```

**Logs (if applicable):**

```
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[13:59:34][VV][esp-idf:000]: I (512) wifi: mode : sta (30:ae:a4:8f:36:5c)
[13:59:34]
[13:59:34][D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[13:59:34][D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[13:59:34][V][wifi_esp32:361]: Event: WiFi STA start
[... 100 lines cut ...]
Linking .pioenvs/testtest/firmware.elf
Building .pioenvs/testtest/firmware.bin
Retrieving maximum program size .pioenvs/testtest/firmware.elf
Checking size .pioenvs/testtest/firmware.elf
RAM:   [=         ]  12.9% (used 42304 bytes from 327680 bytes)
Flash: [=====     ]  47.5% (used 871534 bytes from 1835008 bytes)
====================== [SUCCESS] Took 97.92 seconds ======================
INFO Successfully compiled program.

### Error message:
[09:20:43][W][wifi_esp32:336]: Event: Disconnected ssid='xxxx' bssid=xxxx reason='Auth Expired'[09:20:43][W][wifi:451]: Error while connecting to network.[09:20:43][W][wifi:488]: Restarting WiFi adapter...[09:20:43][I][wifi:193]: WiFi Connecting to 'xxxx'...[09:20:45][W][wifi_esp32:336]: Event: Disconnected ssid='xxx' bssid=xxxx reason='Auth Expired'[09:20:45][W][wifi:451]: Error while connecting to network.[09:20:45][W][wifi:488]: Restarting WiFi adapter...<continuous over and over again, from time to time interrupted with the fallback AP mode, but that's only a second enabled, and doesn't stay 'on'>

I'm using framework espressif32@1.12.4 for the build. (should be latest)I'm using latest ESPHome version 1.15.3 (fresh install within a docker container)On internet there are also other people mentioning this issue. I didn't find a root cause and structural solution to this problem.

### I tried:

- I verified the wifi router + credentials.Other devices at home can connect fluently.Network coverage is fine
- I erased the flash memory completely before flashing the configuration againesptool.py --port /dev/ttyUSB0 erase_flash
- I updated/upgraded the platformio libraries withpython -m platformio platform update
   2 versions of framework expressif32 are available: 1.12.4 and 1.9.0I tried both by specifying the framework to use in the yaml config file (platformio_options)
- I reflashed the device with my (working) arduino code through the Arduino IDE 1.8.4 and again, the wifi is established almost immediately==> no faulty ESP, wifi library of arduino works fine.
- Restart of wifi router
- I tried specifying a manual_ip and a fast_connect: trueNo help (except that no wifi scanning is done; same error messages/behaviour)

--> it's clear to me that esphome and it's libraries are causing the problem. But where to start and how to fix this?

XcaliburJeroen, 2020-12-04:
My own web research leads to the next issues. Issue is quite common, but no structural solution exists.
See also: [https://www.reddit.com/r/Esphome/comments/g81cta/nodemcev2_flashed_under_home_assistant_wifi_issues/](https://www.reddit.com/r/Esphome/comments/g81cta/nodemcev2_flashed_under_home_assistant_wifi_issues/) (but topic is closed)

[https://community.home-assistant.io/t/esphome-not-connecting-anymore/174112](https://community.home-assistant.io/t/esphome-not-connecting-anymore/174112) Exact same behavior with same AUTH_EXPIRED error.

[https://github.com/esphome/issues/issues/936](https://github.com/esphome/issues/issues/936) (same error, same behaviour; no structural solution)

[https://github.com/esphome/issues/issues/1057](https://github.com/esphome/issues/issues/1057) (same error, same behaviour; no structural solution; topic closed without solution)

[https://github.com/esphome/issues/issues/1173](https://github.com/esphome/issues/issues/1173) (same error, same behaviour)

[https://github.com/esphome/issues/issues/455](https://github.com/esphome/issues/issues/455) (may be related; although quite old topic)

XcaliburJeroen, 2020-12-04:
Additional information:
This morning I tried different board definitions, as I was not completely sure which model I have (there are numerous versions around the world, and I bought it from China/overseas.

I tried:
board: esp32doit-devkit-v1
board: nodemcu-32s
board: esp32dev
board: node32s

The info printed on my ESP32 states:
SBC-NodeMCU-ESP32
JOY-IT Europe GmbH, Pascalstr 8 D-47506 NK-VLuyn
www.joy-it.net
On the backside of the board itself, ESP32 DEVKITV1 is printed. On the chip ESP-WROOM-32 is printed.
It's a 2x15 = 30pin board

On my Arduino IDE, I use DOIT DevKit v1 and all works fine.

All board configurations give the same AUTH_EXPIRE error. 
I'm completely puzzled and I'm blocked by this wifi error.

dervogt, 2020-12-08:
same is happening here. I am running in a roaming wifi setup with three AP, two from AVM in the Fritzbox Series (FB 5490 and FB Repeater 1750E in AP Mode) and one Sitecom WLX-2005 AP.
The two AVM APs are giving the Auth Expired message whenever the esp32 is trying to connect, the Sitecom WX access point just accepts the esp

dervogt, 2020-12-09:
I did some additional testing this morning with the esphomedashboard-upload, these are my observations:

Config yaml which does replicate the Auth Expired issue:
`esphome:
  name: esp32_schuppen
  platform: ESP32
  board: nodemcu-32s

wifi:
  #ssid: "Familie.Vogt"
  #password: "Max!2705"
  power_save_mode: none
  #fast_connect: True
  manual_ip:
    static_ip: 10.0.0.130
    gateway: 10.0.0.1
    subnet: 255.255.255.0
    dns1: 10.0.0.1
  networks:
  - ssid: "Familie.Vogt"
    password: "Max!2705"
    bssid: "CC:CE:1E:FC:1B:21"
    priority: 2
  - ssid: "Familie.Vogt"
    password: "Max!2705"
    bssid: "CC:CE:1E:2F:52:2B"
    priority: 1
  - ssid: "Familie.Vogt"
    password: "Max!2705"
    bssid: "64:D1:A3:4A:BF:15"
    priority: 0
# Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32 Schuppen Fallback Hotspot"
    password: "Max!2705"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
sensor:
  - platform: dht
    pin: GPIO14
    model: DHT11
    temperature:
      name: "Schuppen Temperature"
    humidity:
      name: "Schuppen Luftfeuchte"
    update_interval: 60s
  - platform: wifi_signal
    name: "Schuppen WiFi Signal"
    update_interval: 60s`

The following config does well work, same ESP32 device on both configs:

`esphome:
  name: esp32_office
  platform: ESP32
  board: nodemcu-32s

wifi:
  ssid: "Familie.Vogt"
  password: "REDACTED"
  manual_ip:
    static_ip: 10.0.0.124
    gateway: 10.0.0.1
    subnet: 255.255.255.0
    dns1: 10.0.0.1

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32Test Office.Local"
    password: "REDACTED"

captive_portal:

# Enable logging
logger:
  level: debug

# Enable Home Assistant API
api:

ota:
remote_receiver:
  pin: 
    number: GPIO18
    inverted: True
  dump: 
    #- lg
    - nec
    #- panasonic
    #- jvc
    #- samsung
    #- sony
    #- rc5
    #- rc_switch
remote_transmitter:
  pin: 
    number: GPIO5
  carrier_duty_percent: 50%
#esp32_ble_tracker:
sensor:
  - platform: esp32_hall
    name: "Office Hall Sensor"
    update_interval: 60s
  - platform: duty_cycle
    pin: GPIO4
    name: "Duty Cycle Sensor GPIO4"
    update_interval: 60s
  - platform: duty_cycle
    pin: GPIO5
    name: "Duty Cycle Sensor GPIO5"
    update_interval: 60s
  - platform: wifi_signal
    name: "Office WiFi Signal Sensor"
    update_interval: 60s
  - platform: uptime
    name: "Office Uptime Sensor"
    filters:
      - lambda: return x / 3600;
    unit_of_measurement: "h"
  - platform: ultrasonic
    trigger_pin: GPIO2
    echo_pin: GPIO15
    pulse_time: 20us
    name: "Office Window"
    update_interval: 15s
    #timeout: 1m
#  - platform: xiaomi_hhccjcy01
#    mac_address: '80:EA:CA:89:4C:F6'
#    temperature:
#      name: "Garten Bodentemperatur"
#    moisture:
#      name: "Garten Feuchtigkeit"
#    illuminance:
#      name: "Garten Licht"
#    conductivity:
#      name: "Garten Leitfaehigkeit"
#    battery_level:
#      name: "Garten Batterie"
#binary_sensor:
#  - platform: remote_receiver
#    name: "Tristar Airco Power Status"
#    nec:
#      address: 0x20DF
#      command: 0x20DF
#  - platform: ble_presence
#    mac_address: 7C:76:68:F0:85:CC
#    name: "Huawei P30Pro BT"
#  - platform: ble_presence
#    mac_address: 10:32:7E:C1:07:AF
#    name: "Huawei P20Lite BT"
#  - platform: gpio
#      pin: 
#      number: GPIO4
#      name: "Window Sensor"
#  - platform: gpio
#    pin: GPIO4
#    name: "Living Room Window"
#    device_class: window
switch:
  - platform: template
    name: "Büro Airco Power Toggle"
    turn_on_action:
      remote_transmitter.transmit_nec:
        address: 0x20DF
        command: 0x20DF
  - platform: template
    name: "Büro Airco Temp Up"
    turn_on_action:
      remote_transmitter.transmit_nec:
        repeat: True
        address: 0x20DF
        command: 0xA05F
  - platform: template
    name: "Büro Airco Temp Down"
    turn_on_action:
      remote_transmitter.transmit_nec:
        address: 0x20DF
        command: 0x807F
  - platform: template
    name: "Büro Airco Mode"
    turn_on_action:
      remote_transmitter.transmit_nec:
        address: 0x20DF
        command: 0xB04F
  - platform: template
    name: "Büro Airco Speed"
    turn_on_action:
      remote_transmitter.transmit_nec:
        address: 0x20DF
        command: 0x609F
  - platform: template
    name: "Büro Airco Timer"
    turn_on_action:
      remote_transmitter.transmit_nec:
        address: 0x20DF
        command: 0x00FF`

on the first config the connect try fails with AUTH EXPIRES, on the second config it does not....when I am copy and pasting the wifi part of the first config into the second one, it does seem to just work fine. So my hypothesis is, that the settings from the non-working config cause something during the AUTH process. I am now sticking with the working WIFI config as a kind fo workaround, but it may be interesting what the devs would find here...

XcaliburJeroen, 2020-12-15:
I got the ESP32 working after a very very long time of stepwise improvement - far from 'configure and go'.
It appears that the wifi library for esphome is less stable/fast than on Arduino?
I literally had to move my esp32 to a few meters (without walls/floors) from the Access point. There is't working fine. Network is picked up fast.

Today, I try to move my second ESP32 from Arduino own-coding to the esphome approach Again, the same problem: No wifi connection is established. Again the ESP32 is near the AccessPoint but now there is 5m in between AND a thick concrete floor (35cm). But on these distances, the signal strength should still be ok to pick up by the esp32. There are other AP's nearby but as I force the said and password, it should pick the right SSID up.

I'm completely puzzles by the way of working of the wifi component on the ESP32.

`INFO Reading configuration /config/esp32_a.yaml...
INFO Detected timezone 'CET' with UTC offset 1 and daylight savings time from 28 March 02:00:00 to 31 October 03:00:00
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[21:20:41][D][light:265]: 'ESP32_A Blue LED' Setting:
[21:20:42][D][dht:048]: Got Temperature=24.1°C Humidity=61.7%
[21:20:42][V][sensor:013]: 'ESP32_A Temperature': Received new state 24.100000
[21:20:42][D][sensor:092]: 'ESP32_A Temperature': Sending state 24.10000 °C with 1 decimals of accuracy
[21:20:42][V][sensor:013]: 'ESP32_A Humidity': Received new state 61.700001
[21:20:42][D][sensor:092]: 'ESP32_A Humidity': Sending state 61.70000 % with 1 decimals of accuracy
[21:20:42][D][light:265]: 'ESP32_A Green LED' Setting:
[21:20:42][I][wifi:194]: WiFi Connecting to 'MyNetworkSSID'...
[21:20:42][V][wifi:196]: Connection Params:
[21:20:42][V][wifi:197]:   SSID: 'MyNetworkSSID'
[21:20:42][V][wifi:202]:   BSSID: Not Set
[21:20:42][V][wifi:220]:   Password: [redacted]
[21:20:42][V][wifi:227]:   Channel: Not Set
[21:20:42][V][wifi:233]:   Manual IP: Static IP=192.168.1.192 Gateway=192.168.1.1 Subnet=255.255.255.0 DNS1=0.0.0.0 DNS2=0.0.0.0
[21:20:42][V][wifi:237]:   Hidden: NO
[21:20:42][V][wifi_esp32:036]: Enabling STA.
[21:20:42][V][esp-idf:000]: I (25684) wifi: mode : sta (30:ae:a4:8f:1b:4c)
[21:20:42]
[21:20:42][D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[21:20:42][D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[21:20:42][V][wifi_esp32:361]: Event: WiFi STA start
[21:20:42][V][wifi_esp32:352]: Event: WiFi ready
[21:20:42][V][wifi_esp32:114]: Stopping DHCP client failed! 20485
[21:20:43][V][esp-idf:000]: I (25831) wifi: n:11 2, o:1 0, ap:255 255, sta:11 2, prof:1
[21:20:43]
[21:20:43][V][esp-idf:000]: I (25832) wifi: state: init -> auth (b0)
[21:20:43]
[21:20:44][V][esp-idf:000]: I (26835) wifi: state: auth -> init (200)
[21:20:44]
[21:20:44][V][esp-idf:000]: I (26836) wifi: n:11 0, o:11 2, ap:255 255, sta:11 2, prof:1
[21:20:44]
[21:20:44][D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[21:20:44][W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[21:20:44][W][wifi_esp32:386]: Event: Disconnected ssid='XCALI_DLINK' bssid=[redacted] reason='Auth Expired'
[21:20:44][W][wifi:485]: Error while connecting to network.
[21:20:44][W][wifi:522]: Restarting WiFi adapter...
[21:20:44][V][wifi_esp32:038]: Disabling STA.
[21:20:44][V][esp-idf:000]: I (26878) wifi: flush txq
[21:20:44]
[21:20:44][V][esp-idf:000]: I (26881) wifi: stop sw txq
[21:20:44]
[21:20:44][V][esp-idf:000]: I (26888) wifi: lmac stop hw txq
[21:20:44]
[21:20:44][D][WiFiGeneric.cpp:337] _eventCallback(): Event: 3 - STA_STOP
[21:20:44][V][wifi_esp32:365]: Event: WiFi STA stop
[21:20:44][I][wifi:194]: WiFi Connecting to 'MyNetworkSSID'...
[21:20:44][V][wifi:196]: Connection Params:
[21:20:44][V][wifi:197]:   SSID: 'MyNetworkSSID'
[21:20:44][V][wifi:202]:   BSSID: Not Set
[21:20:44][V][wifi:220]:   Password: [redacted]
[21:20:44][V][wifi:227]:   Channel: Not Set
[21:20:44][V][wifi:233]:   Manual IP: Static IP=192.168.1.192 Gateway=192.168.1.1 Subnet=255.255.255.0 DNS1=0.0.0.0 DNS2=0.0.0.0
[21:20:44][V][wifi:237]:   Hidden: NO
[21:20:44][V][wifi_esp32:036]: Enabling STA.
[21:20:44][V][esp-idf:000]: I (27036) wifi: mode : sta (30:ae:a4:8f:1b:4c)
[21:20:44]
[21:20:44][D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[21:20:44][D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[21:20:44][V][wifi_esp32:361]: Event: WiFi STA start
[... 45 lines cut ...]
[21:20:50][V][wifi_esp32:352]: Event: WiFi ready
[21:20:50][V][wifi_esp32:114]: Stopping DHCP client failed! 20485
[21:20:50][V][esp-idf:000]: I (33536) wifi: n:11 2, o:1 0, ap:255 255, sta:11 2, prof:1
[21:20:50]
[21:20:50][V][esp-idf:000]: I (33537) wifi: state: init -> auth (b0)
[21:20:50]
[21:20:51][D][light:265]: 'ESP32_A Blue LED' Setting:
[21:20:51][V][esp-idf:000]: I (34540) wifi: state: auth -> init (200)
[21:20:51]
[21:20:51][V][esp-idf:000]: I (34541) wifi: n:11 0, o:11 2, ap:255 255, sta:11 2, prof:1
[21:20:51]
[21:20:51][D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
[21:20:51][W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[21:20:51][W][wifi_esp32:386]: Event: Disconnected ssid='MyNetworkSSID' bssid=[redacted] reason='Auth Expired'
[21:20:51][W][wifi:485]: Error while connecting to network.
[21:20:51][W][wifi:522]: Restarting WiFi adapter...
[21:20:51][V][wifi_esp32:038]: Disabling STA.
[21:20:51][V][esp-idf:000]: I (34583) wifi: flush txq
[21:20:51]
[21:20:51][V][esp-idf:000]: I (34586) wifi: stop sw txq
[21:20:51]
[21:20:51][V][esp-idf:000]: I (34593) wifi: lmac stop hw txq
[21:20:51]
[21:20:51][D][WiFiGeneric.cpp:337] _eventCallback(): Event: 3 - STA_STOP
[21:20:51][V][wifi_esp32:365]: Event: WiFi STA stop`

XcaliburJeroen, 2020-12-15:
Further research reveals the following:
ESP32_1: connected to the AP; 3m, direct line of sight; -40-45 dB signal strength
ESP32_1: connected to the AP; 5m, 1 concrete floor 35cm, -74dB signal strength

ESP32_2: connected to the AP; 3m, direct line of sight; -40-45 dB signal strength
ESP32_2: connected to the AP; 5m, 1 concrete floor 35cm, -74dB signal strength

==> Why does the signal drop so fast that even no connection can be made?
Other devices (eg my cellphone has no problems whatsoever to connect to AP through the concrete floor even meters and meters away.

AP is an (old) DLINK DIR-655 with wifi enabled on auto-channel (6), with preconfigured WPA/PSK. Although it has some years, that should do nothing to its wifi broadcasting capabilities 802.11 g+n. (I only use it exclusively for some sensors pushing temperature/humidity (every 10 seconds) and some LEDs and relays (ad hoc)).

anotherhobby, 2021-01-24:
I just started having this same issue. I tried flashing multiple ESP32s and got the same result. I searched all over and could find reports but no fixes. The info above from @XcaliburJeroen solved my issue. As soon as I got them chip closer to a WiFi router, the errors went away.

Judge-Tread, 2021-02-25:
So i'm having this exact issue - having flashed multiple d1 mini clones over the last couple of days, with no problem, I suddenly started getting this issue when flashing. 

Oddly, the exact .bin files are now failing with this issue, and d1mini that have previously worked fine have started doing it if flashed.

XcaliburJeroen, 2021-02-26:
I can confirm that your network strength of the wifi clearly has to do with the mentioned errors.
When the esp32 was short to the wifi accesspoint, it worked.
It didn't work with a wall in between.
Recently, I added a 5GHz AC3000 wifi point and no connection errors whatsoever.
==> documentation and information of the "Auth expired" error is not Always clear or to the point. 
I lost a lot of time searching things out and still I didn't find a proper solution that is stable and resilient.

closed 2021-07-10 by stale[bot]