esphome/issues #1671: Mitsubishi Heat Pump will not shut off

RubyBailey, 2020-12-05:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

ESP32 - esp32doit-devkit-v1
**ESP (ESP32/ESP8266, Board/Sonoff):**

Version v1.15.3 per Documentation
**ESPHome version (latest production, beta, dev branch)**

**Affected component:**

Climate, specifically Mitsubishi Heat Pump

**Description of problem:**
We have 4 Mitsubishi Heat pump indoor units: MSZ-A24, MSZ-D36, MSZ-GE18 and MFZ-KA18NA. All turn On and are able to accept temperature commands. Three of the units can be turned OFF. The MFZ-KA18NA unit cannot be turned OFF. I suspect the challenge is somewhere in the IR command strings for these units and therefore in the mitsubishi.cpp file. I would like to test and troubleshoot the issue by customizing the mitsubishi.cpp file; and hoping to propose a solution to the issue. I cannot determine how to alter the .cpp file without ESPHome overwriting the file when it compiles the project. 
The README.txt file in /config/"name of yaml"/src/esphome folder says "ESPHome automatically populates the esphome/ directory, and any changes to this directory will be removed the next time ESPHome is run. For modifying esphome's core files, please use a development ESPHome install or use the custom_components folder." I have searched for on how to use the "development install" and how to use "custom_components" folder to modify the core for testing purposes, yet without success.  I do not want to change the ESPHome configuration core for any of my other yaml configurations.

ESPHome is installed on a Windows 10 desktop using VMWare Workstation with Ubuntu 18.04.5 LTS

Thanks for any suggestions and help in advance. I have enjoyed using ESPHome and would like to contribute to further development and refinements whenever I can.
**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

RubyBailey, 2020-12-07:
Hello again,
I did find https://esphome.io/guides/contributing.html and setup git as well as the development environment. I have downloaded VSC, added extensions for Python and C++ and the Platformio IDE.
I have forked and branched the github files for ESPHome and then edited the Mitsubishi Heatpump file (mitsubishi.cpp) so I can test it. 
My current issue is hat I cannot figure out how to compile ESPHome to then test the Mitsubishi HeatPump file for local testing.
Perhaps these are obvious question, but its just not clear to me from the documentation at  https://esphome.io/guides/contributing.html how to accomplish this task.

closed 2021-08-22 by stale[bot]

mihing, 2021-09-21:
Same issue. Mitsubishi MS-GF25VA cannot be turned OFF.

mihing, 2021-09-21:
This code turns off the Mitsubishi MS-GF25VA normally: 
`switch:
  - platform: template
    name: "AC OFF"
    id: ac_off
    turn_on_action:
      - remote_transmitter.transmit_raw:
          carrier_frequency: 38000
          code: [3426, -1734, 455, -1273, 444, -1274, 449, -415, 397, -473, 399, -467, 403, -1325, 451, -409, 485, -392, 452, -1276, 450, -1274, 448, -418, 420, -1306, 397, -473, 399, -472, 394, -1327, 453, -1273, 398, -470, 398, -1325, 398, -1330, 449, -422, 394, -471, 398, -1324, 451, -417, 395, -470, 396, -1330, 457, -409, 420, -452, 424, -450, 397, -473, 400, -466, 398, -468, 421, -453, 395, -472, 424, -444, 399, -469, 451, -412, 426, -442, 419, -453, 483, -395, 397, -470, 395, -475, 427, -440, 448, -412, 397, -472, 421, -443, 425, -443, 397, -469, 453, -412, 397, -472, 450, -424, 396, -476, 420, -1300, 399, -1322, 398, -471, 453, -415, 399, -468, 418, -451, 449, -1278, 446, -1274, 395, -472, 397, -472, 450, -417, 396, -473, 480, -396, 420, -449, 450, -1272, 398, -1326, 422, -447, 425, -1297, 486, -1255, 395, -471, 449, -418, 424, -447, 423, -440, 397, -467, 454, -415, 395, -471, 451, -418, 397, -1328, 426, -443, 448, -416, 394, -1329, 400, -1325, 397, -1330, 398, -471, 393, -1327, 451, -1274, 453, -415, 448, -417, 422, -443, 450, -413, 419, -452, 452, -416, 392, -471, 425, -439, 420, -448, 452, -416, 394, -473, 396, -471, 397, -471, 424, -438, 452, -412, 451, -416, 423, -441, 451, -417, 395, -469, 398, -470, 395, -475, 449, -417, 396, -470, 396, -472, 451, -414, 397, -473, 395, -471, 424, -439, 456, -408, 454, -413, 397, -469, 484, -392, 396, -470, 398, -470, 425, -440, 427, -440, 396, -472, 448, -415, 425, -443, 396, -472, 451, -415, 395, -472, 451, -421, 398, -474, 430, -433, 396, -470, 450, -417, 397, -472, 426, -441, 395, -1329, 394, -1334, 395, -1327, 398, -470, 396, -1331, 397, -471, 481, -391, 398, -473, 397]
      - remote_transmitter.transmit_raw:
          carrier_frequency: 38000
          code: [3593, -1737, 395, -1328, 451, -1273, 452, -415, 448, -420, 448, -415, 482, -1260, 396, -468, 397, -473, 422, -1299, 450, -1275, 424, -442, 398, -1327, 396, -468, 453, -416, 397, -1330, 396, -1327, 398, -474, 448, -1273, 451, -1273, 398, -468, 451, -416, 419, -1310, 454, -412, 397, -466, 454, -1274, 398, -469, 399, -471, 395, -468, 452, -414, 397, -468, 425, -441, 398, -472, 424, -445, 397, -470, 450, -415, 421, -448, 483, -392, 425, -444, 448, -417, 451, -416, 447, -416, 395, -468, 421, -448, 450, -418, 424, -439, 399, -472, 418, -452, 397, -470, 394, -472, 422, -442, 396, -469, 397, -1330, 395, -1327, 399, -469, 450, -417, 396, -474, 452, -418, 396, -1329, 448, -1275, 423, -445, 449, -418, 398, -474, 397, -473, 451, -412, 398, -471, 425, -1300, 397, -1326, 398, -475, 396, -1330, 419, -1309, 449, -415, 399, -479, 450, -418, 420, -460, 398, -471, 424, -437, 401, -469, 449, -421, 394, -1330, 427, -440, 426, -440, 396, -1331, 398, -1328, 395, -1327, 424, -438, 419, -1313, 425, -1298, 451, -417, 420, -449, 397, -470, 397, -469, 453, -420, 449, -411, 452, -418, 425, -439, 450, -416, 396, -471, 452, -414, 395, -471, 453, -411, 399, -470, 424, -446, 395, -471, 425, -440, 396, -472, 424, -443, 398, -472, 452, -417, 448, -414, 486, -395, 422, -446, 425, -438, 450, -414, 451, -410, 402, -473, 423, -443, 451, -415, 397, -470, 398, -468, 424, -442, 399, -474, 394, -472, 397, -468, 451, -418, 394, -472, 422, -437, 400, -472, 396, -468, 453, -417, 398, -473, 425, -441, 427, -440, 452, -417, 449, -420, 423, -444, 448, -414, 398, -1328, 396, -1329, 397, -1328, 397, -473, 449, -1273, 426, -440, 397, -472, 425, -441, 419]`

RubyBailey, 2021-09-24:
Hi @mihing,
1. What version of esphome are you running? The issue that I posted was fixed in pull request #1414 (Mitsubishi Heat Pump - Fixed default transmit_state to be generic instead of for a specific type of heat pump)
2. I tried to decode your transmit string, but the header does not match any of the Mitsubishi units that I have worked with. The first 5 bytes of the header are typically 0x23, 0xCB, 0x26, 0x01, 0x00 
3. As this specific issue has now been closed I suggest that you create a new issue, giving details of your system, esphome version and a detailed description of what is working and what is not working with your unit.