esphome/issues #1675: RC522_spi scans continiously if no tag is present

MatthiasLienhard, 2020-12-06:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip
**ESP (ESP32/ESP8266, Board/Sonoff):**

esp32 dev board
**ESPHome version (latest production, beta, dev branch)**

v1.15.3
**Affected component:**

https://next.esphome.io/components/binary_sensor/rc522.html

**Description of problem:**
Apparently everything works fine:
If the tag is placed on the reader it is recognized and the corresponding binary sensor gets triggered. When its removed, the binary sensor gets unset after a short delay. However, I noticed that the antenna gets powered in each iteration cycle, not only once per second as specified if no tag is present. This results in quite high power consumption ~250 mWatts (without verbose logging over serial which slows down the code), but if a tag is placed on the reader, this drops to acceptable values. 
I analysed the code and found in addition that the binary sensors are not actively turned off, but time out. 
The reader (at least the ones I tested, all on these cheep blue modules) behave a little bit unexpected during operation: If a tag is presented it gets recognized, but the next scan fails with STATUS_COLLITION - which is the same error code I get if no card is detected. 
I propose the attached solution to this: Uid gets an additional property last_seen, that gets set on successful scans. On first failed scan last_seen is unset, and on second failed scan the binary switch gets unset. I attached the implementation. Relevant lines are 185 in the rcc522_spi.h, the  RC522::loop() function lines 114-147 and in RC522BinarySensor::process to actively unset the binary sensor. 

**Problem-relevant YAML-configuration entries:**
```yaml
spi:
  clk_pin: GPIO18
  miso_pin: GPIO19
  mosi_pin: GPIO23

rc522_spi:
  cs_pin: GPIO2
  #reset_pin: GPIO5
  update_interval: 1s
  on_tag:
    then:
      - homeassistant.tag_scanned: !lambda 'return x;'

binary_sensor:
  - platform: rc522_spi
    uid: 80-37-B9-A4
    name: "NFC Card 1"
    id: 'nfc1'
[rc522_spi.cpp.txt](https://github.com/esphome/issues/files/5649638/rc522_spi.cpp.txt)
[rc522_spi.h.txt](https://github.com/esphome/issues/files/5649639/rc522_spi.h.txt)

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

rradar, 2020-12-14:
@glmnet are you by chance the code owner? There are no labels yet available for next/dev so I allowed me to mention you :wink:

glmnet, 2020-12-14:
Ahh thanks @rradar I don't know why labels don't exist.

@MatthiasLienhard It will be great to test your changes but I cannot get the code from those links

MatthiasLienhard, 2020-12-15:
Should I do a pull request? I am not sure how this works here, since the issues are in a separate repro.

glmnet, 2020-12-15:
Sure go ahead

closed 2021-04-28 by jesserockz