esphome/issues #1678: Can't we use Bluetooth and camera together?

linear82, 2020-12-08:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**ESPHome version (latest production, beta, dev branch)**

**Affected component:**

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: esp_cam
  platform: ESP32
  board: esp32cam

wifi:
  ssid: 
  password: 

# Enable logging
logger:

# Enable Home Assistant API
api:
  reboot_timeout: 0s
ota:

switch:
  - platform: restart
    name: veranda Restart

# ESP32-CAM
esp32_camera:
  external_clock:
    pin: GPIO0
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32
  vertical_flip: false
  horizontal_mirror: false

  name: "veranda"

# Flashlight
output:
  - platform: gpio
    pin: GPIO4
    id: gpio_4
light:
  - platform: binary
    output: gpio_4
    name: "veranda light"

esp32_ble_tracker:

sensor:
  - platform: xiaomi_lywsd02
    mac_address: ""
    temperature:
      name: "Computer ROOM Temperature"
    humidity:
      name: "Computer Room Humidity"

```

**Logs (if applicable):**

```
PASTE DEBUG LOG HERE

```

**Additional information and things you've tried:**

Can't I use Bluetooth and camera at the same time?

back0nline, 2021-03-06:
I have the same problem

linear82, 2021-03-07:
I can't use

buxtronix, 2021-03-23:
There is not enough memory on the device to enable both. It's one or the other I'm afraid.

linear82, 2021-03-23:
ok, thx...

carlosmgr, 2021-03-23:
So something changed becaused I had both enabled and working before.