esphome/issues #1679: Fallback AP Causes Crash when WiFi Signal Returns

shallax, 2020-12-09:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Home Assistant running in Docker

**ESP (ESP32/ESP8266, Board/Sonoff):**

Shelly PM 1 (ESP8266)

**ESPHome version (latest production, beta, dev branch)**

ESPHome v1.15.3

**Affected component:**

https://esphome.io/components/wifi.html and maybe https://esphome.io/components/captive_portal.html

**Description of problem:**
I have 24 ESP8266 devices (2 Shelly EMs, 21 Shelly PMs and a Sonoff RF Bridge). I had a complete power outage which brought down my network and all of the ESP8266 devices. Once power was restored, the Shellies powered on as normal by themselves (where they are configured as such) and my network infrastructure began to reinitialise.

I noticed that as the WiFi APs came online (Ubiquiti NanoHDs), some of the circuits attached to the Shellies briefly lost power causing whatever load was on the circuit to lose power and reboot. When I was finally able to access Home Assistant, I noticed that the "uptime" sensor indicated an uptime of just a few seconds (<60) which is much shorter than the time it took the network to come online.

Some Googling suggested this may be due to the WiFi and API reboot_timeout property in ESPHome. Even though I'm sure the entire process took less than the default of 15 minutes, I added the property to my config and set it to "15m". I then powered cycled my house to see if the issue was fixed. However, the problem persisted. I then changed reboot_timeout to "0ms" to effectively disable this behaviour. Again, the problem persisted.

I went through the pain of modifying all components that call App.reboot() (which I have instantiated in my configurations) to persist a flag to flash when App.reboot() was called. I did this using a call to "global_preferences.make_preference<bool>(this->get_object_id_hash(), true)" in each class after also making the components inherit from "Nameable". In my case, these components were: wifi_component, ota_component and api_server. I also modified the dump_config() method in each of these to print the value of that flag (after restoring it from flash). I tested this by forcing the components to set the flag after 60 seconds and then rebooting + reconnecting to the log to verify that "dump_config()" displayed the flag as set in the event of App.reboot() being called. I then powered cycled my house again. The issue occurred again on multiple (maybe all) devices. However, none of them indicated that this flag had been set. The logs of some devices also indicated that the "Last Boot was an unhandled reset," which, from looking at the code I understand to mean that the "OTA" component's safe mode functionality didn't run for long enough to flag the boot as successful meaning that it's highly unlikely an intentional "App.reboot()" was called in such a short space of time (the 120s default).

Finally, I decided to rip out "ap: ..." and "captive_portal:" configuration directives from a subset of devices (all of the Shelly PMs). From looking at the wifi_component.cpp file, it looks like these are the only things that could interfere with the bootup/ connectivity sequence in such a short space of time with the default settings of enabling the fallback AP in 1 minute. After performing yet another power cycle of my house, none of the Shelly PMs exhibited the brief "reset" behaviour. I checked the uptime sensors for the Shelly PM devices and they all uniformly reported around 680s, which was pretty much exactly how long since I switched the power back on. I looked at the other devices (Shelly EMs and Sonoff RFBridge) and these had uptimes of between 30 and 60 seconds indicating that they had definitely been rebooted unbidden.

I didn't find anything when Googling for this, so am raising this issue.

I haven't dug too deeply into the code, but it would seem that if the WiFi signal/ SSID expected by the device reappears after the fallback AP has been created and the captive portal activated, the device has an unexpected reboot.

**Problem-relevant YAML-configuration entries:**
This is the configuration that causes the problem. Simply removing the "ap:" and "captive_portal:" sections seem to fix it.

```yaml
substitutions:
  name: "shellypmupstairslights"
  friendly_name: "Shelly PM Upstairs Lights"
  short_name: "${friendly_name}"
  default_mode: RESTORE_DEFAULT_ON
  force_switch_back_on: "true"

# Everything below is normally included with an "include" directive so it can be shared with the other devices. However, for the sake of this bug report, I've just pasted it directly below
globals:
  - id: safety_mode
    type: bool
    restore_value: no
    initial_value: "false"
  - id: safety_mode_overheat
    type: bool
    restore_value: no
    initial_value: "false"

esphome:
  name: ${name}
  platform: ESP8266
  board: esp01_1m
  esp8266_restore_from_flash: true

wifi:
  ssid: !secret ssid
  password: !secret ssid_password
  fast_connect: true
  reboot_timeout: 1h

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${short_name} FB"
    password: !secret fallback_password

[... 242 lines cut ...]
[01:15:44][C][resistance:011]:   Configuration: DOWNSTREAM
[01:15:44][C][resistance:012]:   Resistor: 32000.00Ω
[01:15:44][C][resistance:013]:   Reference Voltage: 3.3V
[01:15:44][C][adc:026]: ADC Sensor 'shellypmupstairslights_source_sensor'
[01:15:44][C][adc:026]:   Unit of Measurement: 'V'
[01:15:44][C][adc:026]:   Accuracy Decimals: 2
[01:15:44][C][adc:026]:   Icon: 'mdi:flash'
[01:15:44][C][adc:031]:   Pin: 17
[01:15:44][C][adc:051]:   Update Interval: 10.0s
[01:15:44][C][status:034]: Status Binary Sensor 'Shelly PM Upstairs Lights Status'
[01:15:44][C][status:034]:   Device Class: 'connectivity'
[01:15:44][C][ota:032]: Over-The-Air Updates:
[01:15:44][C][ota:033]:   Address: shellypmupstairslights.local:8266
[01:15:44][C][ota:035]:   Using Password.
[01:15:44][C][ota:045]: OTA REBOOT WAS CALLED 1
[01:15:44][C][api:099]: API Server:
[01:15:44][C][api:100]:   Address: shellypmupstairslights.local:6053
[01:15:44][C][api:110]: API REBOOT UNKNOWN, probably not called
[01:15:44][C][wifi_signal.sensor:009]: WiFi Signal 'Shelly PM Upstairs Lights WiFi Signal'
[01:15:44][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dB'
[01:15:44][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[01:15:44][C][wifi_signal.sensor:009]:   Icon: 'mdi:wifi'
```

**Additional information and things you've tried:**

thomaspedley, 2020-12-09:
I should also note that the calls to App.reboot() don't seem to toggle the GPIO-connected relay. Certainly when I perform an OTA update, I don't notice any attempt to toggle the relay despite the final step in the update being a call to App.reboot(). This leads me to believe that it's very unlikely that its an intentional reboot that's causing this issue. It seems like the entire device is crashing and reinitialising, which I believe causes the GPIOs to reset to their default values briefly.

shallax, 2020-12-09:
Just to add, this would also explain why my TV/ AV centre would seemingly randomly reboot every few days.

I've installed a Shelly PM inside a wall socket pattress behind the TV unit. WiFi reception there is on the edge of losing connection a fair bit of the time (depending on which WiFi AP the Shelly decides to latch on to). It's definitely possible that it was losing wifi for >1 minute, then regaining it after the fallback AP had been started and therefore provoking the crash.

I've been reading so many reports of people having problems with switches randomly switching off/on and I'm really hoping this might be the answer to those people's problems.

shallax, 2021-04-08:
I suppose that’s one way to address the issue without addressing it.

jhenkens, 2021-06-23:
I had been playing around with the fallback ap / captive portal a few weeks ago, and uncovered some weirdness, specifically when WiFi is unavailable during boot up. My notes from the time said that captive portal will prevent the normal boot sequence from completing. However, if the ESP disconnects from WiFi rather than booting into no-WiFi, it will then launch the captive portal in a "post-boot" mode. Seems a bit strange to me, and I wouldn't be surprised if it's related.

I had begun poking around and attempting to change that behavior - to allow full boot up without any WiFi connections, captive portal only, but hadn't completed that. However, that's not necessarily specifically related. I'm trying to setup a test-bed with an ESP32 dev board and a travel router, so I can test my changes out easier, and I'll have to see if I can recreate what you are talking about with a minimal config.

parautenbach, 2022-11-21:
Old (unresolved) issue, but just adding my voice to it: I had a device in a spot where the WiFi is a bit spotty. The ESP32 itself is powered by battery. We have regular power outages which means the nearest AP to that ESP32 will go down if that happens and the signal is too weak to communicate reliably (and ESPHome remembers the AP it connected to so without a reboot it won't reconnect anyway). I have the captive portal enabled and I was just researching this issue when I came across this logged issue. I had the suspicion too that the captive portal is interfering with the reboot timeout. I couldn't find any other documentation on this.

cypherbits, 2024-05-04:
Same experience here with an ESP32-C3-supermini, latest ESPHome version and arduino framework, wifi strength low with ap and portal enabled: if the wifi disconnects and connects under 1 minute it's ok. On the other side if it is more than 1 minute, then the captive portal is enabled but it seems it doesn't restart after 15 minutes. If I reset it manually (button) it connects again fast.

cypherbits, 2024-05-04:
Looking at the ESPHome code I found this. Reboot timeout only works with AP disabled:
https://github.com/esphome/esphome/blob/c7c0d97a5ecbaa235636ba9e527f8c9ff7b4c4b0/esphome/components/wifi/wifi_component.cpp#L192C1-L192C57

ssieb, 2024-05-16:
The most recent comments here are unrelated to the original issue which was about a crash.  The documentation change is valid, but irrelevant to this issue.