esphome/issues #1681: Limit of Bluetooth BLE Devices / Crash during boot after firmware upload

ssahlender, 2020-12-09:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

pip3

**ESP (ESP32/ESP8266, Board/Sonoff):**

ESP32 / LilyGO ESP32 ... without Camera / Tjoranl
https://github.com/lewisxhe/esp32-camera-series/blob/master/docs/T_Joranl.md

**ESPHome version (latest production, beta, dev branch)**

1.15.3

**Affected component:**

esp32_ble_tracker
ruuvitag
xiaomi_hhccjcy01
xiaomi_lywsdcgq

**Description of problem:**
After firmware upload system crashes.
Running 2 ruuvitag, 3 xiaomi_lywsdcgq, 1xiaomi_hhccjcy01
Question: Is there a limit of bluetooth devices?
If i remove the comments for the third ruuvitag system crashes during boot

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: esp32ble1
  platform: ESP32
#  board: nodemcu-32s
#  board: az-delivery-devkit-v4
  board: esp32dev

wifi:
  ssid: "***"
  password: "***"
  use_address: esp32ble1.internal.lan

  manual_ip:
    static_ip: a.b.c.d
    gateway: d.e.f.g
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Test32 Fallback Hotspot"
    password: "****"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
#api:
#  password: "****"

ota:
  password: "****"

esp32_ble_tracker:

#exposure_notifications:
#  on_exposure_notification:
#    then:
#      - lambda: |
#          ESP_LOGD("main", "Got notification:");
#          ESP_LOGD("main", "  RPI: %s", hexencode(x.rolling_proximity_identifier).c_str());
[... 203 lines cut ...]
WARNING Decoded 0x40106921: nvs::Storage::findItem(unsigned char, nvs::ItemType, char const*, nvs::Page*&, nvs::Item&, unsigned char, nvs::VerOffset) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/nvs_storage.cpp:557
WARNING Decoded 0x40106d21: nvs::Storage::readMultiPageBlob(unsigned char, char const*, void*, unsigned int) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/nvs_storage.cpp:557
WARNING Decoded 0x40106e58: nvs::Storage::readItem(unsigned char, nvs::ItemType, char const*, void*, unsigned int) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/nvs_storage.cpp:557
WARNING Decoded 0x40105dd6: nvs_get_str_or_blob(unsigned int, nvs::ItemType, char const*, void*, unsigned int*) at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/nvs_api.cpp:521
WARNING Decoded 0x40106449: nvs_get_blob at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/nvs_api.cpp:525
WARNING Decoded 0x400f52a8: load_cal_data_from_nvs_handle at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/phy_init.c:428
WARNING Decoded 0x400f571c: esp_phy_load_cal_data_from_nvs at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/phy_init.c:457
WARNING Decoded 0x400f5ab2: esp_phy_load_cal_and_init at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/phy_init.c:613
WARNING Decoded 0x401593be: esp_bt_controller_enable at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bt.c:1433
WARNING Decoded 0x400f17d5: btStart at /home/stefan/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-bt.c:44
WARNING Decoded 0x400d37dd: esphome::esp32_ble_tracker::ESP32BLETracker::ble_setup() at /data/syncthing/data/esp/esphome/esp32ble1/src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.cpp:241
WARNING Decoded 0x400d3992: esphome::esp32_ble_tracker::ESP32BLETracker::setup() at /data/syncthing/data/esp/esphome/esp32ble1/src/esphome/components/esp32_ble_tracker/esp32_ble_tracker.cpp:241
WARNING Decoded 0x401ebdd5: esphome::Component::call_setup() at /data/syncthing/data/esp/esphome/esp32ble1/src/esphome/core/component.cpp:176
WARNING Decoded 0x401ebebd: esphome::Component::call() at /data/syncthing/data/esp/esphome/esp32ble1/src/esphome/core/component.cpp:176
WARNING Decoded 0x400df439: esphome::Application::setup() at /data/syncthing/data/esp/esphome/esp32ble1/src/esphome/core/application.cpp:38
WARNING Decoded 0x400e524e: setup() at /data/syncthing/data/esp/esphome/esp32ble1/src/main.cpp:1424
WARNING Decoded 0x400f2d33: loopTask(void*) at /home/stefan/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:14
WARNING Decoded 0x4008f0e1: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
[12:26:01]
[12:26:01]Rebooting...
[12:26:01]ets Jun  8 2016 00:22:57

```

**Additional information and things you've tried:**

ssahlender, 2020-12-09:
Small update from further tests.
Tried adding another 
`xiaomi_hhccjcy01`
Same result, esp crashes during boot
So it looks like there is a limit of 6 BLE devices

ssahlender, 2020-12-09:
Another update:
It doesn't depend on the board even with ESP32 board from AZdelivery it crashes during boot.

ssahlender, 2020-12-09:
Update 3:
Completely disabling the display block seems to solve the problem:
`i2c
font
display`

buxtronix, 2021-03-23:
That would be because when BLE is enabled, there is not enough free RAM to also support the display component. Can't do anything about it I'm afraid (except perhaps looking at migrating BLE to NimBLE).

closed 2021-08-04 by stale[bot]