esphome/issues #1682: Dallas sensor not working on GPIO36

lucagiudici, 2020-12-09:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Mac OS install

**ESP (ESP32/ESP8266, Board/Sonoff):**
NodeMCU-32s

**ESPHome version (latest production, beta, dev branch)**

1.15.3

**Affected component:**

Dallas sensor

**Description of problem:**
a Dallas sensor connected to GPIO36 is not recognized, Sensor not found, and the same setup work with other pins like GPIO32

**Problem-relevant YAML-configuration entries:**
```yaml

**NOT WORKING:** 

esphome:
  name: bread_test
  platform: ESP32
  board: nodemcu-32s
  
logger:
    level: VERBOSE

dallas:
  - pin: GPIO36

---

**WORKING**: 

esphome:
  name: bread_test
  platform: ESP32
  board: nodemcu-32s
  
logger:
    level: VERBOSE

dallas:
  - pin: GPIO32

```

**Logs (if applicable):**

```
**NOT WORKING:** 

INFO Successfully uploaded program.
INFO Starting log output from /dev/cu.usbserial-0001 with baud rate 115200
[16:23:33]\x85[\x8b\x95\xe9DIO, clock div:2
[16:23:33]load:0x3fff0018,len:4
[16:23:33]load:0x3fff001c,len:1044
[16:23:33]load:0x40078000,len:8896
[16:23:33]load:0x40080400,len:5828
[16:23:33]entry 0x400806ac
[16:23:33][I][logger:166]: Log initialized
[16:23:33][I][app:029]: Running through setup()...
[16:23:33][V][app:030]: Sorting components by setup priority...
[16:23:33][C][dallas.sensor:032]: Setting up DallasComponent...
[16:23:33][I][app:059]: setup() finished successfully!
[16:23:33][E][dallas.sensor:126]: Requesting conversion failed
[16:23:33][I][app:105]: ESPHome version 1.15.3 compiled on Dec  9 2020, 16:22:29
[16:23:33][C][logger:185]: Logger:
[16:23:33][C][logger:186]:   Level: VERBOSE
[16:23:33][C][logger:187]:   Log Baud Rate: 115200
[16:23:33][C][logger:188]:   Hardware UART: UART0
[16:23:33][C][dallas.sensor:072]: DallasComponent:
[16:23:33][C][dallas.sensor:073]:   Pin: GPIO36 (Mode: INPUT)
[16:23:33][C][dallas.sensor:074]:   Update Interval: 60.0s
[16:23:33][W][dallas.sensor:077]:   Found no sensors!

**WORKING**: 

INFO Successfully uploaded program.
INFO Starting log output from /dev/cu.usbserial-0001 with baud rate 115200
[16:26:26]:0x00,wp_drv\xc1
[16:26:26]mode:DIO, clock div:2
[16:26:26]load:0x3fff0018,len:4
[16:26:26]load:0x3fff001c,len:1044
[16:26:26]load:0x40078000,len:8896
[16:26:26]load:0x40080400,len:5828
[16:26:26]entry 0x400806ac
[16:26:26][I][logger:166]: Log initialized
[16:26:26][I][app:029]: Running through setup()...
[16:26:26][V][app:030]: Sorting components by setup priority...
[16:26:26][C][dallas.sensor:032]: Setting up DallasComponent...
[16:26:26][I][app:059]: setup() finished successfully!
[16:26:26][I][app:105]: ESPHome version 1.15.3 compiled on Dec  9 2020, 16:25:02
[16:26:26][C][logger:185]: Logger:
[16:26:26][C][logger:186]:   Level: VERBOSE
[16:26:26][C][logger:187]:   Log Baud Rate: 115200
[16:26:26][C][logger:188]:   Hardware UART: UART0
[16:26:26][C][dallas.sensor:072]: DallasComponent:
[16:26:26][C][dallas.sensor:073]:   Pin: GPIO32 (Mode: INPUT)
[16:26:26][C][dallas.sensor:074]:   Update Interval: 60.0s
[16:26:26][D][dallas.sensor:079]:   Found sensors:
[16:26:26][D][dallas.sensor:082]:     0x0B030C9794530428

```
**Additional information and things you've tried:**

the above test was done with a simple breadboard with nothing else connected than the dallas sensor, resistor 4.7K doesn't change the overall issue.

I can't be sure at 100% but when I tested before making the PCB it was working, and was epshome v. 1.4.3 was working, then after assembly the final PCB and after the last upgrade to 1.5.3 is not working anymore... :( !

randybb, 2020-12-09:
GPIO34, GPIO35, GPIO36 and GPIO39 are input only - 1-wire is a bi-directional bus.

lucagiudici, 2020-12-09:
Ok - 
understood, many thanks.

How we can make this clear into the doc ? ( https://esphome.io/components/sensor/dallas.html )

closed 2020-12-09 by lucagiudici

randybb, 2020-12-09:
To check a reference guide is one of the essential steps you have to do when you work with electronics, especially with MCU's. Sure, we are dealing only with ESP8266 and ESP32, but the same approach is related to every single peripheral that it can support. ESP is quite simple, but the options and limitations of other MCU's are crazy and you have to always check a dedicated reference guide for your part number.