esphome/issues #1684: OTA stopped working. Error: Finishing update failed.

davida72, 2020-12-09:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

**Home Assistant OS / Wemos D1 Pro**

**ESPHome version (latest production, beta, dev branch)**

**Affected component:**

**Description of problem:**

**Problem-relevant YAML-configuration entries:**
```yaml
esphome:
  name: nullo
  platform: ESP8266
  board: d1_mini_pro

wifi:
  ssid: "timor"
  password: "password"
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 192.168.86.69
    # Set this to the IP address of the router. Often ends with .1
    gateway: 192.168.86.1
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0  

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Tunnelleds Fallback Hotspot"
    password: "IxFsIotrRrSE"

light:
  - platform: fastled_clockless
    chipset: WS2811
    pin: GPIO2
    num_leds: 50
    rgb_order: RGB
    name: "FastLED WS2811 Light"
    effects:
      - addressable_rainbow:

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

```

**Logs (if applicable):**

```
INFO Reading configuration /config/esphome/nullo.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/nullo
Processing nullo (board: d1_mini_pro; framework: arduino; platform: espressif8266@2.6.2)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 16MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 3.20704.0 (2.7.4) 
 - tool-esptool 1.413.0 (4.13) 
 - tool-esptoolpy 1.20800.0 (2.8.0) 
 - toolchain-xtensa 2.40802.200502 (4.8.2)
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.3
|   |-- <ESP8266WiFi> 1.0
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.7
|   |-- <ESPAsyncTCP-esphome> 1.2.3
|   |   |-- <ESP8266WiFi> 1.0
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <FastLED> 3.3.2
|   |-- <SPI> 1.0
|   |-- <EspSoftwareSerial> 6.8.5
|-- <DNSServer> 1.1.1
|   |-- <ESP8266WiFi> 1.0
Retrieving maximum program size /data/nullo/.pioenvs/nullo/firmware.elf
Checking size /data/nullo/.pioenvs/nullo/firmware.elf
RAM:   [====      ]  44.7% (used 36632 bytes from 81920 bytes)
Flash: [====      ]  39.8% (used 416192 bytes from 1044464 bytes)
========================= [SUCCESS] Took 11.60 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.168.86.69
INFO Uploading /data/nullo/.pioenvs/nullo/firmware.bin (420352 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
ERROR Error Update end: Error: Finishing update failed. See the MQTT/USB logs for more information.

```

I've been able to compile and upload via USB and OTA was working fine previously but I've done something that's stopped it working. I've tried reinstalling ESPHome on Home Assistant, I've rebuilt new firmware, I've added and removed the device from Home Assistant. I can't see anything useful in the logs. 

Any idea why I'm no longer able to upload my firmware OTA?

davida72, 2020-12-10:
Update: As a workaround I've switched from _Wemos D1 mini Pro_ to  _Generic ESP8266_ and that seems to work fine.

qbunt, 2020-12-18:
@davida72 I just had a similar issue with a Wemos D1 clone, end erasing the flash memory first via esptool before flashing over USB did the trick.
`esptool.py -p /dev/[usb-port] erase_flash`

thecreaser, 2020-12-20:
Same issue here.
Raspberry Pi Home Assistant OS 5.8  Core version 2020.12.0
ESPHome latest production version via Home Assistant Integrations

I have clones of Wemos D1 Minis and D1 Mini Pros to compare them.
No problems with D1 Mini but the D1 Mini Pros no longer not flash OTA.  I had seen one successful flash once but can't identify what might have changed or been updated since.

If the Mini Pro device types are Generic ESP8266 then they will flash OTA with no errors.  It's only if they are set up as D1 Mini Pro that the OTA fails.
No other error messages in any logs.

davida72, 2020-12-20:
> erasing the flash memory first via esptool before flashing over USB did the trick.

Thank you, although this didn't work for me. I erased using the latest version of esptool.py over USB, then uploaded a _board: d1_mini_pro_ version over USB, which worked fine. When I tried to upload the same version OTA I got the error message. When I recompiled a _board: esp01_1m_ version then that uploaded OTA without a problem.

davida72, 2020-12-30:
USER ERROR.

I thought my Wemos was a pro but it's not. The pro is 16MB, not 4MB. My bad!

closed 2020-12-30 by davida72

SAOPP, 2021-03-02:
Guys, latest esphome dev in docker and wemos d1 mini pro, the same things... I can't update firmware via OTA.

`ERROR Error Update end: Error: Finishing update failed. See the MQTT/USB logs for more information.`

556duckvader, 2021-04-30:
This is still any issue.  I still can't upload to an Wemos D1 mini pro.

pirx7, 2021-08-13:
**same with esp32-Cam**
using ESPHome-Flasher-1.3.0-Windows-x64.exe from https://github.com/esphome/esphome-flasher/releases
for initially flashing.
But it is working with WEMOS-D1-Mini (cheap clone from China)

Johansamreji, 2021-10-06:
Switching from `board: d1_mini_pro` to `board: d1_mini_lite` fixed my issue.
I think Cause _**D1 mini Lite**_ has **4MB** Flash while _**D1 mini Pro**_ has **16MB** flash...

_Basically Just Check Which Type of D1 mini you have Before Uploading the .yaml._