esphome/issues #1687: DHT Temp sensor convert to Fahrenheit still gives me Celsius but with *F

TCPMeta, 2020-12-12:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

ESP8266MOD
**ESP (ESP32/ESP8266, Board/Sonoff):**

Latest Production
**ESPHome version (latest production, beta, dev branch)**

**Affected component:**

https://esphome.io/components/sensor/dht.html
https://esphome.io/components/sensor/index.html#sensor-filters

**Description of problem:**
Converted Celsius to Fahrenheit but still gives me a Celsius output. For a example it's around 68 Degrees Fahrenheit but the output says 20*F but if I convert the so called 20*F it says 68.

**Problem-relevant YAML-configuration entries:**
```yaml
PASTE YAML FILE HERE

esphome:
  name: magicmirror
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: ""
  password: ""

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Magicmirror Fallback Hotspot"
    password: ""

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: ""

ota:
  password: ""

sensor:
  - platform: dht
    pin: GPIO5
    temperature:
      name: "Magic Mirror Temperature"
      filters:
        - lambda: return x * (9.0/5.0) + 32.0;
      unit_of_measurement: "°F"
    humidity:
      name: "Magic Mirror Humidity"
    update_interval: 60s
    model: DHT11
```

**Logs (if applicable):**

[11:12:24][C][logger:185]: Logger:
[11:12:24][C][logger:186]:   Level: DEBUG
[11:12:24][C][logger:187]:   Log Baud Rate: 115200
[11:12:24][C][logger:188]:   Hardware UART: UART0
[11:12:24][C][dht:017]: DHT:
[11:12:24][C][dht:018]:   Pin: GPIO5 (Mode: INPUT)
[11:12:24][C][dht:022]:   Model: DHT11
[11:12:24][C][dht:027]:   Update Interval: 60.0s
[11:12:24][C][dht:029]:   Temperature 'Magic Mirror Temperature'
[11:12:24][C][dht:029]:     Unit of Measurement: '°F'
[11:12:24][C][dht:029]:     Accuracy Decimals: 1
[11:12:24][C][dht:029]:     Icon: 'mdi:thermometer'
[11:12:24][C][dht:030]:   Humidity 'Magic Mirror Humidity'
[11:12:24][C][dht:030]:     Unit of Measurement: '%'
[11:12:24][C][dht:030]:     Accuracy Decimals: 0
[11:12:24][C][dht:030]:     Icon: 'mdi:water-percent'
[11:12:24][C][captive_portal:169]: Captive Portal:
[11:12:24][C][ota:029]: Over-The-Air Updates:
[11:12:24][C][ota:030]:   Address: magicmirror.local:8266
[11:12:24][C][ota:032]:   Using Password.
[11:12:24][C][api:095]: API Server:
[11:12:24][C][api:096]:   Address: magicmirror.local:6053
```
PASTE DEBUG LOG HERE
[16:28:58][D][dht:048]: Got Temperature=-6.5°C Humidity=4.0%
[16:28:58][D][sensor:092]: 'Magic Mirror Temperature': Sending state 20.30000 °F with 1 decimals of accuracy
[16:28:58][D][sensor:092]: 'Magic Mirror Humidity': Sending state 4.00000 % with 0 decimals of accuracy
```

**Additional information and things you've tried:**
Replaced the Sensor, Tried different GPIO pins and tried a different ESP Module, Yelled at it, spit on it and hitting it.

ronytomen, 2020-12-17:
The log entry is `dht` component and the sensor filter has no effect on how the `dht` components log values.

This is not an issue, it is working as designed. :)

If you are using Home Assistant this does not matter for the logs or even the sensor value is in Celsius.  Home Assistant will automatically adjust temperature sensors from Celsius to Fahrenheit if configured to do so.

glmnet, 2020-12-17:
this answer is correct, closing

closed 2020-12-17 by glmnet