esphome/issues #1688: Sonoff Basic upgrade 1.15.1 → 1.15.3 breaks connectivity when adding web server or text_sensor

wsw70, 2020-12-12:
**Operating environment/Installation (Hass.io/Docker/pip/etc.):**

Official docker image (`esphome/esphome`)

**ESP (ESP32/ESP8266, Board/Sonoff):**

Sonoff Basic

**ESPHome version (latest production, beta, dev branch)**

The issue is with the upgrade from 1.15.1 (works) to 1.15.3

**Affected component:**

Access to the WiFi AP

**Description of problem:**

I had a working Sonoff Basic with 1.15.1, connected to my WiFi. Though the web interface, I updated it to 1.15.3. As part of the upgrade, I added to the YAML configuration the following entries:

```
web_server:

text_sensor:
  - platform: version
    name: "ESPHome version"
  - platform: template
    name: "Configuration version"
    lambda: |-
      return {"1"};
```

After a successful flash, the device did not reconnect.

**Problem-relevant YAML-configuration entries:**
```yaml

```

**Logs (if applicable):**

```

```

**Additional information and things you've tried:**

The exact description of the issue is:

- Sonoff with "base" configuration + 1.15.1 → works
- Sonoff with "base" configuration + 1.15.1 + the extra YAML entries above → works
- Sonoff with "base" configuration + **1.15.3** → works
- Sonoff with "base" configuration + **1.15.3** + the extra YAML entries above → **does not work - the device does not connect to WiFi**

**So the problem is either with the web server or the extra `text_sensor` entries that would have had appeared between 1.15.1 and 1.15.3.**

The device reacts to the button press. It does not propose its own WiFi AP (the fallback one) despite having one defined.

mcc05, 2020-12-15:
I confirm I have the same issue on two sonoff basics. with the webserver component present, they fail to connect to wifi, if I remove it they work fine

user897943, 2021-01-06:
It's unbelievable. I am still holding on version 1.13.6 because of WiFi issues. 1.13.6 works, I tried the .14 branch but that was not connecting to WiFi, I had to flash with a serial to usb cable to get it back to 1.13.6 which worked fine. It's been several versions since and this issue still exists. Now I know I'm not contributing anything here but if I was, I'd feel extremely ashamed. Imagine embedding your sonoff behind drywall, then sending a OTA that kills WiFi, and now you have to break the wall to fix it. It's a seriously stupid issue and brings this project into disrepute. Please fix or take this whole thing offline, it's embarrassing and a huge waste of time for everyone.

wsw70, 2021-01-23:
I checked the diff between the two versions and the `web_server` component was not changed, and neither was something obviously related to WiFi.

It may be a change in the other components (or core) but this is way above my coding capabilities :)

glmnet, 2021-01-25:
Are you explicitly setting the output power on the wifi component?

wsw70, 2021-01-25:
> Are you explicitly setting the output power on the wifi component?

Sorry, I do not know ESPHome enough to understand your question. What exactly should be set? (I provided the complete configuration in the description of the issue)

wsw70, 2021-05-28:
Has the issue been fixed? (I am pinging because the "stale bot" is marking this for automated close)